marq24/ha-senec-v3 #1: Neue

Cyrelion, 2023-04-14:
Hallo Matthias,

danke für die Weiterentwicklung des HACS Addons, das ist eine wirklich große Hilfe!

Ich habe nur eine kurze Frage: wäre es möglich bei deinem Fork evtl. die Netzfrequenz mit aufzunehmen und als Entity bereitzustellen?

PM1OBJ1 FREQ -> Zeigt bei mir 49.96
PM1OBJ2 FREQ -> wobei das bei mir 0 anzeigt, kA warum.

Das ist auch in der read_senec_v31 function schon definiert, aber ich hab mich jetzt nicht wirklich mit dem Python Code beschäftigt, ich komme eher aus der PHP Ecke :-)

Vielen Dank, wenn ich irgendwie helfen kann sag gerne Bescheid.

Gruß,
Cyrelion

Cyrelion, 2023-04-14:
Okay PM1OBJ2 ist wohl for nen zweien Enflui, hab ich nicht, also ist es richtig dass ich da ne 0 habe :-D

marq24, 2023-04-14:
just update to 2.2.3...
and thanks a lot for your suggestion!

PM1OBJ2 is listed as 'ENFLURI - VERBRAUCH' here - and also contains no data here

closed 2023-04-14 by marq24

Cyrelion, 2023-04-14:
Works, great! Thank you