marq24/ha-senec-v3 #3: Support HTTPS

marq24, 2023-08-30:
Recent Senec release introduced (self signed) https communication between browser & device

marq24, 2023-08-30:
Please let me know, if this PRE-Release will work for you
https://github.com/marq24/ha-senec-v3/releases/tag/pre3.0.5https

Have in mind in HACS you have to add this [https://github.com/marq24/ha-senec-v3] repo first and second you have to enable the BETA-BUILD switch

marq24, 2023-08-30:
ok looks like that in the new build the `STATISTIC` are empty

```
    "STATISTIC": {
        "LIVE_BAT_CHARGE": "VARIABLE_NOT_FOUND",
        "LIVE_BAT_DISCHARGE": "VARIABLE_NOT_FOUND",
        "LIVE_GRID_EXPORT": "VARIABLE_NOT_FOUND",
        "LIVE_GRID_IMPORT": "VARIABLE_NOT_FOUND",
        "LIVE_HOUSE_CONS": "VARIABLE_NOT_FOUND",
        "LIVE_PV_GEN": "VARIABLE_NOT_FOUND"
    },
```

So the question is, when you have the original app running, from which fields of the new lala.cgi response the new LIVE data values are read...

Basti-Fantasti, 2023-08-30:
can you paste the API call to put in curl or postman? 
Then I can paste the results from my v2.1 here.

I just received the https update 2-3 hours ago

Basti-Fantasti, 2023-08-30:
I've checked the files in the component and found the old url to be:

http://senec-ip/lala.cgi

is this correct to call the API? Are any additional parameters needed?
because if I call it with http or https using postman I always get this reply:

```html
<html><head><title>Not Found</title></head><body><h1>Not Found</h1><p>Resource `lala.cgi` could not be found.</p></body></html>
```

marq24, 2023-08-30:
Have in mind when you use postman, that you actually make a POST not a GET to the lala.cgi... as raw request body you can use (and use JSON as ContentType):

```
{
"BAT1":{},
"BAT1OBJ1":{},
"BMS":{},
"BMS_PARA":{},
"CASC":{},
"DEBUG":{},
"DISPLAY":{},
"ENERGY":{},
"FACTORY":{},
"FEATURES":{},
"FILE":{},
"GRIDCONFIG":{},
"ISKRA":{},
"LOG":{},
"PM1":{},
"PM1OBJ1":{},
"PM1OBJ2":{},
"PV1":{},
"PWR_UNIT":{},
"RTC":{},
"SELFTEST_RESULTS":{},
"SOCKETS":{},
"STATISTIC":{},
"STECA":{},
"SYS_UPDATE":{},
"TEMPMEASURE":{},
"TEST":{},
"UPDATE":{},
"WALLBOX":{},
"WIZARD":{}}
```

Basti-Fantasti, 2023-08-30:
Ok, got it 😄 

here you go, this is the pretty output, only serialnumbers slightly adjusted:

```json
{
    "BAT1": {
        "CEI_LIMIT": "u1_0000",
        "DRM0_ASSERT": "u8_00",
        "ISLAND_ENABLE": "u8_00",
        "NSP2_FW": "u1_00F3",
        "NSP_FW": "u1_00F3",
        "REQ_HV_UPD": "u8_00",
        "REQ_LV_UPD": "u8_00",
        "RESET": "u8_00",
        "SELFTEST_ACT": [
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000"
        ],
        "SELFTEST_LIMIT": [
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000"
        ],
        "SELFTEST_OFF": [
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000",
            "fl_00000000"
        ],
        "SELFTEST_OVERALL_STATE": "u8_00",
        "SELFTEST_STATE": [
            "u8_00",
            "u8_00",
            "u8_00",
            "u8_00",
            "u8_00",
            "u8_00",
            "u8_00",
            "u8_00"
        ],
        "SELFTEST_STEP": [
            "st_",
            "st_",
            "st_",
            "st_",
            "st_",
            "st_",
            "st_",
            "st_"
        ],
        "SELFTEST_TIME": [
            "u1_0000",
            "u1_0000",
            "u1_0000",
[... 1240 lines cut ...]
        ],
        "SETUP_USED_PHASE": "u8_01",
        "SETUP_USE_ABS_POWER": "u8_00",
        "SETUP_USE_DRM0": "u8_00",
        "SETUP_USE_RCR": "u8_00",
        "SETUP_WALLBOX_SERIAL0": "st_1W22G123456",
        "SETUP_WALLBOX_SERIAL1": "st_",
        "SETUP_WALLBOX_SERIAL2": "st_",
        "SETUP_WALLBOX_SERIAL3": "st_",
        "SG_READY_CURR_MODE": "u8_01",
        "SG_READY_ENABLED": "u8_00",
        "SG_READY_ENABLE_OVERWRITE": "u8_00",
        "SG_READY_EN_MODE1": "u8_00",
        "SG_READY_OVERWRITE_RELAY": [
            "u8_00",
            "u8_00"
        ],
        "SG_READY_POWER_COMM": "u1_FFFF",
        "SG_READY_POWER_NORMAL": "u1_0000",
        "SG_READY_POWER_PROP": "u1_FFFF",
        "SG_READY_TIME": "u1_02D0",
        "ZEROMODULE": "u8_00"
    }
}
```

Basti-Fantasti, 2023-08-30:
it's strange that the **STATISTIC** topic is totally empty and no **LIVE_** Data seems to be included at all

marq24, 2023-08-30:
it's `STATISTIC`, `DISPLAY`, 'SELFTEST_RESULTS', `STECA`, `TEST `and `UPDATE` that are empty (now)... well - what I make sure is, that the integration will at least not crash when trying to access these fields... Most of them I do not care much... but `STATISTIC` will hurt - since it's the source for the HomeAssistent Energy Information... Of-Course I can create a OWN "sum from now" Sensor... but *sigh* only SENEC knows what's the reason behind this...

marq24, 2023-08-30:
> Nice but I cant find this "Fork" from you in HACS. Only the original one ....
you have to add a "custom repo in HACS...
[image]

marq24, 2023-08-30:
> Thanks I´ve found it and now what I have to put in here? I didnt find any description.
Use SHIFT-REOLAD in order that the translation text's will be displayed... The second input field is the IP/HOST-name of your device

electricdelicate, 2023-08-30:
> > Thanks I´ve found it and now what I have to put in here? I didnt find any description.
> 
> Use SHIFT-REOLAD of your browser in order that the translation text's will be displayed... The second input field is the IP/HOST-name of your device

Failed to connect, no matter what I type. 
[image]

afraschetti, 2023-08-30:
> > > Thanks I´ve found it and now what I have to put in here? I didnt find any description.
> > 
> > 
> > Use SHIFT-REOLAD of your browser in order that the translation text's will be displayed... The second input field is the IP/HOST-name of your device
> 
> Failed to connect, no matter what I type. [image]

Same problem

neubi4, 2023-08-30:
Same for me. I think the config flow is missing the option for https. I tried pre3.0.5https-v2

marq24, 2023-08-30:
actually there is no need for the user to select http OR https - since the code will try http first, and if connection failed, then the code is trying to connect via https - at least that's the theory... I have tested this here locally with my device here (but with the inverse logic - trying https first and then fallback to http - and this is working here as expected... I will debug further...

In the meantime - is there any output in the ha log ?! - if yes please share... TIA

marq24, 2023-08-30:
just enter '_192.168.178.172_' - do __not__ include http:\\ or https:\\

electricdelicate, 2023-08-30:
> just enter '_192.168.178.172_' - do **not** include http:// or https://
I have it already, every time the same.

marq24, 2023-08-30:
log say something different... but I am currently enrich the logging information... so we will see

neubi4, 2023-08-30:
Can you clarify which version we should use? I am on pre3.0.5https-v2, using just the IP i get this log:

```
2023-08-30 20:41:06.877 ERROR (MainThread) [custom_components.senec.config_flow] Could not connect to SENEC.Home at 192.168.251.135, check host ip address
```

I looked at the code and i think the problem is that when using https, aoihttp is getting the senec self singed certificate, which is not valid, so it gets an error.

You shoud set ssl=False every time you make a request in pysenec_ha to an url (get and post) to ignore the invalid self singed certificate. https://docs.aiohttp.org/en/stable/client_advanced.html#ssl-control-for-tcp-sockets

marq24, 2023-08-30:
beside additional logging in this build I have set 'ssl=False' :-D
https://github.com/marq24/ha-senec-v3/releases/tag/pre3.0.5https-v3

marq24, 2023-08-30:
> Can you clarify which version we should use? I am on pre3.0.5https-v2, using just the IP i get this log:
> 
> ```
> 2023-08-30 20:41:06.877 ERROR (MainThread) [custom_components.senec.config_flow] Could not connect to SENEC.Home at 192.168.251.135, check host ip address
> ```
> You shoud set ssl=False every time you make a request in pysenec_ha to an url (get and post) to ignore the invalid self singed certificate. https://docs.aiohttp.org/en/stable/client_advanced.html#ssl-control-for-tcp-sockets

Thank you very much! - I just googled this few minutes ago - and created the new build (v3) - after that I saw your reply - I am happy that this might/will solve the issue... (at least I have some hope) :-D

marq24, 2023-08-30:
> beside additional logging in this build I have set 'ssl=False' :-D https://github.com/marq24/ha-senec-v3/releases/tag/pre3.0.5https-v3

@neubi4 @electricdelicate - please update and try again (I am also trimming any http or https entered by the user - so the code is for sure only handing the senec host/ip...)

neubi4, 2023-08-30:
It works, thank you very much :)

Besides the missing Statistics, all values are working. I think i might having a workaround for this in the other issue.

Please send me an Paypal "Geld an Freunde senden" Link or Mail Adress, i want to send you at least the money for an pizza or something for your hard work.

marq24, 2023-08-30:
> Please send me an Paypal "Geld an Freunde senden" Link or Mail Adress,...
you can use '@marq24' - but I am happy that we found a solution (step1)...

I will now prepare a "real" build - so that everybody can update to this fork (without the beta step)...

Basti-Fantasti, 2023-08-30:
thanks for your great and quick assistance 👍

marq24, 2023-08-30:
- cool - solved - in less then 24h

closed 2023-08-30 by marq24

electricdelicate, 2023-08-30:
> @neubi4 @electricdelicate - please update and try again (I am also trimming any http or https entered by the user - so the code is for sure only handing the senec host/ip...)

Work, thank you! :)

Basti-Fantasti, 2023-08-30:
> > Please send me an Paypal "Geld an Freunde senden" Link or Mail Adress,...
> 
> you can use `@marq24` - but I am happy that we found a solution (step1)... I will now prepare a "real" build - so that everybody can update to this fork (without the beta step)...

So the 3.0.5 version / branch should be selected when adding your github repo?

Showazter, 2023-09-03:
Are there any todal day values (total grid imported/exported, solar generatet... in (kWh))?

marq24, 2023-09-03:
> Are there any todal day values (total grid imported/exported, solar generatet... in (kWh))?

nope (not yet) - this data is read from the web-api - but actually I did not include any sensor entities for these values because:

1. These values had not been available in the original `lala.cgi`-based-integration - so nobody had them before
2. The HA-Energy Tab don't need these values either (the Energy Tab calculate these values automatically)
3. If you want/need monthly/weekly/daily/hourly entities you can easily set them up via 'Utility Meters' Helpers (Verbrauchszähler in German)

So I continue enjoy the rest of my short weekend...

Showazter, 2023-09-03:
Thanks for your reply.
In energy tab I can only choose consumption values (Wh, kWh), not power values (W, kW).

marq24, 2023-09-03:
> Thanks for your reply. In energy tab I can only choose consumption values (Wh, kWh), not power values (W, kW).

probably I am confused - is this a new question - or a reply... Obviously in EnergyTab there are only consumption (kW__h__) values... displayed as total today/week/month (when this is running for a while)...

The current power values [which have to be kW (and not W - I just released 3.0.6-test-04 to fix that)] are not available as any kind of total or total_day... Since it's the "actual" power usage... I have put **actual** in quotes here, since we just have every 5 minutes a value... (via the new WebAPI)

Showazter, 2023-09-03:
For e.g: which entity do I have to insert here? In older version I used total_grid_import.
[image]

marq24, 2023-09-03:
> For e.g: which entity do I have to insert here? In older version I used total_grid_import. 

depending on your selected 'name' - (if you use the default 'senec_webapi') it's `sensor.senec_webapi_gridimport_total`

Showazter, 2023-09-03:
Thankyou, till now I didn't use webapi.

marq24, 2023-09-04:
Hi @melotti892!

could you please be so kind (to double check), that

a) https://mein-senec.de/endkunde/api/context/getAnlageBasedNavigationViewModel?anlageNummer=0

returning the data for your battery A and...

b) https://mein-senec.de/endkunde/api/context/getAnlageBasedNavigationViewModel?anlageNummer=1

is returning the data for your second hardware? - if this is the case, then I have to make sure, that for users like you, there is the option to specify the number of devices that have to be captured... I will see, if (and how) I find there some working concept - since I would not like the "simplest approach" - just using two web-sessions and polling twice...

melotti892, 2023-09-04:
> 

in a) i get the slave one, and in b) i get the master one, is this you need right?

marq24, 2023-09-04:
thanks for the quick check - I assume this (specially that the anlageNummer=0 is tagged as 'slave'... So I guess I can iterate though the numbers, till a system will return master... and use then "this" number... 

But I hope you can understand, that I will spend my Feierabend with something different ;-)

melotti892, 2023-09-04:
@marq24 
i studied what you said
i can confirm all (like 0 is slave and so on)
and obv as you spend your time in what you prefer
just hope to have helped :)

marq24, 2023-09-04:
@melotti892 can you check the latest build please? [remove the installed integration first - and than reinstall it...]
This time the code will query the webportal till it found the master... and use then the master-number in all later requests

melotti892, 2023-09-04:
@marq24 3.0.6 test 5, reinstalled all the senec related integration but it still recognise the slave one (i don't know what tipe of sistem of battery i have, like the 4 or 3 or 2 )
i think 3 accordingly the senec.de site

marq24, 2023-09-04:
... after you have updated the integration - did you actually restarted your HA? - and after the restarted added the integration again?

And if you have a V3 - what was your selection at the integration start? - I have "adjusted" the web-api paort only (since that's the new stuff)... now I am confused, if you are using the 'lala.cgi" LAN-local-polling based variant?

melotti892, 2023-09-04:
@marq24 
now i've uninstalled all
restarted ha
reinstalled it
restarted ha
and configured straight on the part of web-api
it gives me the slave one again
 (i haven't opened the battery case in the parallel guide where there is to connect it trought lan)
i know the installator placed a wifi port throught it is connected to wifi and to the senec portal i guess but i didn't made anything more of "hardware"

marq24, 2023-09-04:
thanks... for your clarification - so we are in the new webapi context... I'll check again the code... A bit difficult to debugging "blind"...but we will see...

Concerning connected SenecV3 via LAN... -> https://sollis.de/wp-content/uploads/TD140-057.10_SENEC.Home_V3_Installationsanleitung_1.0.pdf - page 20 - the 'ETH' Port is the one that should be connected to your LAN (but I guess that's already done)

melotti892, 2023-09-04:
If you need anything more specific just askKeep in mind that i don’t know for real anything, ha was installare by a friend 3 month ago and i’m still understanding itAlessandro Melotti

melotti892, 2023-09-04:
Here you are
As you can see it’s the slave one
Previously where you gave me something that end ed With =1 it said that it was correctly the master
With it’s own serial
{"steuereinheitnummer":"S452202814460xxxxxxxxxxxxxx","master":false,"produktName":"SENEC.Home V3 hybrid","statusVisible":true,"autarkieVisible":true,"akkuVisible":true,"statisticDataDownloadVisible":true,"dokumenteVisible":true,"steckdosenVisible":false,"steckdosenEnabled":false,"peakShavingVisible":false,"heizstaebeVisible":false,"technischeDatenVisible":true,"econamicGridVisible":false,"econamicGridVarioVisible":false,"zusatzpaketeVisible":true,"wartungsladungVisible":false,"serviceVisible":true,"zoneId":"Europe/Rome","senecCloudVisible":false,"offpeakVisible":false,"digitalAssistantVisible":false,"wallboxVisible":false,"senecOpenApiVisible":false,"sgReadyVisible":false,"emergencyPowerVisible":false}

marq24, 2023-09-05:
thanks - please rerun the procedure with https://github.com/marq24/ha-senec-v3/releases/tag/3.0.6-test-07

melotti892, 2023-09-05:
@marq24 
ok now the integration recognise the master one and choose it
and only it's values are calcolated
at this point it's just left out the part of the slave one

melotti892, 2023-09-05:
@marq24 made it as you suggested
for the other part of the w recognised as Kw, maybe the "," is recognised as the "." and viceversa?
is there something that i have to change or is the senec-portal?

Thomk-63, 2023-09-07:
Thank you marq24 for the fix!!! I have a Senec V3 hybrid, recognized the problem today in the morning and could fix it with your integration.
Thanks alot! Great work!