marq24/ha-senec-v3 #4: Since SENEC Application v825 no STATISTIC data is provided [No Data for multiple Sensors]

marq24, 2023-08-30:
When your SENEC.Home will be updated to v825 the integration is not able to provide data for a couple of sensors..

Affected Sensors:

- House consumed
- Solar generated
- Battery charged
- Battery discharged
- Grid Imported
- Grid Exported
- Wallbox charged

The reason behind is, that SENEC has decided to exclude 'STATISTIC' information from the response of the 'lala.cgi'...

electricdelicate, 2023-08-30:
Oh no ....

neubi4, 2023-08-30:
If i got is correctly, these are all "daily usage" statistics data, e.g. how many kWh House consumed today.

This should be replaceable by using an Home Assistant Riemann sum integral (https://www.home-assistant.io/integrations/integration/#energy) which simply can summarize House Power (W) to kWh like in the linked example. 

Since this counts infinitely higher, to get an daily used you can use Utility Meter to get an sensor on the integral which is resetted daily/monthly value.

These Utility Meter sensors then could be used in Home Assistents Energy View.

We cloud provice YAML examples for this in the readme of this repo.

neubi4, 2023-08-30:
After looking at the old data, it seems that the Utility Metern is not needes since the old data also only counts upwards infinitely.

So it should be fine only using the Riemann sum integral sensors. I will try this.

marq24, 2023-08-30:
FUNNY - while I am tyring - you already provide the info... 

the values (that are still available in my environment) are constantly growing - so there is no need to reset them on a monthly/weekly/daily/hourly basis - so looks like using the suggested integration from @neubi4 sounds like a decent replacement...

Once the http-v3 build is confirmed as "working" I can spend some time working on example yaml parts that you can copy&paste in your ha configuration.yaml...

electricdelicate, 2023-08-30:
Sounds great, thank you for your support!

egonkab, 2023-08-30:
Hello marq24
Thank you for your hard work. The beta version works fine.
I try to use powercalc to calculate the daily kWh. I'll report after my test.

marq24, 2023-08-31:
added workaround information...

egonkab, 2023-08-31:
hello marq24,

thanks for your hard work. It works very well. 
I use Powercalc to calculate the total sensors,  this also works very well.

sigcarunchio, 2023-08-31:
Hello marq24, just discovered your fork after firmware update, great work!

1) About missing entities, i also redo them using custom sum integration templates, but instead using yaml code, i use helpers; do you think it's the same?

2) My problem is that i have a master and a slave senec v3; i think that missing statistics gave already total values, now (but i confirm this later) with custom sensor i have only data about master senec (the one that i register in the integration); do i need to register also slave battery and do total custom sensors?

3) Worst thing when i change sensors in energy panel with custom ones, I LOST all past data of about one year!!!!! sigh!!!!

anyway again great work, thank you!

neubi4, 2023-08-31:
Hi @sigcarunchio 

1.: yes, using Helpers and the yaml code is the same.

3.: i left the old sensors with their current `VARIABLE_NOT_FOUND` value in the energy dashboard and only added the new helper sensors. This way i still have the old stats data in my energy dashboard.
I dont know exactly how the dashboard works internally, it is possible that readding the old sensors will not work for you.

2.: This is an interesting one since not many users have multiple systems. Does the slave senec also has an seperate IP and the webinterface? If yes, then i would try to add the slave additionally to the master.
To get the total stat values of both systems, you could try to calculate the sum of the values from both senecs and calculate the Riemann sum integral on this value.

sigcarunchio, 2023-08-31:
3. Ok, i replace them, try to reinsert also old sensors

2. Yes, it has a separate IP with the same web interface, try this way!

sigcarunchio, 2023-08-31:
Readded also old sensors, past stats availables again!
As soon as i will add also slave senec, I will report if it works

bobthebuilder4711, 2023-08-31:
It also seems like the update reset the settings for our Home V2.1 to poll our three SMA inverters. Solar Power just stays at 0W. Anyone else has this problem?

Don't have the installer's password for the senec battery, so I guess I'll have to add the inverters to home assistant directly...

Kuechenfenster, 2023-08-31:
[](https://github.com/mchwalisz/home-assistant-senec/issues/47#issuecomment-1701014858)

marq24, 2023-09-01:
Please note, there was a typo in my provided configuration.yaml

in the `senec_grid_total_export_fb` the source has to be: `source: sensor.senec_grid_exported_power` (and not `source: sensor.senec_senec_grid_exported_power`) - sorry...

marq24, 2023-09-01:
@Kuechenfenster 
> [https://github.com/mchwalisz/home-assistant-senec/issues/47#issuecomment-1701014858](https://github.com/mchwalisz/home-assistant-senec/issues/47#issuecomment-1701014858)

could you let us know, what you wanted to say?

marq24, 2023-09-01:
Since here (locally) I still have the http version - I am able to compare the values [created via the `integration` platform vs data from the senec statistics json data]

[image]
[image]

Kuechenfenster, 2023-09-01:
> @Kuechenfenster
> 
> > [mchwalisz#47 (comment)](https://github.com/mchwalisz/home-assistant-senec/issues/47#issuecomment-1701014858)
> 
> could you let us know, what you wanted to say?

my apology.. please ignore it.

electricdelicate, 2023-09-01:
> > added workaround information in the initial post
> 
> Good job, work also like a charm. One question, is the "Riemann sum" exact enough for a proper calculation? I will compare that to the Senec App.

As far as I can tell at my system, the Riemann sum is only correct at my charging value, the discharging value is very deviating to MySenec/official App ....

marq24, 2023-09-02:
here is a comparison of the 'discharging value' here @ my home... YES the numbers are different... 
[image]
[image]

Just as Update (working on a better replacement) - I have spend already some hours trying to access mein-senec.de - but failed with the aiohttp python classes to get an appropriate login (with the request-lib this works smooth like silk) - one of the many redirects is failing

egonkab, 2023-09-02:
Hello, 
the powercalc result from 09/01/2023

[image]

[image]

marq24, 2023-09-02:
Passt über den dicken Daumen - oder? [It just took me a while to realize, that top part (of the image) is from SENEC-App and the lower part is from HA-Energy-Tab (I assume feeded by the 'integration' sensors...

egonkab, 2023-09-02:
Ich schwenke mal auf deutsch um. Die Werte aus dem HA Energy Tab sind auf der PowerCalc Berechnung. 
Ich habe am zum Vergleich folgende Sensoren aufgesetzt (senec_grid_exported_power), Ergebnis: 

PowerCalc: 14,53kWh   -   RMS Trapez: 14,12kWh   -   RMS Left: 14,19kWh  -  Senec App: 14,5kWh

BMeyendriesch, 2023-09-02:
Könntest Du mal bitte die powercalc.yaml für obige Ausgabe mitteilen.

BMeyendriesch, 2023-09-03:
Hat sich erledigt. Ich hatte mir mit gleichzeitigen Konfigurationsversuchen per GUI und yaml selbst auf dem Fuß gestanden . . .

marq24, 2023-09-03:
hier teilen wäre 'trotzdem' schön ;-) - Gute Morgen btw..

marq24, 2023-09-03:
updated initial post -> added a solution alternative: __mein-senec.de WebPortal Integration Entry__

BMeyendriesch, 2023-09-03:
Ich habe hier einen Haufen von Helpern und darauf aufbauenden Sankey-Diagrammen, die auf den ursprünglichen Entities der v2.1-Integration 

```
sensor.senec_solar_total_generated
sensor.senec_house_total_consumption
sensor.senec_grid_total_import
sensor.senec_grid_total_export
sensor.senec_battery_total_charged
sensor.senec_battery_total_discharged

```
basieren und würde nur sehr ungern sämtliche Abhängigkeiten von diesen Entities anfassen und modifizieren. Gibt es nicht eine schlanke Methode, die Ausgangswerte von powercalc in diese Entities umzuleiten? Bisher habe ich powercalc nicht dazu überreden können. Möglicherweise habe ich auch das gesamte Konzept der Entities in Homeassistant nicht hinreichend verstanden . . .

marq24, 2023-09-03:
> Ich habe hier einen Haufen von Helpern und darauf aufbauenden Sankey-Diagrammen, die auf den ursprünglichen Entities der v2.1-Integration
>
> ```
> sensor.senec_solar_total_generated [...]

ich verstehe Deine Herausvorderung - mit der gerade veröffentlichten web-api variante kommst Du hier 'so' erstmal nicht weiter, da die entitie_id's der web-api sensoren 'natürlich' nicht den namen der 'verlorengegangenen entsprechen'... 

Da ich zu Hause noch die "alte" (nicht https) variante habe, kann ich auch nicht einfach diese bestehenden sensoren überschreiben... Ich muss mal einwenig überlegen, ob es da noch eine elegante/end-user-freundliche variante gibt. Aber dazu benötige ich etwas Zeit.

electricdelicate, 2023-09-03:
> 

perfekt, ich habe ein zusätzliches WebAPI Device hinzugefügt.
[image]

bobthebuilder4711, 2023-09-03:
> Ich habe hier einen Haufen von Helpern und darauf aufbauenden Sankey-Diagrammen, die auf den ursprünglichen Entities der v2.1-Integration
>
> ```
> sensor.senec_solar_total_generated [...]

Bau Dir Template Sensoren, deren IDs Du dann in deine Helper einbauen kannst. Dann kannst Du relativ einfach in den Template Sensoren ändern, wenn sich wieder was ändert, ohne dass Du die Helper nochmal anfassen musst.

zum Beispiel (in der configuration.yaml eintragen):
```
template:
  - sensor:
    - name: "Fester Sensor PV Erzeugung Gesamt"
      unique_id: "hier was Einzigartiges ausdenken"
      unit_of_measurement: kWh
      device_class: energy
      state_class: total_increasing
      state: >
        {{ float(states('sensor.senec_solar_total_generated')) }}
```

In der letzten Zeile kannst Du dann den jeweiligen Sensor eintragen, den Du eigentlich nutzen willst.

Home Assistant baut die entity ID aus dem name zusammen. In dem Fall wäre das dann sensor.fester_sensor_pv_erzeugung_gesamt.

marq24, 2023-09-03:
> [edit (bob): Die 'unique_id' und die 'entity_id' sind zwei verschiedene paar Schuhe. Die unique_id braucht Home Assistant intern zur eindeutigen Identifizierung, selbst wenn man eine 'entity_id' angibt. Meine Templates zum Beispiel haben da UUIDs drin stehen (zB: "49da82ca-4d3b-4db3-ac86-e8c3b0dafd50"). Gibt es keine 'unique_id', dann kann es sehr gut sein, dass Home Assistant meckert, dass man den sensor für viele Dinge nicht benutzen kann.]

wieder was über HA gelernt! - danke Bob!

bobthebuilder4711, 2023-09-03:
Hatte halt selber schon das Vergnügen, stundenlang dumm auf den Bildschirm zu starren, als ich meine ESPEasy Steckdosen per MQTT an HA anbinden wollte und dann in der GUI nicht richtig benutzen konnte. Da stößt man dann schnell auf diese Seite, die es gut erklärt: https://www.home-assistant.io/faq/unique_id/

blackfed, 2023-09-05:
Hi,

Just started using today this integration, I have a V2 running firmware 825, so I set it up using the WebAPI option by logging in. As soon as it creates the device in my HASS I can see data on the sensors, but then it doesn't get any other update, and after a couple of minutes they all go "not available". Is there something I'm doing wrong?

Thanks

egonkab, 2023-09-05:
So wieder ein bisschen Zeit für den JA

> updated initial post -> added a solution alternative: **mein-senec.de WebPortal Integration Entry**

Wie hoch ist die Abfragerate?

marq24, 2023-09-05:
> Wie hoch ist die Abfragerate?

5 minutes

bobthebuilder4711, 2023-09-05:
For my V2.1 the portal updates ever 5 minutes. But I've been told that you can poll the portal much quicker if you have a V4. 30 seconds seems to still be okay. But I can't say if the portal updates even quicker than that.

@egonkab you should definitely use both the webportal for the kWh data and the normal local connection for the kW data since that updates much faster. I'm currently polling the local connection every 10 seconds without any problems.

jc2308, 2023-09-06:
Hi, today i ran into this issue. So far i got the things to work now, but how can i add the WepApi? i can´t chose it by adding a new integration. There are only the 3 Senec variants.

My integration verision is 3.0.5 and beta is active.
[image]

marq24, 2023-09-06:
well the screenshot is from 3.0.5 - so what ever you did (when you tried to install 3.0.6-test-xx) did not work - might have forgotten to restart HA? Tried to use SHIFT+Reload in the Browser?

jc2308, 2023-09-06:
Oh, sry. i didn´t see that there is a new Version. The latest version in HACS is 3.0.5 for me. I´ll have a look at the new version. Thanks for your Work :)

marq24, 2023-09-06:
> Oh, sry. i didn´t see that there is a new Version. The latest version in HACS is 3.0.5 for me. I´ll have a look at the new version. 

the "new build" is currently just in BETA-state... so that's probably the reason why you did not see/saw it...

sigcarunchio, 2023-09-07:
Hi marq24, i'm trying the new webapi sensors, but value doesn't match the ones of senec ios app.
They are lower than app; i have two senec modules, a master and a slave; with the old integration missing values reported total values of produced energy, battery in/out, grid in/out; actual webapi value are total?

Also, is this correct?

Battery discharged -->  sensor.senec_battery_total_discharged --> sensor.senec_wabapi_accuimport_total
Battery charged --> sensor.senec_battery_total_charged --> sensor.senec_wabapi_accuexport_total

or new sensors are inverted?
Thank you!

marq24, 2023-09-07:
> Hi marq24, i'm trying the new webapi sensors, but value doesn't match the ones of senec ios app. They are lower than app; i have two senec modules, a master and a slave; with the old integration [...]

please see the conversation #8 concerning the process of "multiple" stacked SENEC Systems... 

> Also, is this correct?
> 
> Battery discharged --> sensor.senec_battery_total_discharged --> sensor.senec_wabapi_accuimport_total Battery charged --> sensor.senec_battery_total_charged --> sensor.senec_wabapi_accuexport_total
> 
> or new sensors are inverted?

this question should go to SENEC master-brain developed the (internal)API... I fully agree with you, that (internal) naming is quite confusing... All I can say (based on my personal observation) is, that the `accuimport` fields from the webapi containing data when my home is consuming power from the battery - and the `accuexport` fields when the battery will be charged... 

So IMHO the SENEC developer took the perspective from the point of view of Main unit (and not from the Battery)...

bobthebuilder4711, 2023-09-07:
Yes, the sensors are inverted to what we're used to.

The names are like that because mein-senec.de calls them that internally. And they correspond to how import and export works **from the view of the house**.

Think about it this way:

Energy flowing from your home into the grid is: gridexport.

So energy flowing from your home into the battery is: batteryexport.

And if you need energy from the grid then that is: gridimport.

If you need energy from the battery then that is: batteryimport.

mstuettgen, 2023-09-07:
> For my V2.1 the portal updates ever 5 minutes. But I've been told that you can poll the portal much quicker if you have a V4. 30 seconds seems to still be okay. But I can't say if the portal updates even quicker than that.

I can confirm that you can poll data from mein-senec.de at least every 30 seconds if you have a V4. But I also have not tried shorter polling intervals, as 30s is absolutely fine for me. Maybe the V4 transmits its data more often "back home"? I might investigate that when I return from my vacation. 

If you wanna play around with it  yourself, run "senec_webgrabber.py" from https://github.com/mstuettgen/homeassistant-addons/blob/main/senecweb2mqtt/senec_webgrabber.py. You just need to fill in your senec credentials.

The script will print out the that it read from the website.

marq24, 2023-09-07:
> I can confirm that you can poll data from mein-senec.de at least every 30 seconds if you have a V4. But I also have not tried shorter polling intervals, as 30s is absolutely fine for me. Maybe the [...]

thanks! - I actually have already (this morning) adjusted the polling interval to 60sec if the system identifies itself as 'Home V4'... but this only takes effect, when you re-install the integration entry [since the interval is currently not part of the configuration]

Thomk-63, 2023-09-07:
HI, I am now using the WebAPI for getting the statistics values. This seems to work and I can take these WebAPI sensors for the energy dashboard. BUT: All the energy dashboard history is then gone, as this is saved under the original senec entity IDs. Is there any way to write the WebAPI values to the senec entity IDs. Ideally this would be done directly in the integration by mapping the WebAPI to the unavailable statistics entity IDs. Unfortunately, I am not a software programmer. Any idea?

bobthebuilder4711, 2023-09-07:
Seems like you can just delete the old sensors and then rename the new sensors to have the entity_ids of the old ones: https://community.home-assistant.io/t/migrate-lts-history-to-replacement-sensor/460115/3

marq24, 2023-09-08:
**> To combat that just leave the old entities in the energy dashboard. I have both the old and the new sensors in there and it looks like this on the day of the update. It's yellow instead of orange from there on but that doesn't really bother me. !**

exactly! - as documented here - https://github.com/marq24/ha-senec-v3/issues/11

SteffenS27, 2023-09-10:
Hi, sorry for confusing that technical discussion. Bit can someone please tell what I have to do with that beta version and so on ? Sorry bit HA is new for me

marq24, 2023-09-10:
@SteffenS27 - have you read this? https://github.com/marq24/ha-senec-v3/issues/14

SteffenS27, 2023-09-13:
@marq24 thanks that should be what I need, but I have at the moment another problem, in HACS it is not possible to install or renew a Integration, the circle is just turning, but nothing happens

marq24, 2023-09-13:
@SteffenS27  I hope it's obvious, that I can't provide HA support in general... I would do:

0. CREATE A BACKUP
1. login via ssh to get shell access
2. rename the `custom_components` folder in your config directory (e.g. to `xustom_components`
3. restart HA

this will disable all custom_components you might have installed - and you can re-enable them one by one to find the one causing the loop... [also checking the ha system logs might be already a way simpler start to find the root of your problem]...

IF it's the SENEC.Home integration - double check, if you have specified the correct credentials (WEBAPI) - If your system was http only before (and now https is required) - make sure that you remove the LaLa.cgi based integration - and reinstall it.

Cyrelion, 2023-10-02:
### @marq24,

Heute hab ich zufällig in einem der Facebook Posts gelesen dass eine neue Firmware ausgerollt wird welche die Probleme mit der aktuellen NPU-Error etc. lösen soll.

Dabei gibt es ebenfalls ein Absatz über unser Vars.html Problem, welches aber leider eher weniger zur Freude einlädt.

Anbei ein Screenshot der Nachricht aus der ich gerne mal Zitire:

"Bisher konnten Endkunden über /display.html bestimmte Werte des Speichers auslesen, diese Werte finden sich in der App bzw. in mein-senec.de. Im Zuge der Optimierung des Speicherverbrauchs der Firmware setzen wir diese Funktion nicht fort.
Dies gilt auch für den direkten Zugriff auf Entwicklungsvarieablen der Systeme, die zuvor durch Endkunden und Fachpartner selbst ausgelesen werden konnten (/Vars.html). Im Zuge einer allgemeinen Erhöhung der Sicherheitsstandards der Systeme wird der Zugriff auf diese Entwicklungsinformationen auf SENEC selbst beschränkt."

Heißt im Klartext: Weil die Mainboards anscheinend zu wenig RAM haben und vermutlich die Firmware jetzt durch die SSL Implementierung mehr Speicher benötigt haben sie solche wichtigen Informationen wie Gesamtverbrauch etc. einfach wegoptimiert? Das kann doch nicht deren Ernst sein.

[image]

Evtl. kannst du ja zumindest die Funktionen der WebAPI und des Speicherzugriffs dann zusammenfassen und die WebAPI Daten dann in die fehlenden Speicherentities packen?

Gruß,
Andree

bobthebuilder4711, 2023-10-02:
> Evtl. kannst du ja zumindest die Funktionen der WebAPI und des Speicherzugriffs dann zusammenfassen und die WebAPI Daten dann in die fehlenden Speicherentities packen?

Mit einem V4 geht ja sowieso nur noch die Abfrage über mein-senec.de oder die App. Die reine WebAPI hat also auf jeden Fall ihre ganz eigenständige Daseinsberechtigung.

Und wenn mal eins der beiden nicht geht, ist auch die Fehlersuche einfacher.

marq24, 2023-10-02:
> Evtl. kannst du ja zumindest die Funktionen der WebAPI und des Speicherzugriffs dann zusammenfassen und die WebAPI Daten dann in die fehlenden Speicherentities packen?

In der Ankündigung von SENEC steht, das die Vars.html und die display.html abgekündigt wird [was natürlich an sich schon erbährmlich ist - aber so ist es eben - beides sind aber Funktionen, die ich persönlich bislang nicht verwendet habe]... Wie es mit den STATISTICS-Daten aussieht, werden wir dann alle merken, wenn das Update tatsächlich mal ausgerollt worden ist... 

Und vor allem, was noch alles so zusätzlich mt dem Arsch umgeworfen wurde - bis dahin will ich aber erstmal abwarten... 

Die WEB-API Felder aber in die Local-lala.cgi Implementierung zu integrieren halte ich im Augenblick aber noch nicht so zielführend - schon allein wegen der unterschiedlichen "polling-intervallen" und der Tatsache, das das eine mit dem anderen nicht wirklich viel gemeinsam haben... 

>... im Zuge einer allgemeinen Erhöhung der Sicherheitsstandards der Systeme...
Humor haben'se wenigstens!

loeffelpan, 2023-10-02:
Hab auch gerade die Meldung in der App bekommen.
Klingt nach als kämen die Statistikdaten wieder. Aber mal sehen.

Cyrelion, 2023-10-02:
Dann hoffen wir mal das beste das die Statistiken in die lala.cgi wieder kommen. Ansonsten hab ich jetzt alles erstmal auf die WebAPI umgestellt in meinen Dasbboads und für die Automatisierungen ob und wieviel Überschuss gerade da ist habe ich ja gott-sei-dank mit Discovergy direkten Zugriff auf die Daten des Stromzählers sodass ich hier sogar genauer und direkter als mit Senec arbeiten kann.

bobthebuilder4711, 2023-10-03:
> direkten Zugriff auf die Daten des Stromzählers

Dann noch einen Shelly 3EM an das Stromkabel vom Senec Speicher klemmen und Du brauchst die Daten theoretisch gar nicht mehr vom Speicher selber abzurufen.

marq24, 2023-10-03:
> ... direkten Zugriff auf die Daten des Stromzählers ...

Seit ein paar Wochen habe ich einen Tibber Pulse im Keller - und heute habe ich mich mal hingesetzt und damit auseinandergesetzt, wie ich direkt an die Daten komme...

-> https://github.com/marq24/ha-tibber-pulse-local

[8 more replies]