marq24/ha-senec-v3 #5: Energiedashboard

SimonFuVo, 2023-09-04:
Hallo, bei mir werden vom Energiedashboard die einzelnen Entitäten nicht akzeptier. Muss ich hier selber etwas anpassen?

Ansonsten vielen Dank für die bereitstellung der neuen Integration.

[image]

marq24, 2023-09-04:
That's the name of the sensor when using the V2.x/V3 Variant (polling lala.cgi) in your LAN) - when you use V4/WebAPI, then the sensor name is different:  `sensor.senec_webapi_gridexport_total` (when you use the default 'Name')

SimonFuVo, 2023-09-04:
Name of sensor from v3 localy which not work is
sensor.senec_solar_generated_power

marq24, 2023-09-04:
sorry - updated my initial reply

SventeHof, 2023-09-04:
same Problem.. the following entitys are not queried and displayed:

- sensor.senec_grid_total_import
- sensor.senec_grid_total_export
- sensor.senec_solar_total_generated
- sensor.senec_battery_total_charged
- sensor.senec_battery_total_discharged

marq24, 2023-09-04:
I have created for you gentlemen here https://github.com/marq24/ha-senec-v3/issues/4#issue-1873971875 a table showing old (lala.cgi incl. STATISTICS) and new sensor names (web-api based)

marq24, 2023-09-06:
I hope with this additional info/issue #11 we can consider this as solved?

closed 2023-09-06 by marq24

SittingDuck52, 2023-10-08:
Additional Info:
[](https://www.home-assistant.io/docs/energy/electricity-grid/)
If you are unable to select your energy sensor in the grid consumption drop-down, make sure that its value is being recorded in the Recorder settings.