marq24/ha-senec-v3 #7: Webapi stopped pulling Data after 401 from Senecs Website

neubi4, 2023-09-04:
Hey @marq24 i an running 3.0.6-test-04 since yesterday, today at 12:33 the integration stopped pullling data, and i got this error message in log:

`2023-09-04 12:33:38.375 ERROR (MainThread) [custom_components.senec] Error requesting senec data: 401, message='', url=URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=endkunde-portal&scope=endkunde&state=randomid&redirect_uri=https://mein-senec.de/endkunde/login/oauth2/code/endkunde-portal')
`

After an reload of senec_WEBAPI it is now working fine. Maybe we could do limited number of retries here if we get an error from senecs website?

marq24, 2023-09-04:
Thanks for sharing this... 

This 401 during login is caused by a (IMHO) bug in the cookie implementation of aiohttp client - I have created a local patch... but this implies, that no other code will "reset" the used CookieJar for the HTTP Session(s)...

Looks like that this happens at your system (that some code is resetting the client-session - and my patched CookieJar is not used anylonger)... We will see, if I manage to "check" what JAR is currently in use (and "force" mine)... OR if it's faster, that aiohttp team will accept my PR (https://github.com/aio-libs/aiohttp/pull/7577) that I just have submit...

marq24, 2023-09-04:
@neubi4 can you please install - https://github.com/marq24/ha-senec-v3/releases/tag/3.0.6-test-05 this one will force the SenecCookieJar when ever the code accessing the web-api... I am courious, if you will see at a certain point of time in the log:

`WARNING CookieJar is not of type MySenecCookie JAR any longer... forcing CookieJAR update`

scottgoo, 2023-09-05:
I have installed the test-05. To get test-07 do I have to deinstall the integration completely or is there a way of updating it? Like to know if there still is the 401 in test-07.

marq24, 2023-09-05:
update works as "usual" via HACs - you man choose "install it again" ... and then select the new build

scottgoo, 2023-09-06:
It worked a little longer with the 07 version so the energy dashboard started showing some numbers. But then I got the 401 problem again:

Logger: custom_components.senec
Source: helpers/update_coordinator.py:229 
Integration: SENEC.Home ([documentation](https://github.com/marq24/ha-senec-v3/blob/master/README.md), [issues](https://github.com/marq24/ha-senec-v3/issues)) 
First occurred: 06:48:25 (2 occurrences) 
Last logged: 06:54:14

Error requesting senec data: 401, message='', url=URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=endkunde-portal&scope=endkunde&state=XXX&redirect_uri=https://mein-senec.de/endkunde/login/oauth2/code/endkunde-portal')
Error requesting senec data: 401, message='', url=URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=endkunde-portal&scope=endkunde&state=YYY&redirect_uri=https://mein-senec.de/endkunde/login/oauth2/code/endkunde-portal')

marq24, 2023-09-06:
... and did you saw the mentioned warning in ha log? ['CookieJar is not of type MySenecCookie JAR any longer... forcing CookieJAR update']... 

the "good" thing is, that since yesterday my SENEC was also updated to the STATISTICS missing version - so I need to use the WEB integration too :-D - My one is running since 23:00 o'clock yesterday so we will see, how this going to develop... I guess with a restart of HA (or just the integration - if that is possible?!) solves the current situation... but obviously the "plan" was something different - should try to auto-re-login if there is any response != 200...

scottgoo, 2023-09-06:
At first thank you very much for your quick replies and your commitment in hellping us senec guy. Unfortunately I am not so deep into HA to answer every question as quick as you can. ;-)

I tried to find the cookie warning but I couldn't. Neither under System/Log (where I find the 401 warning) nor in file system under /config/home-assistant.log Are those spots the right ones to look for die cookie warning?

When I restart the integration manually things are running for a while but then they fail. Interesting: I restarted the integration  13 minutes ago and it's still fine. That is more I ever had since installing the 07 version. Keep your fingers crossed! :-)

marq24, 2023-09-06:
... IF they are present, then you find them here... https://[YOUR_HA_ROOT]/config/logs...

But when you say, that 13minutes are already a long period, then I have to check something (in general) - since I would had expect, that this problem would occur after 24h (or later)

marq24, 2023-09-06:
> Short update: System is running for nearly an hour now. Don‘t know why it didn‘t do this before. I let you know tomorrow morning if it is still alive. And investigate my log files a little more… Thank you!!!

I knock on wood...  ;-)

marq24, 2023-09-07:
Good Morning - yesterday evening I released a new (beta) version https://github.com/marq24/ha-senec-v3/releases/tag/3.0.6-test-08 - this include code that should handle possible 401's... this might not be required - but who knows... 

Here locally my system had not any 401's so far - BUT since I am obviously test the released before they go live this does not mean that much (probably my system never run's 24h without a restart)

scottgoo, 2023-09-07:
Systems are still up and running. :-) No cookie mentioning anywhere but the 24 hour line is not crossed yet. Thank you for the next release. I will install it the next days.

scottgoo, 2023-09-07:
Version 07 still working as hell more than 24 hours after restart. Nevertheless I will switch to 08 on sunday. Kudos for @marq24 again! 😊

marq24, 2023-09-09:
@neubi4  @scottgoo con we consider this as solved?

scottgoo, 2023-09-09:
I think so. Thank you very much!

neubi4, 2023-09-09:
I was on 3.0.6-test-07 since 5 days and had no issues anymore, so i think this can be closed.

closed 2023-09-09 by neubi4