marq24/ha-senec-v3 #8: Supporting Master/Slave in Homeassistant webapi integration

melotti892, 2023-09-05:
here we are
now as found before the integration recognise the master battery, but not the slave one
is there a way to get all the data and then use them as the user prefer?
or best combine it all in an unique output for every info?

marq24, 2023-09-05:
... this will become quite a challenge in the next couple of days/weeks to get this solved correctly - since i don't know anybody (yet) in my area with such a setup - so I have to to develop blind...

First of all my assumption was (but I guess this is totally wrong), that the MASTER would show you ALL collected data (from all subsystems)... But looks like this is not the case?

Just as explanation - here in my local setup I have an additional Inverter (from SMA) installed - and the power that's generated by this additional Inverter is INCLUDED in the data my Senec device is reporting (in both implementations - in WEB and in LAN)... So I had the HOPE, that the master is collecting all the data of all "additional" devices...

But from from your input this sounds like, that's not the case, and that each `anlage` is reporting it's own/separate values?

Could you please be so kind and share the output of the following requests:
https://mein-senec.de/endkunde/api/status/getstatusoverview.php?anlageNummer=0
https://mein-senec.de/endkunde/api/status/getstatusoverview.php?anlageNummer=1

https://mein-senec.de/endkunde/api/status/getstatus.php?type=powergenerated&period=all&anlageNummer=0
https://mein-senec.de/endkunde/api/status/getstatus.php?type=powergenerated&period=all&anlageNummer=1

https://mein-senec.de/endkunde/api/status/getstatus.php?type=gridexport&period=all&anlageNummer=0
https://mein-senec.de/endkunde/api/status/getstatus.php?type=gridexport&period=all&anlageNummer=1

https://mein-senec.de/endkunde/api/status/getstatus.php?type=consumption&period=all&anlageNummer=0
https://mein-senec.de/endkunde/api/status/getstatus.php?type=consumption&period=all&anlageNummer=1

I am also asking for this output in order to understand your kW vs W Unit issue you mentioned - IMHO the data that is reported from the web is in kW - the json response include tha value(s) in the `fullkwh` field.

melotti892, 2023-09-05:
unfortunately ( in my case) i can only confirm that your assumption were wrong
i have the data from both the battery divided, even in the app, if i need to see the slave i have like to select it manually
and the metrics tabs shows all the funny graphic but only from 1 battery at a time
now what you asked:
1) {"wartungsplan":{"possibleMaintenanceTypes":[],"applicable":false,"maintenanceDueSoon":false,"maintenanceOverdue":false,"minorMaintenancePossible":false},"suppressedNotificationIds":[],"steuereinheitState":"AKKU_VOLL","wartungNotwendig":false,"state":13,"firmwareVersion":825,"gridexport":{"today":2.7667236328125,"now":2.591309},"gridimport":{"today":0.072998046875,"now":0.000000},"powergenerated":{"today":3.88134765625,"now":2.742188},"consumption":{"today":1.4736328125,"now":0.158203},"accuexport":{"today":2.470947265625,"now":0.000000},"accuimport":{"today":1.328369140625,"now":0.002930},"acculevel":{"today":83.25547790527344,"now":100.0},"machine":"MCU","lastupdated":1693906121}
2) {"wartungsplan":{"possibleMaintenanceTypes":[],"applicable":false,"maintenanceDueSoon":false,"maintenanceOverdue":false,"minorMaintenancePossible":false},"suppressedNotificationIds":[],"steuereinheitState":"AKKU_VOLL","wartungNotwendig":false,"state":13,"firmwareVersion":825,"gridexport":{"today":5.53564453125,"now":5.194599},"gridimport":{"today":0.145263671875,"now":0.000000},"powergenerated":{"today":12.3818359375,"now":5.909988},"consumption":{"today":4.7255859375,"now":0.724149},"accuexport":{"today":4.533203125,"now":0.000000},"accuimport":{"today":2.26708984375,"now":0.000000},"acculevel":{"today":84.90243530273438,"now":100.0},"machine":"MCU","lastupdated":1693906060}
3) {"fullkwh":5264.5751953125,"lastupdated":1693864800,"val":[[1640991600000,1573.658691406249999],[1672527600000,3690.9165039062500]]}
4) {"fullkwh":12152.0,"lastupdated":1693864800,"val":[[1640991600000,3642.149658203125003],[1672527600000,8509.850830078125]]}
5) {"fullkwh":1723.7591552734375,"lastupdated":1693864800,"val":[[1640991600000,101.59199523925781258],[1672527600000,1622.16716003417968875]]}
6) {"fullkwh":3388.733642578125,"lastupdated":1693864800,"val":[[1640991600000,162.626215934753416410],[1672527600000,3226.1073760986328135]]}
7) {"fullkwh":5155.1318359375,"lastupdated":1693864800,"val":[[1640991600000,2198.7570800781249995],[1672527600000,2956.374755859375]]}
8) {"fullkwh":11106.095703125,"lastupdated":1693864800,"val":[[1640991600000,4575.944986343383788],[1672527600000,6530.15039062500]]}

marq24, 2023-09-07:
... could you please share your expectation?

Would it be cool, if the Integration would report in your scenario only a single accumulated value? Or would you prefer that each system have it's own sensor's that then deliver the values independent from each other?

I am asking since the one could be implemented quite easily (at least that's my current expectation) - and the other one would mean some sort of redesign of the complete interface...

And obviously the answer "I would like to have both" is understandable - but once you have single values you can create the sum by yourself ;-)

melotti892, 2023-09-07:
Thanks for Asking
I know how to merge things together With the helper, so i think it’s good (even for you having less to bother) to have the data of the two battery delivered separately.

marq24, 2023-09-07:
> Thanks for Asking I know how to merge things together With the helper, so i think it’s good (even for you having less to bother) to have the data of the two battery delivered separately.

*sigh* that was not the answer I was hoping for... ;-) Separating the devices would mean to provide additional sensor entities (per additional device)... and blow up the interface - I have no clue how many devices actually can be stacked (I assume there is a limit of 4 - but who ever knows)... 

But I guess: Who have asked - might get answered (he not wanted to hear)... 

When the next release (beta) would include code, that create the SUM-UP of all fields (except battery level in %) - would you use it - or would you stick to the current version? (not promising, that I am develop a sensor-separated-version in the next couple of weeks)...

melotti892, 2023-09-07:
In reality i would prefer to have all merged together
I answered to let you work less like “i would do the rest by myself” 😆 
Even the button to let the battery Charge manually (1 button, for all the battery)
I would use it and be your beta-helper :)

marq24, 2023-09-07:
then give https://github.com/marq24/ha-senec-v3/tree/3.0.6-test-09 a try ;-)

sigcarunchio, 2023-09-07:
Hi marq24, hi melotti892, I have also a master and slave senec v3
For me, in the ios app I see (I think!!!) total value, melotti892 where you can select master or slave unit?
Until recent firmeare update, I thought that present missing values were total value of both master and slave unit, and they matched ios app values.
Now webapi values no longer match app values...

melotti892, 2023-09-07:
Hi @sigcarunchio 
I’m not using the app since i use HA
Now that i checked for you How to switch from battery to battery it seems that it’s gone this option in one auto update
I know for shure (for me) the app tells me only the master value Because in the little panel/screen of the battery (master and slave one) the value that they say aren’t merged in the app, and this confirm the app (for me) says only the master one

marq24, 2023-09-07:
@sigcarunchio you might be able to dig a little bit yourself by sending the URL's posted here https://github.com/marq24/ha-senec-v3/issues/8#issuecomment-1706208446 to the webapi - and check if the values reported in the `anlageNummer=0` requests are included in the `anlageNummer=1` values - or if they are separated by slave=0 and matser=1 ?!

melotti892, 2023-09-07:
@marq24 i uninstalled the previous version
Restarted ha
Downloaded the last beta
Restarted
Configured it
Now it says “configuration not succeded, it will retry later”
And every 30/60sec it says “initializing” But stays “configuration not succeded, it will retry later”

sigcarunchio, 2023-09-07:
The guy who install my system told me that on the slave battery lcd display shows itself, while the lcd master display show total values from both battery....
now i'm away, i will check next week when i return home if data on master lcd display match total value of ios app, or it's the sum of the two

sigcarunchio, 2023-09-07:
> you might be able to dig a little bit yourself by sending the URL's posted here https://github.com/marq24/ha-senec-v3/issues/8#issuecomment-1706208446 to the webapi - and check if the values [...]

Will try this afternoon

marq24, 2023-09-07:
@melotti892 shame on me - then give https://github.com/marq24/ha-senec-v3/tree/3.0.6-test-10 another try ;-)

melotti892, 2023-09-07:
@sigcarunchio 
i didn't thought about this, that the value are already merged in the master, i will ask by email to be shure, thank you

melotti892, 2023-09-07:
@marq24 i think you made it!
now all the sensor says the data summed together
i checked all and yes,
just the battery %  says just the master one but amen
great job

marq24, 2023-09-07:
looking again closer at the result of the URL's above, I do believe, that the statement from @sigcarunchio is valid (and also my initial assumption), that the MASTER values include the values of all slaves...

| field | slave (0) | master (1) |
| --- | --- | --- |
| gridexport | 2.766 | 5.535 |
| gridimport | 0.0729 | 0.1452 |
| powergenerated | 3.881 | 12.381 |
| accuexport | 2.470 | 4.533 |
| accuimport | 1.3283 | 2.2670 |
| powergen (full kwh) | 5264.57 | 12152.0 |
| gridexport (full kwh) | 1723.75 | 3388.73 |
| consumption (full kwh) | 5155.13 | 11106.09 |

so if you ask me, the master values seams to include the values of the slave... so creating the sum of both does not seams very reasonable to me...

melotti892, 2023-09-07:
@marq24 it seems legit like you both said, i've sent a clarification email to the installation support that placed them in my home, at this point i can only say that's better to return at the point of reading only the value from the master one and keep just it

marq24, 2023-09-07:
you have the option to stick with test-v10 if you like to keep the "sum-up" version... for all other's I have reverted the creation of master and slave and only return the master values (so more or less it's the same status then test-v09)

melotti892, 2023-09-07:
@marq24 at the end you "had" made it, now confirmed 😆 
effectively if i'm not wrong, times ago i asked something to the senec direct support and they replied me that every battery pack can send 3kw at a time, so reading 5kw from the master is, following this, 2,5+2,5 of each battery

sigcarunchio, 2023-09-07:
Just update to test-v11, still have problems about different value between webapi, home assistant energy panel and senec ios app.

[image]
Here, from webapi sensor, i have today a production of 36,23 kWh

[image]
Here hass energy panel

[image]
Senec IOS app

sigcarunchio, 2023-09-07:
Boom!!!
20min after I update to test-v11, value perfectly match!!!
Still monitoring but confident about good results!

marq24, 2023-09-07:
> Boom!!! 20min after I update to test-v11, value perfectly match!!! Still monitoring but confident about good results!

good that sometimes things get solved by themselves (while I am riding the bike outside)... 

> How often the webapi integration ask data? It's the same about ios app? I think webapi is sometimes a little behind ios app

every 5 minutes - I did not want to kill SENEC Servers... and started with this conservative update interval...

melotti892, 2023-09-07:
@marq24 
maybe you can think to add a "timer" in seconds to refresh the servers
with at minimum like 60 seconds?
don't know, it's an idea
because like i would use that parameter for charging the car, and increase (or decrease) the power depending of how near is to the cap (6Kw)
and 5 minute is quite long for have a precise automation

marq24, 2023-09-07:
IMHO it does not make sense to spam mein-senec.de with requests, when we (currently) do not know in which frequency the device are uploading the data to SENEC...

If you want to do more "real time" automation - I strongly recommend to use the LAN access - where you can use an update interval <= 5sec

melotti892, 2023-09-08:
answer email received, the master one is the one who summ all the data of the 2 battery pack

sigcarunchio, 2023-09-08:
> If you want to do more "real time" automation - I strongly recommend to use additional the LAN access - where you can use an update interval <= 5sec

Yes, I think that statistics data in kwh from webapi are ok every 5 min.
melotti 892 For precise automation you can use power value in kw from direct integration.

melotti892, 2023-09-08:
@sigcarunchio 
i re-added the direct integration (senec.home v3 direct integration) and now it shows everything, like before the update that stopped it
@marq24 you made something?

melotti892, 2023-09-08:
@marq24 i have to study the lan access you said, cause i would like to have the definitive solution, but i'm afraid do made some kind of fatal mistake, you know

marq24, 2023-09-09:
@melotti892 - I hope you can understand, that I can't provide you with any additional information "how to connect your Senec to your LAN"... IMHO your Senec is communicating with the Internet - therefore it's somehow connected to your local router - therefor you should find it in your LAN (via the user interface of your Internet Router)... 

If your Solar installateur have build completely own infrastructure in your house (with own internet provider), so to say a second/parallel network then I would get in contact with him - since you should have access to the LAN as well... 

The only additional hint I am able to provide is the "official" installation documentation I have found (only in German) https://sollis.de/wp-content/uploads/TD140-057.10_SENEC.Home_V3_Installationsanleitung_1.0.pdf - on page 20 is described, when the ETH connector is located... and this connector is the one that's connected to my LAN here at my home.

closed 2023-09-22 by marq24

mariobon123, 2024-02-23:
> answer email received, the master one is the one who summ all the data of the 2 battery pack

Hello, I know this thread is a bit old and it was answered, but I'm just looking for a small clarification regarding the merging issue: the percentage shown on the app and on the battery displays is also summed and averaged by the Master? i.e. is the master shows 50% and the slave 88%, is the 50% shown by the master an average between the two?

marq24, 2024-02-23:
> Hello, I know this thread is a bit old and it was answered, but I'm just looking for a small clarification regarding the merging issue: the percentage shown on the app and on the battery displays is [...]

at least I wanted to reply - I just can guess you have to check that by yourself - looking at the battery level of each of the devices via the local web frontend (local-ip) - and then compare the two values with the number displayed by the SENEC-App...

IMHO just adding the both percent values and divide by 2 is not the correct approach - this would be only valid of both systems have identical capacities... [but again - since I just have "one" unit in my basement I can't provide you with an answer...