marq24/ha-senec-v3 #13: Wrong unit in senec_house_power

MrIcemanLE, 2023-09-08:
Hi!

I noticed that the entity sensor.senec_house_power has the wrong unit. In my case I get around 800kW. Thats way to much. I think it has to be 800W. Unfortunately i changed the entity_id back to the old ones so all my dashboards continue working. The entity shows the current power the "house" is using.

[image]

marq24, 2023-09-09:
I can confirm, that `sensor.senec_house_power` should/is WATT
[image]

also code wise it's configured as WATT
https://github.com/marq24/ha-senec-v3/blob/5aff26b31d3b82c43295583993add08510c671a1/custom_components/senec/const.py#L272

But it could be, that in your installation something is mixed up with the additional WEB-API Integration - where "House Power" is measured in KILO_WATTS ?! see `sensor.senec_webapi_consumption_now` 
[image]
But this WebAPI Sensor should have a different Entity Name/ID?!

io-debug, 2023-09-15:
Hi, I can also confirm that the sensor is configured correctly (in the code as well as in Home Assistant) in WATT.
@MrIcemanLE do you still experience this issue?

MrIcemanLE, 2023-09-15:
So maybe something mixed up due to the "old" integration that stoped working some days ago. As I said I reused the old entitiy IDs to keep the energy dashboard working.

In my case both entities are showing Kilowatts. Any Idea how I can fix that? Can't change the unit manually. Maybe I should try to reinstall the integration

[image]

marq24, 2023-09-15:
> As I said I reused the old entitiy IDs to keep the energy dashboard working.

I just can guess, that this is the root of the problem... currently you can not reuse the original sensor ids... For the EnergyDashboard just add the new ones... and do not delete the (currently) not feeded ones... This will give you the best possible backward compatibility... We all dont know if the statistic data will return in the lala.cgi...

bobthebuilder4711, 2023-09-16:
> Any Idea how I can fix that?

I would try opening the settings of the sensor and simply changing the unit. Or does that lead to home assistant converting the data behind the scenes?

closed 2023-10-01 by marq24