marq24/ha-senec-v3 #18: Update time seems not to work as set

Thomk-63, 2023-09-11:
Hi, I have a Senec V3 and using this integration. So far this is working very well with the latest firmware update 825. Today, I wanted to shorten the update time from 30s to 15s. So I have reconfigured the device and reloaded it, but no change in the update time of the Senec V3 entities.
Do I do something wrong or why is a shorter update time not effective?

marq24, 2023-09-11:
... probably you need to restart your complete HA instance so that the new update interval kick's in...

Thomk-63, 2023-09-11:
I tried this already, but it did not help!

marq24, 2023-09-12:
... after you have restarted ha - and open the configuration dialog (of the Integration again) - do you see the NEW or the OLD value then? - So does it display 15 or 30 as update interval?

marq24, 2023-09-12:
Ok - looks like a but - that was working before - but in the current build this functionality seems to be gone... Till I am able to provide a fix, as workaround, you can remove the integration entry - and (without restarting) re-add it with the same display name and the changed update interval.

marq24, 2023-09-12:
can you please check https://github.com/marq24/ha-senec-v3/releases/tag/3.0.6-test-rc02 and close this issue, when you agree that this issue is solved - THANKS for reporting it!

Thomk-63, 2023-09-12:
Your bug fix is working perfectly! Thank you!

closed 2023-09-12 by Thomk-63