marq24/ha-senec-v3 #20: IP change not working

janikthali, 2023-09-19:
Recently I redid my home network and the ip of the senec contoller changed.
I thought it would be that easy to just change the ip of the device in the homeassistant gui
but in the logs I can still see the old ip being used even after reboot.

marq24, 2023-09-19:
Hi @janikthali,

as quick solution for your problem (and assuming you have not manually (via the GUI) modified any of the Sensors, you simply can remove the current Integration and RE-add it with the new IP... 

I am currently super bussy@work so I will not have the appropriate time to fix & test the required change - I hope the workaround is fine for now - but of course this needs to be fixed!

janikthali, 2023-09-19:
Hey, yes that's the workaround I did. No problem just wanted to share so it can get fixed.
Keep up the great work!

marq24, 2023-10-08:
I have to admit, that's not "easy" to test this for me - code wise it looks like that it "should be fixed" (famous developer words)...

marq24, 2024-03-10:
to my best knowledge this 'should be resolved'... Please reopen, if the problem still exists

closed 2024-03-10 by marq24