marq24/ha-senec-v3 #21: HACS does not find Repo/Integration

h4nnes, 2023-09-19:
Hi,
any idea why I can’t find the integration in hacs, when I search for „Senec“?

The custom repo was added:

[image]

Thanks!
Hannes

PS: Manual adding was possible, but I would like to use the repo to get updates etc.

marq24, 2023-09-20:
well - more or less I am clueless here - when you added he repro in HACS and then installed the custom_integration from this repro in HACS, then the SENEC.Home "should" appear under Devices and Services... But if after the installation in HACS the custom integration would not appear in HACS itself, then I am clueless what would be the root cause... 

[image]

h4nnes, 2023-09-20:
I can’t download/find the repo in hacs. But it’s saying, the custom repo exists. Really strange

marq24, 2023-09-20:
well - so when you have the repo added, then (adding another one) should/will look like this:
[image]

and when you then press the blue (at the bottom) `Explore & Download Repositories` button and then type 'senec' in the search bar you should get the two possible HACS-Integrations [the 'original' version and "this" one here...]
[image]

After you have selected `SENEC.Home V2.x/V3/V4 Systems` from this list, you need to press again the blue `download` button at the bottom/right... to finally install the custom integration in HA...

marq24, 2023-09-20:
... but could it be, that you are using a "beta" version of HACS? - since the error message reports, that this repo is already part of the default store... 

But this PR 'https://github.com/hacs/default/pull/2035' is not merged yet - so I don't understand how ha-senec-v3 could be already in your hacs installation?

h4nnes, 2023-09-20:
I'm using HACS version "1.32.1". But yes, looks different to your version. Which version are you using?
[image]

When I search for Senec, I only get the repo from mchwalisz.

Really strange

h4nnes, 2023-09-20:
So, I just reinstalled hacs and it's working now. Thanks a lot!

closed 2023-09-20 by h4nnes