marq24/ha-senec-v3 #22: Spare-Capacity will not be added on "fresh" HA installation

marq24, 2023-09-21:
Having the integration installed (the custom_component), but not configured any integration - restart HA - then add the WEB-API - for what ever reason only 13 Entities will be shown in the GUI - the spare_capacity entity is not showing up - even if the EntityRegistry event-logger is showing that the entity will be created!

When you restart HA, then the 14'th Entity ill show up and works as expected...

When you remove the WEB-API - and do not restart HA - and re-add it again - the 14'th entity is also instant present... When you remove the integration - restart HA - and then re-add it again - there are only 13 entities!

For local debugging I use here a docker HA-installation running in WSL on my Windows Notebook - this is the easiest way for me to test the code that I develop in my IDE.

io-debug, 2023-09-21:
I could reproduce the issue.
In my case, when I see 13 Entities the 14th Entity appears after some time. No reload of the integration or restart of Home Assistant is necessary. 
For testing purposes I've added several additional Logger-Calls, but could not find any hint, why the Entity doesn't apper immediately. It seems as if the Number-Plattform as well as the Number-Entity is set up as expected.

Currently I don't have any thesis, where the problem ist and have to analyse this further.

marq24, 2023-09-21:
indeed - if I wait 5min+ the missing entity does appear automatically ... this is really  strange - HA logic ?!

closed 2023-09-21 by marq24

io-debug, 2023-09-22:
As you see, sometimes I struggle as well to understand the behaviour of Home Assistant....
I will continue thinking, what the reson could be.