marq24/ha-senec-v3 #25: Second wallbox Data?

toastbroat13, 2023-09-23:
Hi and first of all a big thanks to you and your great work on the SENEC integration! :-)

I am running two wallboxes on my senec v3 and your wallbox sensor provides data for wallbox01 (main wallbox) but does not include the power of wallbox02. 

Is there any way to get another sensor for the second wallbox? how could I test the POST-request?

Many thanks in advance and kind regards,
Matthias

marq24, 2023-09-23:
first of all I assume you are talking about the local-polling version (not the WebAPI) - right? For the local polling and support for a second wallbox I would need indeed your help...

Have you ever used a tool called Postman? - Or any other CMD-Line tool (like Curl) that is able to send a POST request your your SENEC?

POST https://[my-ip]/lala.cgi HTTP/1.1
content-type: application/json

and as RAW-CONTENT:
```
{
"DEBUG":{}
}
```
I just can guess, that you have then a second `"WALLBOX???":{},` entry - or the data inside the WALLBOX section will look different... as soon as I have some sample data, it should be some sort of easy to add support to that... 

And beside that - it's obvious, that I would have the need for a sponsor for my first wallbox ;-)

marq24, 2023-09-23:
never mind - please forget my initial response - I have a good reason to believe where the data for the second wallbox could be found... please be patient a couple of days (and make sure you have enabled the BETA option in HACS for this Integration) - in a couple of days you will find additional `Wallbox II` Sensors (that need to be enabled manually)

marq24, 2023-09-24:
@toastbroat13 I think with this https://github.com/marq24/ha-senec-v3/releases/tag/3.0.7-test-001 we can close this - right?

toastbroat13, 2023-09-24:
@marq24 Wow... you are fast as hell in answering and finding solutions. 
Many thanks and waiting for the next Beta. 🥇

closed 2023-09-24 by toastbroat13