marq24/ha-senec-v3 #27: No module named 'constants'

MrIcemanLE, 2023-09-25:
Hi,

After updating to the latest preRelease Version I get the following error:

```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant.setup
Source: custom_components/senec/pysenec_ha/__init__.py:26
Integration: SENEC.Home (documentation, issues)
First occurred: 11:06:19 (1 occurrences)
Last logged: 11:06:19

Setup failed for custom integration senec: Unable to import component: No module named 'constants'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/senec/__init__.py", line 19, in <module>
    from custom_components.senec.pysenec_ha import Senec, Inverter, MySenecWebPortal
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 26, in <module>
    from constants import (
ModuleNotFoundError: No module named 'constants'

```

What went wrong? Any Ideas?

marq24, 2023-09-25:
YUCK! - I am sorry - obviously I missed a essential TEST-STEP this morning (was probably too early? - or need more coffee)...

In any case please rollback to test-001 - the test-002 release is BORKEN (and already removed)

marq24, 2023-09-25:
THANKS for your quick reaction... I have a "fixed" build ready - that at least works on my HA installation :-) - lucky enough I just had my 3'rd coffee and realized I am still a python NOOB!

https://github.com/marq24/ha-senec-v3/releases/tag/3.0.7-test-003

closed 2023-09-25 by marq24