marq24/ha-senec-v3 #28: Senec entities can‘t be chossen in Energy Dashboard

Steffenito84, 2023-09-25:
I installed everything according to the instructions. I can now also create a new map in the dashboard and select the Senec entities. However, I have the problem that in the Energy Dashboard the Senec entities cannot be selected because they are not shown.

marq24, 2023-09-25:
I assume this is related to the change by SENEC -> please see additional documentation here #11 - if this is the solution, please be so kind and close this issue - TIA

closed 2023-10-01 by marq24

SittingDuck52, 2023-10-08:
Additional Info:
[https://www.home-assistant.io/docs/energy/electricity-grid/#troubleshooting]()
If you are unable to select your energy sensor in the grid consumption drop-down, make sure that its value is being recorded in the Recorder settings.