marq24/ha-senec-v3 #33: Master & Slave Setup for Senec.HOME V4 Systems - 4 Unit support

marq24, 2023-10-11:
@marq24 
Thank you for your amazing job, I have a request:
my setup is 4 v3 units  configured as two pair of master and a slave (yes this is so stupid, but v3 doesn't support having more than one slave per master...). This means that I have two separated systems. When I connect to the mein-senec.de (or the mobile app) I have to choose the unit I want to monitor. Crazy...
So this is my problem with the HA integration: I only see the default Master/slave pair and therefore only a half of my current statistics. 
I read that it should be possible to specify a serial number when connecting to the mein-senec.de api, do you think you can add this property in the configuration panel? This way it would be possible to create two instances of the integration, one for each pair of master/slave, and then with a helper summarize the values.
Also if you need an italian translation for all the statuses, I have them ready because, because when I tried to reverse ingeneering how to triggher some commands (like the charge and storage function you added) I found the languages translation hidden in the mein-senec.de web portal..
Thank you

_Originally posted by @mgmatteo in https://github.com/marq24/ha-senec-v3/issues/4#issuecomment-1756240706_

marq24, 2023-10-11:
reply from @bobthebuilder4711

> The webapi integration uses these links:
> 
> For the momentary values: https://mein-senec.de/endkunde/api/status/getstatusoverview.php?anlageNummer=0
> 
> And these for the total values with the appropriate type put in: https://mein-senec.de/endkunde/api/status/getstatus.php?type=powergenerated&period=today&anlageNummer=0
> 
> So if you change the 0 at the end to a 1 you should in theory get the data of your second setup.
> 
> You could check if that is true by logging into mein-senec.de and then using this link for the momentary values: https://mein-senec.de/endkunde/api/status/getstatusoverview.php?anlageNummer=1
> 
> If that works and you check that the numbers are correct then @marq24 would "only" need to add a config entry to input what number you want to poll.
> 
> Edit: It's actually already in there. The variable master_plant_number is just set to default to 0. There really only needs to be a config entry to set that variable to something else. And I have to say that I'm simply too dumb to understand how to add that. But I guess that if you know what you're doing, it's not that big of a deal to add it?

marq24, 2023-10-11:
and again from @mgmatteo

> > So if you change the 0 at the end to a 1 you should in theory get the data of your second setup.
> 
> Hi, yes I can confirm this. On the web portal I have 4 units (two masters and two slaves) And if I i use 0 to 3 i can select the different units! Curiosly on the web portal the units ar sorted in a serial number order, and not as pairs.. So in my case: slave 1, slave 2, master 2, master 1... but this is not a problem as log as the order doesn't cange :)
> 
> Unfortunately I don't any idea about how to change it too.. hopefully @marq24 knows :D

marq24, 2023-10-11:
good morning - I hope it's fine, that I have "deleted" your posts in the other issue - since this is a total new/different topic...

Before I am going to implement something I would have the need to get some "confirmation" of few things

As you might can see, there is "master & slave" support already build into the system - https://github.com/marq24/ha-senec-v3/issues/8...

Based on what @mgmatteo just have said, I expect, that
https://mein-senec.de/endkunde/api/status/getstatus.php?type=powergenerated&period=today&anlageNummer=2 is a MASTER, and the data provided by this already INCLUDING the values of the corresponding SLAVE? So anlageNummer=2 contains the data of anlageNummer=1 - and the second MASTER anlageNummer=3 includes the data of the SLAVE anlageNummer=0 - is this assumption correct?

mgmatteo, 2023-10-11:
Thank you for creating a new topic!
My consideration on top of all is that moving from V2 to V3 Senec removed the support for a multi-battery cascade, allowing only one slave.
This means that for anyone who owns more than two units, it's a pita! 
And the only way to handle this, from an electrical and monitoring perspective, is a workaround with many flows.
That said, the website's also not well implemented; I can only select a unit at a time and not as a cluster of master/slave.
The data shown is even more confusing if the unit I select is:
> master --> it shows the sum of the two units (master+slave)
> slave --> it shows only the slave data
So if I want to know the values produced by the machine, I need to calculate them (e.g., the 4 mppt production).
BUT! not for all the data; as an example, the battery data always refers to the single unit.
It's a confusing mess; it looks like Senec has problems finding a way.

In lala.cgi, the behavior is different since each unit is always showing its own dataset with no sum whatsoever.
To me, it looks like if I query:https://mein-senec.de/endkunde/api/status/getstatusoverview.php?anlageNummer=x
If it's a slave, I always read the slave data only.
If it's a master, I read the sum (master+slave) for all the values that report the "today" and "now" fields.

About your question:
In HA I've implemented your integration with four direct connections to all the units and the web integration as well.
For the latest, I couldn't specify a particular unit, therefore it took it automatically.
Couriously it didn't connect to #0 (as specified in your code, I belive), but connected to #2, which is the first of the two masters. Strange.
I tried to match all the information, and it looks like I can confirm that it's actually showing the sum of #2 and #1, so I guess that selecting #3 would contain #2.
BUT as noted before, not for all the data, only for the this values 
- forgridimport 
- powergenerated 
- consumption 
- accuexport 
- accuimport  
- acculevel 
- gridexport    

Sorry for the long post. I hope I was clear in my reporting. please let me know if you want me to perform any tests.Thank you

mgmatteo, 2023-10-11:
Ops sorry for the strange formatting

mgmatteo, 2023-10-11:
I'll try to replay your questions based on my setup and osservations

> ## webapi
> ### How the code currently working
> 
> When you use this HA WebApi Integration, then the code goes through the anlagenNumber from 0 and check, if this is a master, if not (it must be a slave) and checks, request anlagenNumber +1 - if this is also not a master, it will check then anlagenNumber 2... Once one master is found, this anlagenNumber will be used...

That clarifies why, in my WebApi integration, I see the data coming from 2 (my first master). I thought you only queryed for number 0, my bad :)

> So the sensors added by the webapi.integration will request the (first) anlagenNumber that has been found that is tagged as "master"... and the sensors will return the data that is the summary of master & slave data [at least that's what senec web-api is returning]...

> will be provided for one of the two of your senec systems - also important (IMHO) is the the value returned by the webapi for the `acculevel_now` value - in the situation, that slave have a accu_level of 0% and master have accu_level of 100% the webapi will return 50% as acculevel_now value! - so this is also some sort of "sum of master & slave"! [at least that was the outcome of the discussion in #8 ]

I don't think this is correct.
Based on my experience, 'acculevel_now' refers to the single unit, not the sum of master+slave.
For example, my current battery status is:
Master 1 -> 11% 
Slave 1 -> 27% (slaves always return a single value)
11% can't be the sum, in fact even if master is 0 the sum would be 13.
Sometimes I have a master 100% and a slave discharged, again not possible.
I'll collect some data to send you about this.

> ### possible steps to support the second master
> 
> As I tried to explain the current webapi-integration imply is trying to find (the first) master and only request data for this anlagenNumber via the web. I am not 100% confident, if just trying to find an additional "master" would be a smart move... I need to check some stuff... 

I try to think loudly (free translation from an Italian sentence..)
I suppose Senec will always number the units starting from 0 adding consecutive number for each new device added
If so, I think you can iterate through the numbers until you get an empty response (or error?)
At this point, you should have collected all the masters (slaves are irrilevant) and their corresponding "id"
But I would not sum the masters because it's more convenient to have the values separated and manipulated, if necessary, in HA.
For example, my two clusers are not the same: only one has the 'Backup Power pro' licence, and only one has the EVcharger.
Moreover, I have a 3-phase setup, but since Senec does not support 3-phases, I have cluster 1 on L1 and L2, cluster 2 on L2 and L3, each cluster has it's own meter to work correctly.
I also added a 'master' meter, not connected to any unit, just to double check that the sum of the values coming from the  two meters are correct (and other problems due to the on-site phase compensating mechanism).

Also IMHO there is no way to get a info which "slave" belongs to which master - this might be not so important [but right now the Integration shows the SenecNumbers of the master&slave together in the device information]... 
In your case you should see the two "slaves" together with the one master... But the data that is available would be only one of the two slaves...

Yes, I don't think it's really important, anyway I only have the master number and not the saves ones..

> ## lala.cgi / local polling
> 
> You wrote, that you have added for all four of your Senec's Systems to your HA installation - and that each system only shows it's "local" data... I have to admit, that I am wondering, if this is really the case - following the 'senec-logic' I would really expect, that the data you read/receive from a master unit will also include the data of the assigned slave device... So IMHO two of your four systems would have almost doubled values for generated PV...

This is the way I was used before the introduction of the new firmware.
I wrote a program that was collecting data from lala.cgi and was also able to control some basic functions of the units, plus some other nice things, but I don't think it is relevant now.
I can assure you that there are no duplicate values. Each unit reports its own value as an excepion of the statistics ones.
In this case, slave is slave only, anS master is the sum.
All other info were/are specific to the machine.
As an example, temperatures, statuses, allarms, etc., and also battery levels ;-)
I was able to identify probems on the machine that even the first level of Senec support didn't get on the first instance. They were really surprised about what I was able to discover :D 
Having the four direct connections also allows me to manage the 'manual battery recharge' and the 'storage' function, plus the production of each single string. 

> Also I am really interested what kind of data the "house_consumption" sensors are reporting for you (same for the grid import/export data). In a single "master & slave" setup I would expect, that only the master unit would provide data in the "ENERGY" subject... When you would request the data from the slave (IP), then I would expect, that house_consumption is identical to the data the master(IP) is providing...

Nope, slaves return slave data only.

> 
[... cut ...] inted on different cardinal points (like in my case). This let me correct some inefficency I had in the strings setup.

Back to the integration,
It seems to me that there was the possibility to query a specific serial number; this could also be an option.
In the case of multiple masters IMHO it's probably better to create two distinct configurations rather than having one that pulls everything.
Keeping the values separated it's very important, adding a sum in HA, if needed, is always easy to do.

marq24, 2023-10-12:
## webapi

> I don't think this is correct. Based on my experience, 'acculevel_now' refers to the single unit, not the sum of master+slave.
I am sorry - looks like that my memories where not correct - the `acculevel_now` might really the only value from the webapi that is indeed AnlagenNummer (0-x) specific... This implies that with the current version of the WebAPI the accu-level for a Slave Senec V4 is not available - for Senec V2/V3 it's the recommendation to use the value from the lala.cgi polling... 
 
### possible steps to support the second master in the WebAPI

> I try to think loudly (free translation from an Italian sentence..) I suppose Senec will always number the units starting from 0 adding consecutive number for each new device added If so, I think [...]

IMHO the different configurations (of the clusters) do not have any impact on the data currently delivered via the WebAPI - I don't get any information about tec-packages or anything like that... But in order to make the setup process for the User easier I might just would add an (export) option to specify an AnlagenNummer - default is AUTODETECT (which is the current implementation). This will give you all the options you went/need.

## lala.cgi / local polling

> You wrote, that you have added for all four of your Senec's Systems to your HA installation - and that each system only shows it's "local" data... I have to admit, that I am wondering, if this is [...]
 
> This is the way I was used before the introduction of the new firmware.

So just allow me to repeat what I have understood - BEFORE the Firmware Update (that introduced the HTTPS access and removed the STATISTICS Data) the local polling of the master (via lala.cgi) returned you the SUM's of master & slave - and AFTER the Update it's only the master? 

This is "Funny" since then the software-quality is even worse then expected! (mhhh - can it even be more worse?) Just a little background info concerning my private setup: Additionally to the Senec I have an additional SMA-Inverter installed. And the power values of this second Inverter had been always included in the data the Senec Unit has provided (web & local polling)... After the firmware update the data of the second Inverter (actually since I have an Hybrid Duo it's technically the third inverter) is included only for some hours of operation - but this effect is totally random! Sometimes the data is included for days - sometimes just for few hours - it became USELESS - so at the end of the day I have decided to do "everything" by myself and I have documented this here ["Better" house consumption data (based on AC power)](https://github.com/marq24/ha-senec-v3/discussions/30)

So I am just curious, if this randomness is also in your setup - OR - is it, that with the firmware update the data from the slave unit is no longer included in the master values (for the complete time since the update)?

> I wrote a program that was collecting data from lala.cgi and was also able to control some basic functions of the units, plus some other nice things, but I don't think it is relevant now. I can assure you that there are no duplicate values. Each unit reports its own value as an excepion of the statistics ones. In this case, slave is slave only, and master is the sum.

You still have STATISTICS data from your Senec lala.cgi?! - here it's gone since v0825... and I would be surprised if it would be still available in Italy...

> All other info were/are specific to the machine. As an example, temperatures, statuses, allarms, etc., and also battery levels ;-) I was able to identify probems on the machine that even the first [...]

Re 'plus the production of each single string' -> so you mean you simply read the data, that is available via the `PV1` section - right? Or is there something else?

---
#### I guess we have a misunderstanding here...

> Nope, slaves return slave data only.
>
>
> The "house_consumption" values of the two maser are completely different. As previously said, my two cluser work on different phaes combination (L1+ L2 and L2 [...]

---
I guess cause of the fact, that we are both not English native speakers some "information" get's lost... I'll try to explain with different words what I wanted to say...

From end user perspective I __would expect__, that values like grid_import/export & house_consumption values have to be identical for all Senec devices (at least the two masters) that are attached to your regional-power grid. BUT this is just my expectation how it __should be__. Since as I tried to explain - you have only "one" connection to the grid - and only "one" house/home that is consuming power... 

#### BUT

With the knowledge I have about the theory of operation of my single Senec unit here I believe, that the two Senec Masters at your location report different values - at least for "house_consumption"! And actually IMHO I have read from your reply, that this is really the case - different values at master1 and from master2. So you even installed additional (Senec independent) hardware in order to be able to actually measure the real grid import/export and the real house_consumption... 

my takeaway here is: Do everything by yourself - and make sure, that you do it on the "right" level - Power data from the MPP's is in DC - but you import/export from the grid (and consume in your home) AC - an for sure the inverters have some loss while converting from DC to AC... 

I have here installed also an additional Sensor/Hardware that provides the data directly from the Electricity meter (in order to be able to judge, if the Senec values are plausible - which they are - but just cause of my single device setup)... So my recommendation for you is, to wire the senec inverters directly and access the AC Power values from there (and use the link-power values to monitor the AC power that battery consume/provide.

mgmatteo, 2023-10-12:
> I am sorry - looks like that my memories where not correct - the `acculevel_now` might really the only value from the webapi that is indeed AnlagenNummer (0-x) specific... This implies that with the [...]

Yes It's exactly my setup:
direct connection for everything except satisctical data that has been removed with the latest firmware

> IMHO the different configurations (of the clusters) do not have any impact on the data currently delivered via the WebAPI - I don't get any information about tec-packages or anything like that... But in order to make the setup process for the User easier I might just would add an **expert** option to specify an AnlagenNummer - default is AUTODETECT (which is the current implementation). This will give you IMHO all the options you want/need.

Absolutely! Your proposal it's perfect. Simple for everyone and flexible for people like me that have more "wired" setups..
 
> So just allow me to repeat what I have understood - BEFORE the Firmware Update (that introduced the HTTPS access and removed the STATISTICS Data) the local polling of the master (via lala.cgi) returned you the SUM's of master & slave - and AFTER the Update it's only the master?

BEFORE the firmware update:
STATISTICS from the slave were always referred to the values generated only by the slave
STATISTICS from the master were the sum of master+slave (for the most part of the data, not everythng)
I still have old json from the lala.cgi with the indication of what was a sum and what was not, but I think now it is useless...
In my old setup (based only on the lala.cgi) I used the STATISTICS from my two master only, and all the other information from each single integration.

> This is "Funny" since then the software-quality is even worse then expected! (mhhh - can it even be more worse?) 

I completely agree with you. Senec software is well behind its HW counterpart.
And the choise to be so "close", even more with the recent changes, it's stupid to say less.
Just think about the limitation of the number of cascade units. That forced me to have two completely indipendant and not shared cluster. What a stupid setup.
Think of this: I have cluster 1 on L1 and L2, cluster 2 on L2 and L3, now imagine cluster 1 with 0% battery and cluster 2 with 100% battery. All my consumption in L1 would only go trough cluster 1, therefore I would use the grid power even if I have 20Kw of power in batteries of cluster 2!!! 
This is so stupid. 
I could workaround this if I had the possibility to force a battery discharge. In this case if my automation system detects the above situation I could manually foce a discharge of cluster 2 that would send power to the grid, in this case it would "cover" the usage derivating from L1 (cluster 1). But nope. Senec doesn't provide the control of the batteries...
Even the charge function (well hidden in the sw) is ment only for an emergency recharge, but as you wrote (and how I always use) it's very convenient to charge the battery during off peak time. by night I have a cost of 0€, so for me it's so much convenient to not discharge the battery by night (and save cycles) or even charge it if needed to compensate moring lack of sun... 
ok sorry for the deviation, but Senec just drive me crazy.

Just a little background info concerning my private setup: Additionally to the Senec I have an additional SMA-Inverter installed. And the power values of this second Inverter had been always included in the data the Senec Unit has provided (web & local polling)... After the firmware update the data of the second Inverter (actually since I have an Hybrid Duo it's technically the third inverter) is included only for some hours of operation - but this effect is totally random! Sometimes the data is included for days - sometimes just for few hours - it became USELESS - so at the end of the day I have decided to do "everything" by myself and I have documented this here ["Better" house consumption data (based on AC power)](https://github.com/marq24/ha-senec-v3/discussions/30)
> So I am just curious, if this randomness is also in your setup - OR - is it, that with the firmware update the data from the slave unit is no longer included in the master values (for the complete time since the update)?

Actually i didn't noticed loss of data from the slave. But it's Senec on Senec.. so hopefully this will not be a issue.

> You still have STATISTICS data from your Senec lala.cgi?! - here it's gone since v0825... and I would be surprised if it would be still available in Italy...

Sorry perhaps I wrote something not clear (my bad english). No I confirm I lost STATISTICS

> Re 'plus the production of each single string' -> so you mean you simply read the data, that is available via the `PV1` section - right? Or is there something else?

I'm not at home right now and I shoud chek on my other pc, but as far I remember yes. PV1 was the source.
I'll more precise later.

> From end user perspective I **would expect**, that values like grid_import/export & house_consumption values have to be identical for all Senec devices (at least the two masters) that are attached to your regional-power grid. **BUT** this is just my expectation how it **should be**. Since physically you have only "one" connection to the grid - and only "one" house/home that is consuming power... so values **should be** all the same...

In my prticulary setup this can't happen.
It would be logical in case of a single phase house, but in my case:
[... cut ...] ing the led pulse for the consumptions. Super spot on, but not really usefull. There are some devices that connects to the energy meter using elecrical protocols but are very expensive and I don't see the need of that, as I said my meter it's doing a very good job and I trust it so far. Off course I'm not aware if my energy provider start drifting for some strange reason hahaha :D  

Just let me know if I've been clear enoughf.. It's difficult for me to be clear in italian, imagine in english...

marq24, 2023-10-12:
... to get rid of all the quotes...

### WebAPI
a) please be so kind and remove your current WebAPI integration
b) update via HACS to 3.0.8-rc2 (BETA have to be enabled)
c) reboot HA
d) Reinstall 2 times the WebAPI and select your Master AnlagenNummer

### LocalPolling / Reading 'Electricity meter'
a) We agree that Senec SW just sucks
b) JFYI: My single Senec V3 Hybrid Duo provide power to all three phases (L1,L2 and L3)
c) JFYI: just bought here a [Tibber Pulse IR](https://tibber.com/de/store/produkt/pulse-ir) and [wrote this to solve my local challenges](https://github.com/marq24/ha-tibber-pulse-local)

mgmatteo, 2023-10-12:
Ok great,

WebAPI
a) removed the WebAPI integration, but not the direct ones
b) updated to 3.0.8-rc2 
c) rebooted
d) reinstalled the WebAPI

but now all the integrations are broken and no one is actually working, Any idea?

LocalPolling / Reading 'Electricity meter'
a) yes this is a simple one :D 
b) This is really courious! I followed the installation and also read the manual (I have the installateur manual if you need), and in my unit there are phisically just two connection points for the phases. So it is not possible to connect all the pahases to the inverter. The third one is compensated on site directly by the utility meter. For example if L1 and L2 are connected L3 uses the "in house exchange" (not shure how to translate it). In a way we can say that the Senec V3 Hybrid Duo provide power to all three phases even if in a logical way and not by a phisical connection. (But the result is that it's still energy self consumed an not bought from th grid).
But in my case is mo subdole!
As an example: since the two cluster doesn't talk to each other, If a cluster has battery, and the other one is still full the one with the empty battery will ghet power from the grid beaceuse the secod cluster doesen't know it has to send power from the battery, therefore I would pay for that energy.
c) I read your post, very intresting. Unfortunately Tibber it's not aviable here in Italy :/

mgmatteo, 2023-10-12:
Forgot to mention:
the error message is "failed to setup" for all the configurations (local and WebAPI)

mgmatteo, 2023-10-13:
I see you upgraded to 3.0.8-rc2
I deleted everyting from previous install and entities and trying it now..

mgmatteo, 2023-10-13:
Yeah!
Now i have 4 direct and 2 web connections working.
Let me collect some data and I'll be back.
Thank you!

marq24, 2023-10-13:
I am happy that this finally was sorted out - and that you are happy with the solution...

b) here is a screenshot from my german install manual... at least L1-L3 are connected to the Senec Device istelf - how this connection in then internally wired with the two build in inverters is obviously not known
[image]

c) IMHO you can use this Tibber Pulse IR without being Tibber customer - I am a future customer - but still have to wait till my current contract is candled - it's a Device reading data via IR from you main e-meter - transfer this data to a second device (Tibber Bridge - included in the package) - and from this bridge you can read the data to HA... 

and finally - when you are happy - please consider
[image]

[[image]](https://www.buymeacoffee.com/marquardt24)

closed 2023-10-14 by marq24

marq24, 2023-10-18:
First of all THANK YOU very much for your generous Coffee!

> HI, sorry it took some time for this feedback.

We all have our life and there are always good reasons, why a reply might take a bit longer...

> So far so good. I receive the data from both the cluster and both the integrations options (local and web). Unfortunately 3 days ago one of the slaves stopped to work. It turns off with the message "Switch off system. Lost display communication." Opened a ticket with Senec for this. And this is the third failure in 6 moths!!

I had a similar issue some weeks ago - also with a "lost communication" message - I tried "everything" to restart the device manually - also tried to disconnect it from the grid (by unpluging all fuses) - all did not work... Some days (could be a ~week) the device just recovered "by itself" - obviously I have contacted my local installateur as soon as the error came up... But except notifying Senec there was IMHO not much they could do)... 

> I'll have a look to the Tibber Pulse IR as you suggest, thank you.

It's still possible, that they only "sell" the device to already registered customers - but I am curious! 

> Yes 3 phases can be connected to the unit but only two of them are managed internally. The third one is compensated "on site". Sorry I don't know how to say this in english. But something like 2 of them are real the 3rd one is virtual... This was a major problem we discussed with the installateur, but the problem is only if you have more than a master/slave cluster... :/

Ok thanks for this clarification then - I thought, when 3 Phases go in, that the hardware would do something with that... But when I understood you correctly, then in the moment I am starting to use the (not really wired) 3'rd phase, the power will come always from the GRID (no matter of the battery capacity or PV power)?

mgmatteo, 2023-10-18:
> I had a similar issue some weeks ago - also with a "lost communication" message - I tried "everything" to restart the device manually - also tried to disconnect it from the grid (by unpluging all [...]

Same here. A few hours ago, the unit resurrected..
No message from Senec so far, but I guess they did some kind of "reset"?
Just a curiosity:
I had a blakout of data (web-api integration) for a couple of days, but only in the cluster with the defective unit and only on the web-api integration route. The local integration kept working just fine.
This interruption didn't begin on the same day as the problem on the unit rose, but a day later, and fixed itself this morning. Coincidence?

> Ok thanks for this clarification then - I thought, when 3 Phases go in, that the hardware would do something with that... But when I understood you correctly, then in the moment I am starting to use [...]

Yes, BUT the inverter senses this (enfluri) and "pushes" the same amount of energy into the grid. By doing this, we are not charged for the consumption in the 3rd phase.

marq24, 2023-10-18:
> Same here. A few hours ago, the unit resurrected.. No message from Senec so far, but I guess they did some kind of "reset"? Just a curiosity: I had a blackout of data (web-api integration) for a couple of days, but only in the cluster with the defective unit and only on the web-api integration route. The local integration kept working just fine. This interruption didn't begin on the same day as the problem on the unit rose, but a day later, and fixed itself this morning. Coincidence?

Ahhh - yes - I remember - (but since I am 50+ years old I can't remember the details) - I also faced a "complete" data loss at mein-senec.de portal (almost all data was wiped from the history)... This data returned after a certain amount of time... But when I can remember correctly, then this was before the "connection-loss" issue - so I would like to call it just 'Coincidence'... Or better "everything-is-possible-cause-it's-senec"

mgmatteo, 2023-10-18:
Same here 51
my brain logistic is FIFO. First in first out...