marq24/ha-senec-v3 #42: Suggestion to use app-gateway-prod.senecops.com to retrieve webapi Data

oakdesign, 2023-10-26:
Fist of all thank you for integrating the webapi into the the integration. 
I personally use nodered in combination with Homeassistant alongside with influxdb and grafana. 
In oder to get the webapi data I use node-red and ran into a similar issue like you did in 
https://github.com/marq24/ha-senec-v3/issues/7
After further inspection, there is an alternative way to get the same data but through a different source which is the senec 
https://app-gateway-prod.senecops.com which host the service the senec app is using to retreive its data. this is a overview of all available service endpoints
https://documenter.getpostman.com/view/10329335/UVCB9ihW#3e5a4286-c7d2-49d1-8856-12bba9fb5c6e
In comparision to the mein-senec the login and authorisation is not based on a session cookie rather the a token. From my current testing the token only has to be obtained once alongside with the value for the anlagenid and does not expire in contrary to the JSESSIONID cookies used by mein-senec.de

marq24, 2023-10-26:
Hi @oakdesign!

when you run into issues with following the login redirects of mein-senec.de (when using aiohttp) - then you need https://github.com/aio-libs/aiohttp 3.9+ (since the new version - not released yet) will include a fix for the cookie handling... If you don't like to wait the code here includes the fix/workaround by setting an own the cookieJar implementation...

The main reason why I would like to stick with mein-senec.de (and don't use the app-gateway-prod.senecops.com) - this host and API is not public at all - yes obviously we can see that it exists right now (and will be used by the App) - but we never know, when this host & API might be replaced/removed at any time...

I don't see (yet) any advantage of a longer cookie expire-duration... IMHO the infrastructure setup of the host is 'suboptimal' - but that's IMHO just matching the rest of the company's IT department.

So - yes - I knew this 'app-gateway-prod.senecops.com' host (and the endpoints) before - but (right now) I don't see any advantage of using them as long as mein-senec.de is officially available. Of course you are invited to change my mind :-D

oakdesign, 2023-10-26:
Only real advantage I see at the moment would be the ability to set wallbox properties though the app gateway endpoints. At least I haven't identified corresponding service endpoints on the mein-senec.de wallbox controllers.

Btw. from my point of view the mein-senec API endpoints are not more or less public than the app gateway. if we grab A from browser devtools or B through monitoring the DNS traffic of the app, both are not officially public documented by senec ;-)

oakdesign, 2023-10-26:
`and forgot to mention - let me know, if you want/need additional information about the using own CookieJar impl with aiohttp (in order to successfully access mein-senec.de)`

THX for the offer, I have some 25 years as Dev and Architect for web based applications, so have no issues to get the mein-senec.de running as well as using this integration in order to have entities for my power-plus enrgy plus cards. 
On the other hand as I use influxdb as a long-term storage of metrics and don't like the data structure HA influxdb integration is creating for strong entities, I had to use my own nodered flow for measurement storage.

On other small advantage of the app-gateway in comparison to the mein-senec.d I see in collecting the statistic total. With the app-gateway I would be possible to get all total values with a single service call compared to the multiple calls with type=xxx calls.

` so would be interesting, if it's possible to manipulate wallbox settings from the mein-senec.de url [actually I hope that the security layer will intercept - but who knows!]`

I don't think so. My mein-senec.de UI is brette messed an limited I guess because I have a V4. Compared to others I only have half of the menu of other users. My first thought was that it's account based restirctions but then realized that they dont even have a proper binding for the Home 4 and therefore result in a "?view.product.unknown?" and therefore not loading a proper NavigationViewModel. 
Before I found your integration had then to search the js controllers in order to identity that I have access to everything, they just are not managing to show it

oakdesign, 2023-11-11:
@marq24 might have found some data that is not available from the mein-senec.de API but through the app-gateway in terms of systemstatus. I'm currently quite busy with a large software package delivery for 2 automotive customers. But as soon as I find some time might contribute with a app-gateway code. Meanwhile through the latest app version 4.3.3 and 4.4.0_beta01 ther is for example 
`https://app-gateway-prod.senecops.com/v1/senec/monitor/{{SENEC_ANLAGE}}/data?period=MONTH&date=2023-11-14&locale=de_DE&timezone=Europe%2FBerlin`

that returns so for example case and module temperatures for Home 4 systems

`{
    "systemOverview": {
        "systemId": xxxxxx,
        "productName": "Unknown Product",
        "installationDate": 1696513596000
},
    "casing": {
        "serial": "S4H1-xxxxxxxx-xxxx-D4",
        "temperatureInCelsius": 29.0
},
    "mcu": {
        "mainControllerSerial": "v4-xxxxxxxx",
        "mainControllerState": {
            "name": "UNKNOWN",
            "severity": "WARNING"
},
        "firmwareVersion": null,
        "guiVersion": null
},
    "warranty": {
        "endDate": 2012162400000,
        "warrantyTermInMonths": 120
},
    "batteryModules": [
{
            "ordinal": 1,
            "state": {
                "state": "OK",
                "severity": "INFO"
},
            "vendor": "BMZ",
            "serialNumber": "xxxxxxxxxxx",
            "firmwareVersion": "6.3.1",
            "mainboardHardwareVersion": "624026",
            "mainboardExtensionHardwareVersion": "-1610219325",
            "minTemperature": 24.299999237060547,
            "maxTemperature": 29.100000381469727,
            "lastContact": {
                "time": 1699660976857,
                "severity": "INFO"
},
            "flags": []
},
{
            "ordinal": 2,
            "state": {
                "state": "OK",
                "severity": "INFO"
},
            "vendor": "BMZ",
            "serialNumber": "xxxxxxxxxxx",
            "firmwareVersion": "6.3.1",
            "mainboardHardwareVersion": "624026",
            "mainboardExtensionHardwareVersion": "-1610219325",
            "minTemperature": 24.700000762939453,
            "maxTemperature": 30.100000381469727,
            "lastContact": {
                "time": 1699660976857,
                "severity": "INFO"
},
            "flags": []
},
{
            "ordinal": 3,
            "state": {
                "state": "OK",
                "severity": "INFO"
},
            "vendor": "BMZ",
            "serialNumber": "xxxxxxxxxxx",
            "firmwareVersion": "6.3.1",
            "mainboardHardwareVersion": "624026",
            "mainboardExtensionHardwareVersion": "-1610219325",
            "minTemperature": 24.200000762939453,
            "maxTemperature": 27.700000762939453,
            "lastContact": {
                "time": 1699660976857,
                "severity": "INFO"
},
            "flags": []
}
],
    "batteryInverter": {
        "state": {
            "name": null,
            "severity": null
},
        "vendor": null,
        "firmware": {
            "firmwareVersion": null,
            "firmwareVersionHumanMachineInterface": null,
            "firmwareVersionPowerUnit": null,
            "firmwareVersionBidirectionalDcConverter": null
},
        "temperatures": {
            "amb": null,
            "halfBridge1": null,
            "halfBridge2": null,
            "throttle": null,
            "max": null
},
        "lastContact": {
            "time": 0,
            "severity": "ERROR"
},
        "flags": []
},
    "batteryPack": {
        "numberOfBatteryModules": 3,
        "technology": "LITHIUM_ION",
        "maxCapacityInKwh": null,
        "maxChargingPowerInKw": null,
        "maxDischargingPowerInKw": null,
        "currentChargingLevelInPercent": 6.0,
        "currentVoltageInV": 201.0,
        "currentCurrentInA": -1.8619999885559082,
        "remainingCapacityInPercent": 99.0
},
    "installer": {
        "companyName": "xxxxxxxxx GmbH",
        "email": "xxx@xxxxxxx.de",
        "phoneNumber": null,
        "address": {
            "street": "xxxxxxx-Str.",
            "houseNumber": "xx",
            "postcode": "xxxxxx",
            "city": "xxxxxxx",
            "countryCode": "DE",
            "region": null,
            "longitude": xxxxxxx,
            "latitude": xxxxxxxx,
            "timezone": "Europe/Berlin"
},
        "onlineMonitoringAllowed": true,
        "website": null
},
    "slaveSystemInfoDtos": []
}`

marq24, 2023-11-11:
> @marq24 might have found some data that is not available from the mein-senec.de API but through the app-gateway in terms of systemstatus. I'm currently quite busy with a large software package delivery for 2 automotive customers.

GOOD LUCK! - "Your Spaltmaß is killing my Software" - I worked 15+ years in this area - and finally got away...

> But as soon as I find some time might contribute with a app-gateway code. Meanwhile through the latest app version 4.3.3 and 4.4.0_beta01 ther is for example `https://app-gateway.prod.senec.dev/v1/senec/systems/{{SENEC_ANLAGE}}/technical-data`

Actually my current train of thought would be to add an additional "connection-type" - so the user can choose "direct", Mein-Senec or WebAPI (and the additional internal inverter stuff),,,

oakdesign, 2023-11-11:
> Actually my current train of thought would be to add an additional "connection-type" - so the user can choose "direct", Mein-Senec or WebAPI (and the additional internal inverter stuff)

same idea I had in mind.

meanwhile I'm going to ducument the current app-gateway serives of App Version > 4.3.3 here. As the Serviceendpoints and request parameters and response jsons have been changed compared to the previous availabe postman documentation of app version 3.12.0 from 2021

https://documenter.getpostman.com/view/932140/2s9YXib2td

marq24, 2023-11-12:
Might be it was yesterday just too late - or too early this morning - I try to understand the new postman stuff - some URL contains a `set automatically` section - mmmmmmhhh while just writing... - I assume __just NOW I get it__, that this is probably the SENEC_ANLAGEN_ID... 

So basically that's the ANLAGEN_ID that have to be placed into the URL... ok - makes sense - but this will instantly rise the question, about master & slave setup's (that was hard part of the current web-api implementation) - but this should not be a show stopper for now...

iot-sle, 2023-11-30:
At least the possibility to set the wallbox parameters is very useful. To set the limit of charge current in pv mode is not possible by web ui. So no way using mein-senec.de. 
For some unknown reasons Senec decided to set the the upper limit which can be set as "minimum charge current" by senec app to a fix value of 12A. Older versions of the app are using maximum available current of the wallbox instead. There is no other way to limit the wallbox max charge, without (mis)using the pv guided charging and set the mimumim current to wanted value. 
By example: My wallbox is able to do 22kW max charge (32A), but I want to use the 11kW standard charge (16A). Using the new app it isn't possible. Using an old version of the app using in a vm it is still working like a charm. The API is still fine and with an integration I could do the settings in HA.

marq24, 2024-01-14:
@oakdesign I am currently looking "a bit" closer into your request - or better into the data that will be returned... I have to admit, I am basically clueless, what call's I have to make to the APP-API in order to get

A) "lastest" total statistics data
B) "current" values from the "grid"...

Yes I can get current battery status [currentChargingLevelInPercent, currentVoltageInV, currentCurrentInA] from the batteryPack (when calling the technical-data endpoint)...

But I do not manage to create a URL (that does not return 99% mot needed data) for the 'zeitverlauf' (or 'statistik') endpoints - I would like to be able to use something like "after" (and not "before")

marq24, 2024-01-14:
I am already closer to what I was looking for...
https://documenter.getpostman.com/view/932140/2s9YXib2td#155c4e33-9ee2-4c79-ab3f-4df9a0d76c84

bobthebuilder4711, 2024-01-19:
Have you tried the data/custom command with something like 2001-01-01 as the startDate and today's date as the endDate?

marq24, 2024-01-19:
lowest startDate that I can use is 2018-01-01... for endDate I can use current month +1 [JFYI]

marq24, 2024-01-20:
so you have an initial version (beta) now... [no new sensors yet] -> https://github.com/marq24/ha-senec-v3/releases/tag/2024.0.2

marq24, 2024-01-21:
@oakdesign looking forward to your feedback to https://github.com/marq24/ha-senec-v3/releases/tag/2024.0.3

lennartschulte, 2024-01-23:
2024.0.1 was the last working version for me. The 2 beta releases after that did not initialize properly. Did a rollback but have not checked the release 2024.0.6 yet. I think I should not try any beta in my case, although the 2 beta releases in-between don't show up anymore

marq24, 2024-01-23:
of course the installation of the beta version(s) is totally optional - but have in mind, when there is no feedback (bug reports) beta releases will make it to the release channel... 

If a beta version does not work for you - please create a issue (providing details from you log)... TIA

lennartschulte, 2024-01-24:
Totally true. I will provide the logs with a new test of the current beta

marq24, 2024-01-27:
closed with release of 2024.0.9

closed 2024-01-27 by marq24