marq24/ha-senec-v3 #43: Login WebAPI (mein-senec.de) failed

GregorEstebanCardosoSchiller, 2023-11-09:
Hallo,
ich weiß noch nicht genau, was ich falsch mache, da das Issue offenbar bei keinem anderen besteht, aber die Logindaten habe ich inzwischen x mal überprüft und auch mehrfach händisch bei HA als auch bei senec direkt eingegeben. Bei Senec läufts, im HA aber nicht (siehe Screenshot) Ergo: Was mache ich falsch?

[image]

Meine Version ist 3.0.6 - die normale Integration läuft problemlos. Habe jetzt auch händisch den Code nochmal aktualisiert (von hier: https://github.com/marq24/ha-senec-v3/releases/tag/3.0.6) und HA neu gestartet. - Keine Änderung.

GregorEstebanCardosoSchiller, 2023-11-10:
Hi,
PW mit simpelsten Sonderzeichen am Ende geändert. + und - 
HA auf aktuellem Stand, Debug gibt nicht mehr aus als:

[image]

marq24, 2023-11-10:
you might like to try the latest beta version - but my guess goes into the direction, that the cookie jar patching does not work in your specific setup/environment [no clue why this can be the case]... 

Would you be interested in digging deeper into the possible root cause? But this requires some manual efforts - like reading directly the log output from the console (and not via the gui)...

GregorEstebanCardosoSchiller, 2023-11-10:
Everything for the cause =)
Tell me what to do.

HASS OS, Intel Nuc

marq24, 2023-11-11:
Might be, that HASS OS could be already the source of the issue? [Might be that this use a different aiohttp lib?] - So for the start please update to the beta channel, enable the debug log output for the integration [I guess this has to happen in the configuration.yaml] and try to get access to the actual log file of HA [I use here a docker setup which makes it very easy for me to access the complete log output - including all stack traces (and that is what I need - a stack trace)]

GregorEstebanCardosoSchiller, 2023-11-11:
Good morning,
updated to beta (3.0.8-rc5) debug protocoll:

2023-11-11 07:59:50.334 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.262 seconds (success: True)
2023-11-11 07:59:51.336 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.265 seconds (success: True)
2023-11-11 07:59:52.332 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.261 seconds (success: True)
2023-11-11 07:59:52.931 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting MySenecWebPortal... for user: 'verratichnicht@gmx.de' with options: None
2023-11-11 07:59:52.931 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(self) ********
2023-11-11 07:59:53.368 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.296 seconds (success: True)
2023-11-11 07:59:53.450 WARNING (MainThread) [custom_components.senec.config_flow] Could not connect to mein-senec.de with 'gregschill@gmx.de', check credentials
2023-11-11 07:59:53.450 ERROR (MainThread) [custom_components.senec.config_flow] Could not connect to mein-senec.de with User 'verratichnicht@gmx.de', check credentials
2023-11-11 07:59:54.369 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.298 seconds (success: True)
2023-11-11 07:59:55.333 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.261 seconds (success: True)
2023-11-11 07:59:56.330 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.259 seconds (success: True)
2023-11-11 07:59:57.339 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.268 seconds (success: True)

working on trace log...

marq24, 2023-11-12:
... ok looks like that all traces are catched... but since in the console output we read a DEBUG, you ave successfully managed to enable the debug log (and have access to it)... 

I hope you have also access to the filesystem where the `custom_components` folder is located... In this folder you have the `senec` folder (which is this integration)... I am not sure if you have experience with python - but the most important thing is, that you have to keep the intents...

What I am asking you to do is to alter the integration code locally - so that we can finally see the exception trace... 

please open/edit the following file `custom_components/senec/pysenec_ha/__init__.py` file and search for:

```
    async def authenticate(self, doUpdate: bool, throw401: bool):
        _LOGGER.info("***** authenticate(self) ********")
        self.checkCookieJarType()
        auth_payload = {
            "username": self._SENEC_USERNAME,
            "password": self._SENEC_PASSWORD
        }
        async with self.websession.post(self._SENEC_AUTH_URL, data=auth_payload, max_redirects=20) as res:
            try:
                res.raise_for_status()
                if res.status == 200:
                    # be gentle reading the complete response...
                    r_json = await res.text()
                    self._isAuthenticated = True
                    _LOGGER.info("Login successful")
                    if doUpdate:
                        await self.update()
                else:
                    _LOGGER.error("Login failed with Code " + str(res.status))
                    self.purgeSenecCookies()
            except ClientResponseError as exc:
                #_LOGGER.error(str(exc))
                if throw401:
                    raise exc
                else:
                    if exc.status == 401:
                        self.purgeSenecCookies()
                        self._isAuthenticated = False
                    else:
                        _LOGGER.error("Login exception with Code " + str(exc.status))
                        self.purgeSenecCookies()
```

and remove the `#` in front of the _LOGGER.error(str(exc)) - or better replace this line with:

`_LOGGER.error(f"ClientResponseError while authenticate {exc}", exc)`

so the complete method should look like:

```
    async def authenticate(self, doUpdate: bool, throw401: bool):
        _LOGGER.info("***** authenticate(self) ********")
        self.checkCookieJarType()
        auth_payload = {
            "username": self._SENEC_USERNAME,
            "password": self._SENEC_PASSWORD
        }
        async with self.websession.post(self._SENEC_AUTH_URL, data=auth_payload, max_redirects=20) as res:
            try:
                res.raise_for_status()
                if res.status == 200:
                    # be gentle reading the complete response...
                    r_json = await res.text()
                    self._isAuthenticated = True
                    _LOGGER.info("Login successful")
                    if doUpdate:
                        await self.update()
                else:
                    _LOGGER.error(f"ClientResponseError while authenticate {exc}", exc)
                    self.purgeSenecCookies()
            except ClientResponseError as exc:
                #_LOGGER.error(str(exc))
                if throw401:
                    raise exc
                else:
                    if exc.status == 401:
                        self.purgeSenecCookies()
                        self._isAuthenticated = False
                    else:
                        _LOGGER.error("Login exception with Code " + str(exc.status))
                        self.purgeSenecCookies()
```

GregorEstebanCardosoSchiller, 2023-11-13:
Hi,
you can throw anything at me ;-)

Output:

2023-11-13 08:39:32.063 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.261 seconds (success: True)
2023-11-13 08:39:32.639 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting MySenecWebPortal... for user: 'lalaland@xxx.de' with options: None
2023-11-13 08:39:32.640 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(self) ********
2023-11-13 08:39:33.061 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.262 seconds (success: True)
2023-11-13 08:39:33.157 WARNING (MainThread) [custom_components.senec.config_flow] Could not connect to mein-senec.de with 'lalaland@xxx.de', check credentials
2023-11-13 08:39:33.157 ERROR (MainThread) [custom_components.senec.config_flow] Could not connect to mein-senec.de with User 'lalaland@xxx.de', check credentials

marq24, 2023-11-13:
this is a bit strange - I would expect more output from 'pysenec_ha'

`2023-11-13 08:39:32.640 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(self) ********`

either one of this three lines should had generated some output in the log ?!

- _LOGGER.info("Login successful")
- _LOGGER.error("Login failed with Code " + str(res.status))
- _LOGGER.error(f"ClientResponseError while authenticate {exc}", exc)

Are you sure, that your modified code was executed? [you might need to restart HA (one again)?!]

GregorEstebanCardosoSchiller, 2023-11-13:
Strange indeed, but the modified file is live. Tested with "***** authenticate(selftest) ********" instead of (self), just to be sure.

2023-11-13 10:44:37.282 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********

still no further output from pysenec_ha.

WTF....I just tried to debug the file myself (inserted another logger within the try...("Inside try")
and now the whole config did work? Waaaaaas zur Hölle...das ergibt doch keinen Sinn...

2023-11-13 10:48:56.780 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:48:56.999 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:48:59.407 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:48:59.408 INFO (MainThread) [custom_components.senec.pysenec_ha] Login successful
2023-11-13 10:48:59.408 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** update(self) ********
2023-11-13 10:48:59.408 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** update_now_kW_stats(self) ********
2023-11-13 10:49:01.330 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:03.597 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:03.820 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:04.300 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:04.521 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:05.972 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:06.185 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:07.760 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:07.981 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:09.451 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:09.678 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:10.157 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:10.383 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:11.798 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:12.065 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:12.538 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:12.546 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** update_context(self) ********
2023-11-13 10:49:12.546 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(selftest) ********
2023-11-13 10:49:13.933 INFO (MainThread) [custom_components.senec.pysenec_ha] Inside try
2023-11-13 10:49:13.946 INFO (MainThread) [custom_components.senec.binary_sensor] No binary_sensors for WebPortal...
2023-11-13 10:49:13.946 INFO (MainThread) [custom_components.senec.switch] No switches for WebPortal...

The module now collects the data...No explanation why this is now suddenly working...

marq24, 2023-11-13:
a true "software developer WTF moment" - I assume somebody @ senec side have made an adjustment (your login attempts had been monitored and they realized an issue on their end with your account)...

GregorEstebanCardosoSchiller, 2023-11-13:
Zu früh gefreut. Die Statistiken haben sich nicht weiter geupdatet. Kriege nun folgende Nachricht aus dem Log:

Bad logger message: ClientResponseError while authenticate 401, message='', url=URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal') ((ClientResponseError(RequestInfo(url=URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal'), method='GET', headers=<CIMultiDictProxy('Host': 'mein-senec.de', 'User-Agent': 'HomeAssistant/2023.11.1 aiohttp/3.8.5 Python/3.11', 'Accept': '*/*', 'Accept-Encoding': 'gzip, deflate', 'Cookie': 'JSESSIONID=A5C0D7398E8001F38A332A2FF530532F')>, real_url=URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal')), (<ClientResponse(https://mein-senec.de/auth/login) [302 ]> <CIMultiDictProxy('Vary': 'Origin', 'Vary': 'Access-Control-Request-Method', 'Vary': 'Access-Control-Request-Headers', 'Set-Cookie': 'JSESSIONID=268B847AD3C11E46D8EAA498D28CEDE0; Path=/auth; HttpOnly', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'DENY', 'Location': '/auth/', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:54 GMT', 'x-server': 'node2')> , <ClientResponse(https://mein-senec.de/auth/) [302 ]> <CIMultiDictProxy('Vary': 'Origin', 'Vary': 'Access-Control-Request-Method', 'Vary': 'Access-Control-Request-Headers', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'DENY', 'Location': 'https://mein-senec.de', 'Content-Language': 'en', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:54 GMT', 'x-server': 'node2')> , <ClientResponse(https://mein-senec.de) [302 ]> <CIMultiDictProxy('Set-Cookie': 'JSESSIONID=A5C0D7398E8001F38A332A2FF530532F; Path=/; HttpOnly', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'SAMEORIGIN', 'Location': 'http://mein-senec.de/oauth2/authorization/login-portal', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:55 GMT', 'x-server': 'node2')> , <ClientResponse(http://mein-senec.de/oauth2/authorization/login-portal) [301 Moved Permanently]> <CIMultiDictProxy('Content-Length': '0', 'Location': 'https://mein-senec.de/oauth2/authorization/login-portal')> , <ClientResponse(https://mein-senec.de/oauth2/authorization/login-portal) [302 ]> <CIMultiDictProxy('x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'SAMEORIGIN', 'Location': 'https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:55 GMT', 'x-server': 'node2')> ), status=401, headers=<CIMultiDictProxy('Set-Cookie': 'JSESSIONID=CD4160A50074CAE62C2354FFB1EDFEC8; Path=/auth; HttpOnly', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'DENY', 'Transfer-Encoding': 'chunked', 'Date': 'Mon, 13 Nov 2023 13:34:55 GMT', 'x-server': 'node2')>),))

marq24, 2023-11-13:
um mal einwenig ORDNUNG in diese Matsche reinzubringen...

DER request geht auf die Bretter 'https://mein-senec.de/auth/oauth2/authorize':
```
((ClientResponseError(RequestInfo(
URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal'),
method='GET',
headers=<CIMultiDictProxy('Host': 'mein-senec.de', 'User-Agent': 'HomeAssistant/2023.11.1 aiohttp/3.8.5 Python/3.11', 'Accept': '/', 'Accept-Encoding': 'gzip, deflate',
'Cookie': 'JSESSIONID=A5C0D7398E8001F38A332A2FF530532F')>,
URL('https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal')),
```

Es wird die JSESSIONID=__A5C0D7398E8001F38A332A2FF530532F__ verwendet... wenn man sich aber den flow ansieht, dann ist das die session id von der base URL 'https://mein-senec.de'... für die URL 'https://mein-senec.de/__oauth2/__' wird aber eine andere JSESSIONID gesetzt (nämlich die __CD4160A50074CAE62C2354FFB1EDFEC8__)

```
--- FLOW from HERE ---

(<ClientResponse(https://mein-senec.de/auth/login) [302 ]>
<CIMultiDictProxy('Vary': 'Origin', 'Vary': 'Access-Control-Request-Method', 'Vary': 'Access-Control-Request-Headers',
'Set-Cookie': 'JSESSIONID=268B847AD3C11E46D8EAA498D28CEDE0; Path=/auth; HttpOnly', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'DENY', 
'Location': '/auth/', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:54 GMT', 'x-server': 'node2')> ,

<ClientResponse(https://mein-senec.de/auth/) [302 ]>
<CIMultiDictProxy('Vary': 'Origin', 'Vary': 'Access-Control-Request-Method', 'Vary': 'Access-Control-Request-Headers', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'DENY', 
'Location': 'https://mein-senec.de', 'Content-Language': 'en', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:54 GMT', 'x-server': 'node2')> ,

<ClientResponse(https://mein-senec.de) [302 ]>
<CIMultiDictProxy(
'Set-Cookie': 'JSESSIONID=A5C0D7398E8001F38A332A2FF530532F; Path=/; HttpOnly', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'SAMEORIGIN', 
'Location': 'http://mein-senec.de/oauth2/authorization/login-portal', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:55 GMT', 'x-server': 'node2')> ,

<ClientResponse(http://mein-senec.de/oauth2/authorization/login-portal) [301 Moved Permanently]>
<CIMultiDictProxy('Content-Length': '0',
'Location': 'https://mein-senec.de/oauth2/authorization/login-portal')> ,

<ClientResponse(https://mein-senec.de/oauth2/authorization/login-portal) [302 ]>
<CIMultiDictProxy('x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'SAMEORIGIN',
'Location': 'https://mein-senec.de/auth/oauth2/authorize?response_type=code&client_id=login-portal&scope=user&state=jD26m_byP9x5KCsP3CwQpU8786ffDL58U3-JWdm-huQ%3D&redirect_uri=https://mein-senec.de/login/oauth2/code/login-portal', 'Content-Length': '0', 'Date': 'Mon, 13 Nov 2023 13:34:55 GMT', 'x-server': 'node2')> ), status=401, headers=<CIMultiDictProxy(
'Set-Cookie': 'JSESSIONID=CD4160A50074CAE62C2354FFB1EDFEC8; Path=/auth; HttpOnly', 'x-content-type-options': 'nosniff', 'x-xss-protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'x-frame-options': 'DENY', 'Transfer-Encoding': 'chunked', 'Date': 'Mon, 13 Nov 2023 13:34:55 GMT', 'x-server': 'node2')>),))
```

Das ist genau der BUG in der aiohttp lib, den ich fixen musste, dammit der Zugriff auf das mein-senec.de portal überhaupt mit aiohttp funktioniert - leider wird der fix erst ab der version 3.9 (meine ich zumindst) ausgerollt - die gibt es aber noch nicht - deswegen enthält meine Integration einen PATCH, der versucht ein alternatives CookieHandling für die aiohttp sessions zu forcen... Das ganze passiert im Code ab hier: [https://github.com/marq24/ha-senec-v3/blob/849a897bcb80049f486c3e7d1cd1069e5ad19d3a/custom_components/senec/pysenec_ha/__init__.py#L1347](https://github.com/marq24/ha-senec-v3/blob/849a897bcb80049f486c3e7d1cd1069e5ad19d3a/custom_components/senec/pysenec_ha/__init__.py#L1347)

Dieser Austausch des CookieJar's scheint bei Deiner speziellen Konstellation meistens NICHT zu funktionieren (oder Du hast eine weitere Integration am start, die "meine" CookieJar Implementierung wieder auf die default Implementierung zurücksetzt... 

Du kannst bei Dir lokal mal das python script laufen lassen (das habe ich verwendet um den bug in der aiohttp lib zu fixen)...

```
import aiohttp
import asyncio
import logging

_LOGGER = logging.getLogger(__name__)
logging.basicConfig(level=logging.DEBUG)

import custom_components.senec.pysenec_ha
import custom_components.senec.pysenec_ha.util

async def on_request_redirect(session, trace_config_ctx, params):
[... cut ...]
    async with aiohttp.ClientSession(trace_configs=[trace_config]) as session:
        sen = custom_components.senec.pysenec_ha.MySenecWebPortal(user="[YOUR_USER_HERE]", pwd="[YOUR_PWD_HERE]", websession=session)
        await sen.authenticate(doUpdate=False, throw401=True)
        await sen.update_context()
        print('AnlagenNummer: '+ str(sen._master_plant_number))

        await sen.update()
        print(sen._isAuthenticated)
        print(sen._energy_entities)
        print(sen._power_entities)
        print(sen._battery_entities)

asyncio.run(main())
```

GregorEstebanCardosoSchiller, 2023-11-14:
Das ist im HASSOS nicht so einfach. Ich habe zwar meinen Terminal, aber die externen Bibliotheken sind bei ner lokalen Ausführung nicht abrufbar. Da muss ich mal noch ein wenig recherchieren. Die HASSOS Geschichte war am Anfang die einfachste Variante zu starten, gibt auch per se keine Probleme, inzwischen ist jedoch das "komplexere Bedienen" nur sehr nervig durchführbar.
[image]

marq24, 2023-11-24:
sorry - that your reply slippend though my attention - that was a busy week... the error on the second screenshot is (IMHO) fine... that's cause the test called some additional stiff that is not present in your environment - the most important thing is, that the code is able to login (with the patched CookieJar) - so this is indicating, that in your HA environment something is "overwriting" the cookie handler (that will be set by "my" integration)... so we have a race condition here... (that's my interpretation)... 

Not sure if you are able to disable other integrations one by one - till you find the other integration that seams to be in conflict (with the Senec)...

marq24, 2023-11-24:
BUT the code line

`if self._QUERY_SPARE_CAPACITY:`

does not exist in the BETA releases - it should be

`if hasattr(self, '_QUERY_SPARE_CAPACITY') and self._QUERY_SPARE_CAPACITY:`

so I am asking myself, if you might simply just need to update your integration?! to [3.0.9-rc1](https://github.com/marq24/ha-senec-v3/tree/3.0.9-rc1)

marq24, 2023-11-28:
ok - first of all with the new version the script need to be adjusted slightly:

search for:
await sen.authenticate(doUpdate=False, throw401=True)

and replace with:
await sen.authenticate(do_update=False, throw401=True)

marq24, 2023-11-28:
and in https://private-user-images.githubusercontent.com/65222903/286217121-6a308687-b014-4e84-a919-0ab55ac41f87.jpg you have included your email - you might like to remove that image again...

GregorEstebanCardosoSchiller, 2023-11-28:
Siehe hier:

[image]

marq24, 2023-11-28:
image seams to be broken ? :-/

GregorEstebanCardosoSchiller, 2023-11-28:
update

GregorEstebanCardosoSchiller, 2023-11-28:
So it seems, that as you said, the code is working:

[image]

But I have no idea how i should check which integrations might interfere...shall I look for aiohttp? 

Suggestions? ^^

[image]

marq24, 2023-11-28:
just as "proof" - can you please update to 3.1.1, enable DEBUG Log for the senec integration and then (try) to reinstall the WebAPI? - TIA [I hope the messages are now better to understand]

GregorEstebanCardosoSchiller, 2023-11-28:
Updated to 3.1.1:

2023-11-28 13:53:11.287 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.269 seconds (success: True)
2023-11-28 13:53:12.110 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting MySenecWebPortal... for user: '***' with options: None
2023-11-28 13:53:12.110 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(self) ********
2023-11-28 13:53:12.246 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.260 seconds (success: True)
2023-11-28 13:53:13.118 WARNING (MainThread) [custom_components.senec.config_flow] Could not connect to mein-senec.de with '***', check credentials (exception)
2023-11-28 13:53:13.119 ERROR (MainThread) [custom_components.senec.config_flow] Could not connect to mein-senec.de with User '***', check credentials
2023-11-28 13:53:13.295 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.308 seconds (success: True)
2023-11-28 13:53:14.249 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.263 seconds (success: True)
2023-11-28 13:53:15.275 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.289 seconds (success: True)

marq24, 2023-11-28:
I miss in the log output from the  'pysenec_ha'... after the `restarting MySenecWebPortal... for user: 'xxx' with options: None` - I "miss" some lines about the used aiohttp lib... have you now forced the 3.9.x version ?!

GregorEstebanCardosoSchiller, 2023-11-28:
But maybe the last sentence may result in different behaviour within HA...

[image]

marq24, 2023-11-28:
please do not try to manually update the aiohttp lib... 

with the additional output from the log - can you please call (in your browser): https://mein-senec.de/endkunde/api/context/getEndkunde

marq24, 2023-11-28:
this line is the root cause of the issue...

https://github.com/petretiandrea/plugp100/blob/248726a3ed0ce99504e6c7a6dcf00cba7ded9470/plugp100/common/utils/http_client.py#L24C12-L24C12

thanks for nothing! - this code is purging all existing cookies of a session

closed 2023-11-29 by marq24

marq24, 2023-11-29:
after I wrote 'there is nothing I could do'... I had an idea, that I COULD DO something... 
please update to https://github.com/marq24/ha-senec-v3/releases/tag/3.1.2 and let me know, if this solves the situation for you...

GregorEstebanCardosoSchiller, 2023-11-29:
First great appreciation for your dedication and persistence ;-) A Spam for every second of the tapos triggering the cookie purge:

2023-11-29 13:05:42.580 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.myenergi_hub_11357557_power_charging not found.
2023-11-29 13:05:42.583 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.585 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.587 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.589 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.591 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.594 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.596 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.598 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.629 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.642 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.665 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.673 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.679 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.684 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.686 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
[... cut ...] ill keep at least the mein.senec.de cookies! but any other integration might break cause of this call!
2023-11-29 13:05:42.850 WARNING (MainThread) [custom_components.senec.pysenec_ha] CookieJar.clear() have been called without ANY predicates! This would PURGE all cookies (IMHO nobody should do that inside a home assistant integration) - We will keep at least the mein.senec.de cookies! but any other integration might break cause of this call!

But it works. I´ll comment the line in your code....

marq24, 2023-11-29:
with the next update you will get the warning only every 24h...