marq24/ha-senec-v3 #44: [BUG] WebAPI returned values are sometimes wrong

AndreWohnsland, 2023-11-21:
Starting today, the WebAPI seems to have some issues, which result in the values returned from the API are not reasonable. In cases of units in total kWh, they should usually increase monotonically, but some values returned by the API seem to be "shifted" into the past values:

[image]
[image]

This result in the energy dashboard going quite wild, because due to the great difference in values, the calculated energy is wrong and too large:
[image]

This is hopefully just a one time issue of the API, but it would be nice to catch this behavior somehow in the integration.
One way to prevent this IMHO, could be using the previous value stored in HA, if the current value is lower than the past one.

I am quite fluent with Python, but I have never touched any HA Integration up to now, so I don't know if this is "simple" fixable. But as long as you can access the previous value in the logic of the `update_full_kWh_stats` function, I think a simple sanity check (value larger or equal to the current one) can fix this issue for the total values.

Kind Regards
Andre

marq24, 2023-11-21:
Hi - just as short note (while I am bussy with other stuff) - while fetching (new) data you do not have access to previous values - IMHO that's a task of the recorder... 

It might be (quite easy) possible to tell the sensor, that it's accepting only increaing (aka larger then before) values - but i have to check... 

and BTW - thanks Senec!

marq24, 2023-11-21:
> while fetching (new) data you do not have access to previous values
previous values already stored in HA (i though that was obviously)... 

> IMHO that's a task of the recorder...
When I understood the concept of HA correctly, then there are sensors and then there are (data) recorders... sensor's a stupid - they have a current state - that's it - they don't now anything about previous values or anything like that... It's the job of the recorder to "persist" sensor data... and so IMHO the recorder is the only instance in HA that would have knowledge about previous values...

A quick google search did not return anything valuable for me - [You might like to try the google results yourself](https://www.google.com/search?q=home+assistant+only+record+data+if+value+is+higher)...

I was hoping for a property if a sensor entity that would allow to specify, if the value is only increasing... but looks like that this is not existing...

At least I can tell you, that "my" data from senec is still fine and does not have any drops like yours (but might be that's because I am just query it ever 5 minutes)...  [I know, this doesnot help you, when I say "Works fine in my office" - but at least this is not a general issue]

marq24, 2023-11-21:
For me it's a bit late today - can you please be so kind and read this section here: https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics and let me know, if you believe that setting `state_class=total` and `last_reset=None` could solve the issue?!

What I read, this would not solve it ?!

AndreWohnsland, 2023-11-21:
Reading _"For sensors with state_class total_increasing, a decreasing value is interpreted as the start of a new meter cycle or the replacement of the meter."_, this would even be worse for this current situation, since it would think it is reset somehow and then adds the difference (which is none but interpreted due to the jump) to the sensor value.
"_It is important that the integration ensures that the value cannot erroneously decrease in the case of calculating a value from a sensor with measurement noise present._" <- this currently does happen, so I think it would not solve it. 

Would it also be somewhat possible to change this function to get the old value, which is not there at the initial start but should be existing once the class got initiated and run once the data fetching. Or does HA creates the class eacht time again? 

```python
class MySenecWebPortal.
   # this is the data which is problematic
    async def update_full_kWh_stats(self):
        # grab TOTAL stats
        a_url = f"{self._SENEC_API_URL_END}" % str(self._master_plant_number)
        for key in self._API_KEYS:
            api_url = self._SENEC_API_URL_START + key + a_url
            async with self.websession.get(api_url) as res:
                try:
                    res.raise_for_status()
                    if res.status == 200:
                        r_json = await res.json()
                        value = r_json["fullkwh"]
                        entity_name = str(key + "_total")
                        # CHANGE: check here if  self._energy_entities[entity_name] exists and <= value
                        # use old value instead if the new is lower
                        prev_value = self._energy_entities.get(entity_name)
                        if prev_value is not None and prev_value > value:
                            value = prev_value 
                        self._energy_entities[entity_name] = value
                    else:
                        self._isAuthenticated = False
                        await self.update()

                except ClientResponseError as exc:
                    if exc.status == 401:
                        self.purgeSenecCookies()

                    self._isAuthenticated = False
                    await self.update()
```

This assumes that `self._energy_entities[entity_name]` persists after multiple fetches of the data, so the `MySenecWebPortal` class persists after initiated. This is the part where I have no clue how HA is working :(

marq24, 2023-11-22:
> This assumes that self._energy_entities[entity_name] persists after multiple fetches of the data, so the MySenecWebPortal class persists after initiated.

Your assumption is correct...  and also your suggested addon/change looks fine for me... There is only one thing I don't like... When the class will be (re)initialized (e.g. cause of a HA restart) you run in exactly the same situation... you receive a value and you don't have a single clue, if this is "valid" (aka larger then previous value)... So this feels like a temporary "Pflaster" and IMHO not really addressing the issue in a proper way... On the other hand you can argue - "hey that's better then nothing" (which is also true)... 

I would like to come back to `sensor.state_class` for sure total_increasing is not the right one to use... [but in the docu it's said](https://developers.home-assistant.io/docs/core/entity/sensor/#how-to-choose-state_class-and-last_reset):

> It's recommended to use state class total without last_reset whenever possible, state class total_increasing or total with last_reset should only be used when state class total without last_reset does not work for the sensor.
> Examples
> - The sensor's value never resets, e.g. a lifetime total energy consumption or production: state_class total, last_reset not set or set to None

So this sounds to me like an appropriate approach - but then you should not use the sensor.state (as value) anylonger - then you should use sensor.sum_increase ?! At least that's what I understand from the doc... You might need an additional template sensor, that capture the sensor.sum_increase from the original kWh sensor as it's state to use it then in the energy dashboards?

Disincentivato, 2023-11-22:
[image]
[image]

Good morning,
this happen to me for all the webapi total entities.
Now my personal stats from grafana are totaly incorrect.. :(

AndreWohnsland, 2023-11-22:
> Your assumption is correct... and also your suggested addon/change looks fine for me... There is only one thing I don't like... When the class will be (re)initialized (e.g. cause of a HA restart) [...]

Yes, this occurred to me as well, that the initialization can still be wrong. I also don't like this part. If the problem persists, it will happen at some point also at the init, it's just a band-aid.

I think it's also important to investigate why the value is wrong, I don't think this is a thing on HA or the integration side, because it did work fine and broke without any changes, but I still would somewhat verify that "Senec fucked up once again" (like man, why is their product software side as well as physically filled with flaws). Today, I am busy with work, but I will see what I can do the next days.

> So this sounds to me like an appropriate approach - but then you should not use the sensor.state (as value) anylonger - then you should use sensor.sum_increase ?! At least that's what I understand [...]

In theory, I think this is exactly the sensor we want. But as I read somewhere (even if its set to never reset) the value provider needs to take care that the value never really resets. Or a reset is not counted as one, but then just added. e.g. 1000 → 1010: +10, 0: nothing happens, the sensor seems to reset. 0 → 10:  +10, new value is 20. So the measurement never decreases, but if our reference value makes the jump from decrease to increase again, the jump is also added to the value, if I understand the docs correctly. This was made in case you switch your measurement unit, and it starts once again at a lower value. At least that's how I do understand the setting. I did some additional research, but up to now, could not find examples how to check that the value does not decrease from an integration dev POV.

For now, I've disabled the API integration because my Influx + retention policy don't like manually deletion of data. How (and even if it's possible without removing the sensors) to fix the energy dashboard, I need to figure out. One wrong day is not an issue, but the huge values make week / month / year data false as well, but that's another problem for me 🙈.

Currently, it was no longer possible (since the http > https change) to get the total values local, or am I mistaken?

AndreWohnsland, 2023-11-22:
Ok, I've used my coffee break to make some API calls myself:

Timestamp `1640991600000` is start of January 1st, 2022, so that's the previous year.
Somehow, this value "jumps", since total value is the sum of all years (`1672527600000` is start of this year), the sum gets also the jump. 

Some data of `powergenerated`, I used 4 time points:
t (here is the right value)
2151.87 + 2608.27 = 4760.14 ← correct calc

```json
{
	"lastupdated": 1700607600,
	"fullkwh": 4760.1484375,
	"val": [
		[
			1640991600000,
			2151.8767033219337441
		],
		[
			1672527600000,
			2608.271728515625
		]
	]
}
```

t + 1 (here is the wrong value)
2074.76 + 2608.27 = 4683.10 ← past val change make this change

```json
{
	"lastupdated": 1700607600,
	"fullkwh": 4683.10888671875,
	"val": [
		[
			1640991600000,
			2074.7687321305274951
		],
		[
			1672527600000,
			2608.340087890625
		]
	]
}
```

t + 2 (here is the wrong value)
2074.76 + 2608.27 = 4683.10, but somehow full is back to the "right" value from above.
I have no idea what the backend is doing.

```json
{
	"lastupdated": 1700607600,
	"fullkwh": 4760.25146484375,
	"val": [
		[
			1640991600000,
			2074.7687321305274951
		],
		[
			1672527600000,
			2608.374755859375
		]
	]
}
```

t + 3 (here is the right value again)
2151.87 + 2608.1 = 4760.29

```json
{
	"lastupdated": 1700607600,
	"fullkwh": 4760.2958984375,
	"val": [
		[
			1640991600000,
			2151.8767033219337441
		],
		[
			1672527600000,
			2608.419189453125
		]
	]
}
```

One idea would be to use a "lifetime" sensor on the last value in the list of `val`, this would only reset on new year and would not have current changes.
But it may also break existing values in the current implementation. 
Also, it should not happen on Senec side that past (no more changing) values do change. Like, how is this even happening.

Disincentivato, 2023-11-22:
Thanks for the support you all are giving on this bug, unfortunately I don't have enough experience in this field to be able to help.
Why did this reading error not appear for some user? Don't we all connect to the same senec link?

marq24, 2023-11-22:
> Why did this reading error not appear for some user? Don't we all connect to the same senec link?

Because the root cause is located at Senec's Servers - Senec generate/provide false/invalid data - the integration is just the "person who is delivering the bad news"...

bobthebuilder4711, 2023-11-22:
You can even see this yourself by going to https://mein-senec.de/endkunde/#/0/status/gridexport/all and refreshing the page. For me right now as I'm trying this the total jumped a few times between two values.

---

> When the class will be (re)initialized (e.g. cause of a HA restart) you run in exactly the same situation

Shouldn't you be able to get the current value of a sensor even after a reboot? Or if that goes to "unavailable" after a reboot: Is there no way to get the last value update?

AndreWohnsland, 2023-11-22:
> Shouldn't you be able to get the current value of a sensor even after a reboot? Or if that goes to "unavailable" after a reboot: Is there no way to get the last value update?

AFAIK, sensors start with unavailable after a start until they poll the current value from the “physical” device. In this case it’s the api but you d need to get data once. I don’t know if you can access the recorder data (where the last value should be stored in the Ha database). If anyone got a suggestion, feel free to offer :)

bobthebuilder4711, 2023-11-22:
If there is no function of some recorder class then you should be able to get the value via sql query from the statistics_short_term table with a `WHERE NOT state = 'unavailable' LIMIT 1` at the end.

And I just asked if this is happening with the Home V4 too. There doesn't seem to be jumping for the V4.

AndreWohnsland, 2023-11-22:
Well, it have to be possible, looking at the build in [counter](https://github.com/home-assistant/core/blob/dev/homeassistant/components/counter/__init__.py#L232C48-L232C48) and [here](https://community.home-assistant.io/t/main-purpose-of-restoreentity/135270/5) using async_get_last_state from inheriting from the RestoreEntity, this could be working.

marq24, 2023-11-22:
... since I am under pressure @ work - you can modify the code locally - and if you have a working (fix) solution - create a PR and I would be happy to roll it out... But I have to admit after 10h @ work spending my time with creating swagger content I am not in the best mood to fix Senec's crap

bobthebuilder4711, 2023-11-22:
I'm still using the modified script from mstuettgen and simply added that after the webgrabber has pulled the values from the website and stored them in a dictionary, said dictionary is copied to dictionary_old and is also saved to a file. That way on reboot you can simply load the file into dictionary_old and can always compare the new values with the values in dictionary_old and replace them with the values in dictionary_old if the new ones are lower.

I just checked and it seems that you can read and write files in integrations: https://community.home-assistant.io/t/is-it-possible-to-write-read-text-file/15214/3

Sounds a bit dirty but should enable you guys to create a fast fix for this.

Disincentivato, 2023-11-22:
> > Why did this reading error not appear for some user? Don't we all connect to the same senec link?
> 
> Because the root cause is located at Senec's Servers [they have a history of inpredicatble/inconsistent data] - Senec generate/provide false/invalid data - the integration is just the "person who is delivering the bad news"...

what I wanted to say is that there are users with the same integration on HA set equal and have the correct data. I don't understand what causes this incorrect reading.

marq24, 2023-11-22:
> what I wanted to say is that there are users with the same integration on HA set equal and have the correct data. I don't understand what causes this incorrect reading.

the issue is caused by FALSE data generated at SENEC.de webservice - the integration just read the provided data - aka "shit in - shit out"... and not all senec.de users are affected (e.g. I am currently does not have this issue at all) - so the integration istelf works flawless - the data source is BAD/INCORRECT.

Disincentivato, 2023-11-22:
[image]
[image]
[image]
I'm not shure if it can help, the wrong data coming out from senec are always those of 2022 in my case.

Disincentivato, 2023-11-22:
>  I'm not shure if it can help, the wrong data coming out from senec are always those of 2022 in my case.

[image]
[image]

More specific May data appears and disappears.

marq24, 2023-11-22:
... so the time-filter @ senec will not applied correctly - well this does not help "me" (the integration) - this info would help senec developers...

AndreWohnsland, 2023-11-22:
Sadly, I have also been busy with other programming task. 

But I had some time to look into it, the `state` or `_state` just represents one state of an entity not some larger dict of the whole component (like I thought you could store it). TBH, all the files and class struct still confuses me a bit, even after reading through the dev docs.

Could one use the `SenecEntity` or `SenecSensor` class in the `custom_components/senec/sensor.py` file to implement the saving and restoring of the previous state over inheritance with the `RestoreEntity` and according `async_get_last_state` in this entity class? So would this reflect to each entity in the `MySenecWebPortal` class, or how are the `@property` annotated methods handled and converted by HA?

marq24, 2023-11-22:
... since 3h I am trying to understand how the heck HA is storing data in the sqlite db... and when - "my" problem is, that I want to read the last stored value (from db) in a moment when the entities are not "initialized"... 

Even if I know, how to get the Entities from the registry (so I get all properties of the configuration)... the hass.states is empty in this moment (and only will be filled after the MySenecWebPortal have completed the initialization (which is already too late)

bobthebuilder4711, 2023-11-22:
Simply storing the last received values in a file is no option? Senec is likely going to fix this and then you can remove it again.

Change these two functions and you're done:

```
    def __init__(self, user, pwd, websession, master_plant_number: int = 0):
        # here goes whatever else is usually in this function:
        #....
        
        # loading old energy entity values:
        self._old_energy_entities_copy_exists = False
        self._energy_entities_old = {}
        
        sFileName_values_init = '/config/custom_components/senec/last_total_values.json'
        try:
            with open(sFileName_values_init, 'r') as f:
                self._energy_entities_old = json.load(f)
                self._old_energy_entities_copy_exists = True
        except:
            logger.info("File with old values doesn't exist")
```

```
    async def update_full_kWh_stats(self):
        # grab TOTAL stats
        a_url = f"{self._SENEC_API_URL_END}" % str(self._master_plant_number)
        for key in self._API_KEYS:
            api_url = self._SENEC_API_URL_START + key + a_url
            async with self.websession.get(api_url) as res:
                try:
                    res.raise_for_status()
                    if res.status == 200:
                        r_json = await res.json()
                        value = r_json["fullkwh"]
                        entity_name = str(key + "_total")
                        self._energy_entities[entity_name] = value
                        
                        if self._old_energy_entities_copy_exists == True:
                        if self._energy_entities[entity_name] < self._energy_entities_old[entity_name]:
                            self._energy_entities[entity_name] = self._energy_entities_old[entity_name]
                            
                        self._energy_entities_old = self._energy_entities.copy()
                        self._old_energy_entities_copy_exists = True
                        
                        sFileName_values = '/config/custom_components/senec/last_total_values.json'
                        with open(sFileName_values, 'w') as f:
                            json.dump(self._energy_entities_old, f)
                    else:
                        self._isAuthenticated = False
                        await self.update()

                except ClientResponseError as exc:
                    if exc.status == 401:
                        self.purgeSenecCookies()

                    self._isAuthenticated = False
                    await self.update()
```

marq24, 2023-11-22:
yes that would work @bobthebuilder4711 - but I have to admit, that I don't "like" to write a file to the local FS each time the sensor will be updates... -> if this is the only workaround, then I would like to write the file in the "finalize" area (so when the HA will be shut down, that then the file will be written... but at the end of the day It's more the challenge, that I would like to understand, how actually HA persistente is working :-)

bobthebuilder4711, 2023-11-22:
If HA crashes then the file will never be written. So you should at least write it every 5 to 10 minutes or so.

And there are so many sensors constantly writing their data to the database, which is sitting in the local file system as well. So a few bytes more every few minutes hardly matter.

AndreWohnsland, 2023-11-22:
@marq24 Your dev branch got some more commits than master. Should I submit a PR on `develop`? 

I did extend `SenecSensor`  with the `RestoreEntity`, using `async_get_last_state` and also checking state against prev one. Only thing I don't like that much is that this entity is for used for the other values as well. Currently, I only apply the sanity check for sensor values containing "_total" in their description key.

marq24, 2023-11-22:
yes - please on develop

marq24, 2023-11-23:
> @marq24 Your dev branch got some more commits than master. Should I submit a PR on `develop`?
>
> I did extend `SenecSensor` with the `RestoreEntity`, using `async_get_last_state` and also checking [...]

Thanks so much - I solved the challange by allowing the SensorDecription config to "enabled" the new behaviour per Sensor...

Disincentivato, 2023-11-23:
> please update to https://github.com/marq24/ha-senec-v3/releases/tag/3.0.9-rc1
> 
> have in mind - in HACs you have to enable the "beta" switch for this integration...

Hi Thanks a lot, how can I enable the beta version? Do I need to uninstal it from HACs?

marq24, 2023-11-23:
> Hi Thanks a lot, how can I enable the beta version? Do I need to uninstal it from HACs?

https://hacs.xyz/docs/faq/beta/

guevara777, 2023-11-24:
I have an NPU-Error (No IP given) on my Senec Home v2.1. May this be related to this update. Is anyone else having this error?

AndreWohnsland, 2023-11-24:
Currently, I have the impression the senec software is even worse than ever. My storage started an emergency charge from 19-22 o’clock yesterday from grid electricity. Also, past days it did not charge often when excess sun was there, such things could also be Hardware issues. But that’s a whole other topic, and shouldn’t be discussed in this issue.

 It is very unlikely that Ha Integration (which usually just read the values) affect your devices.

scottgoo, 2023-11-24:
@marq24 Thank you so much for your great work, for your quick responses to fill the holes the senec guys dig one after the other!!!

marq24, 2023-11-24:
I guess this is solved - if somebody disagree - please reopen

closed 2023-11-24 by marq24

Disincentivato, 2023-11-24:
Hi and thanks again. I created a sensor using the helper and the Riemann Integration. I think that by doing in this way the webapi is no longer necessary. it is also quite precise and uses local W data to calculate kWh. The two screenshots compared the energy data, one with webapi and the other is the sensor created with the Riemann from the master.house.power.
[image]

AndreWohnsland, 2023-11-24:
Hi @Disincentivato, while this approximation comes near to the real value (I also did it once this way), I really still prefer to have the actual, real value. Because there are some limitation and edge cases where this may also fail terribly. For example, the IT infrastructure to my battery is not that stable and may lose connection for short time (minute) sometimes.
In general, due to the option to set the resolution to 1s, it can come quite close to the real value, tough. So if you got a stable connection, this should also work fine.

BMeyendriesch, 2023-11-26:
I'm sorry, but even after installing 3.1.0, the problem persists for me: The kWh values from the WebAPI fluctuate between two values at intervals of 18kWh. If only the upper or lower value were used, everything would be plausible. But this way it messes up the entire energy panel :-(

[image]

marq24, 2023-11-26:
@BMeyendriesch can you also share the "long term" stats of HA itself - might be that the "fix" is not working for influxDB?!

BMeyendriesch, 2023-11-26:
Voilà

[image]

marq24, 2023-11-26:
thanks! (grr - this really "sucks" (did I mess up the build?!)... let me get another coffee first...

marq24, 2023-11-26:
can you please enable the debug log (of the integration) and restart your ha? and then share the log output - TIA

BMeyendriesch, 2023-11-26:
[home-assistant_2023-11-26T12-19-33.360Z.log](https://github.com/marq24/ha-senec-v3/files/13466721/home-assistant_2023-11-26T12-19-33.360Z.log)

marq24, 2023-11-26:
```
ModuleNotFoundError: No module named 'custom_components.senec-20231023.system_health'
...
ModuleNotFoundError: No module named 'custom_components.senec-20231023.logbook'
...
ModuleNotFoundError: No module named 'custom_components.senec-20231023.media_source'
...
ModuleNotFoundError: No module named 'custom_components.senec-20231023.silabs_multiprotocol'
...
ModuleNotFoundError: No module named 'custom_components.senec-20231023.diagnostics'
```
what are all these `ModuleNotFoundError` ??? - and what is the `senec-20231023`?! - should be only `senec` - do you use some private sources ?!

BMeyendriesch, 2023-11-26:
I'm very sorry! I just realized it may have been my mistake: I had accidentally left a backup directory named senec-20231023 in custom_components so that I could quickly go back if the patch to #44 didn't work. I have now removed it. I will report whether the error has now disappeared.

BMeyendriesch, 2023-11-26:
I guess that's it! The error is gone after I removed my backup directory and rebooted. I'm very embarrassed to have taken up your valuable time. And on a Sunday too. Asche auf mein Haupt!

marq24, 2023-11-26:
Kaum macht' man's richtig - schon funktionierst :-D 

> And on a Sunday too...

that's totally fine... I spent my time anyhow with ha/frigate/go2rts/reolink configuration stuff...