marq24/ha-senec-v3 #46: While Emergency Charge is running battery charge and discharge are 0

iot-sle, 2023-11-28:
seem that only those both values are not filled in. I attached a screenshot showing corrent charge currents for the cells and on right hand a screenshot of "solectrus" using lala.cgi too. Which is showing the correct battery charging power. On upper right corner you can see status is "Notladung" 
[image]

marq24, 2023-11-28:
I am sorry - but I actually do not understand, what you are trying to ask/say...

Can you please be so kind and describe the problem/issue in one sentence? - What is he current behavior - what is the expected behavior - what is causing the behavior? 

TIA

iot-sle, 2023-11-28:
I took a short look at your code, by the way thank you for your great work, the problem is line

BAT_STATUS_CHARGE = {4, 5, 8, 10, 11, 12, 14, 43, 71}

my status is 33: Saftey charge

marq24, 2023-11-28:
You are right - there had been multiple states that have been missing - 23, 24, 25, 33

THANKS for your report - a new build/version is just around the corner

marq24, 2023-11-28:
https://github.com/marq24/ha-senec-v3/releases/tag/3.1.1

closed 2023-11-30 by marq24