marq24/ha-senec-v3 #47: Add or show some more data in the energy dashboard by manipulating HA's SQLite-DB

GregorEstebanCardosoSchiller, 2023-12-01:
### Discussed in https://github.com/marq24/ha-senec-v3/discussions/17

<div type='discussions-op-text'>

<sup>Originally posted by **loeffelpan** September 10, 2023</sup>

Second is a powershell script to create sql statements with api data from senec for adding that data to your energy dashboard.
You can get the script [here](https://gist.github.com/loeffelpan/10bdb7e5bfd21568a9cb05d429aaf214).</div>

So I get from the script the following output, which makes no sense for me, can you assist? To me it seems, that it finds everything (mode ADD), but in the end it says the state or sum is not in db?

[image]

loeffelpan, 2023-12-10:
I will take a look later this day. Sorry for my late response.

loeffelpan, 2023-12-10:
You provided "$first" and "$last" with hours. But as we're trying to replace whole days with data from api, this should be just dates like "01.08.2021" and "01.11.2023".
I added some error handling for this case and improved logging in tha latest revision of my [script](https://gist.github.com/loeffelpan/10bdb7e5bfd21568a9cb05d429aaf214). Please try again.

marq24, 2023-12-11:
would you mind, if i convert this into a discussion (since IMHO this is not really a issue with the Integration) - ok?