marq24/ha-senec-v3 #53: Total entities not increasing

loeffelpan, 2023-12-10:
Even after fix of #44 my energy dashboard does not look correct. The wrong values are gone, but the total sensors are not increasing anymore although there are values from now sensors.

[image]
[image]

marq24, 2023-12-10:
please have a look here: https://github.com/marq24/ha-senec-v3/discussions/48

loeffelpan, 2023-12-10:
Thanks. I did search in issues, but not in discussion.

closed 2023-12-10 by loeffelpan

marq24, 2023-12-10:
NP :-) - so you have also the "thanks for nothing" message in your log - right?!