marq24/ha-senec-v3 #56: Two WEB.API Integrations on the same Home Assistant

ml1nk, 2023-12-11:
I added two Web.API Integration on the same Home Assistant with different logins and displayed some of the data in a dashboard. Some values were identical and after some time (and probably a sync) all values were identical, which can't be true.

Is it possible, that that es retrieval isn't completely separated and is interfering with each other?

It is my first try to work with the Home Assistant, so it could be some glitch somewhere else in my configuration.

Version: 3.1.3

marq24, 2023-12-11:
Hi Marcel,

actually I have to admit, that I don't know the answer to your question - I would expect, that with the current design of the (web) Integration that's not possible to access two different accounts on the same HA instance.

For me there is no way to test/debug/investigate this - since I just have a single account... Have that said, with the integration it's possible to access different systems (aka AnlagenNummer) - but then this different systems are all bound to a single mein-senec.de account.

I don't know your use-case (why there are multiple mein-senec.de logins/accounts) in your scenario... For sure it will be possible to build/implement the integration in a way, that the code could handle this scenario... But (once again) this would be really really special...

ml1nk, 2023-12-11:
Hi Matthias,

no idea why we needed two separate accounts in this case, but the form we filled to senec was rejected because of an duplicate e-mail address. The use-case is probably when different households are using one instance of home assistant together. Ironically it isn't possible to use two energy dashboards until now and like you said, it is a rare case.

Either way i will try to fix this, because it is rather strange that different integrations influence each other and looking up the code there is no separate cookie jar per integration so it shouldn't be too hard.

The Account which is currently connected is working perfectly, thanks for this plugin :+1:

marq24, 2023-12-11:
Thanks for your explanation Marcel - indeed using one HA instance for two (Independent) households is not really common - you obviously have your reasons for this setup - but I honestly would consider installing two docker instances... But that's a total different story.

I think what you need to adjust in the code is, that there are dictionaries in which the data is stored (from the web-api requests) - this data containers have to be enriched to use e.g. the email/user as key and then storing the data...

PR #58 (2023-12-13): fix: use async_create_clientsession instead of async_get_clientsession to prevent cookie leakage
This bugfix replaces all occurrences of async_get_clientsession with async_create_clientsession and as such prevents any cookie leakage between entries. If auto_cleanup couldn't be used (connection test), it is detached inside a finally block.

Looking up [aiohttp_client](https://github.com/home-assistant/core/blob/dev/homeassistant/helpers/aiohttp_client.py) the session returned by the function "async_get_clientsession" seems to be at least bound to the whole component if not the whole thread. Thi [... cut]
closed 2023-12-13 by marq24