marq24/ha-senec-v3 #62: Partially wrong cycle values

electricdelicate, 2023-12-19:
Hi there,

I`ve noticed the module based cycles ist partially wrong. My Senec 2.1 have 3 modules and all installed at the same day. Till 1 or 2 months all modules have more or less the same cycle number. Now has especially module B only 1 cycle. What ist happened?
[image]

marq24, 2023-12-23:
mhhh - strange - I must be day-dreaming from time to time... 

Sorry I thought I had already replied to your issue - Please be so kind and check the data that //vars.html is reporting for your system... Here I have almost the same situation - implausible/inconsistent values for cycles of my four battery modules... my ModuleB is even stuck at '0'... 

So the integration just provide the information that is returned by the device - and I agree looks like the devices does not have a idea what's going on inside.

BMeyendriesch, 2023-12-27:
My SENEC integration can no longer read any values from the BMS. Even deleting the integration and then reinstalling it does not bring up the values again. 
[image]

marq24, 2023-12-27:
then please be so kind an open a new issue... TIA

closed 2024-01-13 by marq24