marq24/ha-senec-v3 #63: Showing Sockets

patricknitsch, 2023-12-20:
Hi,

i have a V2.1 with technical packet. So i have two configurable sockets.

Can you implement values for the sockets?

Many thanks

marq24, 2023-12-21:
Good Morning Patrick,

well - IF the lala.cgi of your device will return data for these "sockets", then the answer is "yes I can" (but you have to provide way more details) -> e.g. share the output of the https://[YOUR-SENEC-IP]//vars.html 

If this "socket" data will be only available via the mein-senec.de - then the answer is also "yes" - but then the process is way more complicated... And since I do not have a 2.1 and also no "sockets" it will take quite long... If you are familiar with command line tools like CURL, then the process could be much faster... but still time consuming.

patricknitsch, 2023-12-21:
Guten Morgen, 

habe gesehen, du kommst ja aus Deutschland. 

Ich habe dir ein Bild der Variablen für die Sockets angehängt.
[image]

Wenn du weiter Informationen benötigst, unterstütze ich gerne. Mit ein paar Anweisungen kann ich auch mit Befehlen herausfinden, was du benötigst

marq24, 2023-12-21:
ok - das sieht erstmal "einfach" umsetzbar aus... kannst Du mir das auch als TEXT (und nicht als Photo) zukommen -> ggf einfach speichern als HTML? - dann kann die Namen/Felder einfach COPY & PASTEN...

marq24, 2023-12-21:
... sehe gerade ich hab' die auch ;-)

patricknitsch, 2023-12-21:
Ich glaube mein Techniker hat mal gemeint, das is ne Option, die einfach nur frei geschaltet wird.

marq24, 2023-12-21:
So since I have started to implemented them - in the meantime - what the heck this sockets are doing?!

patricknitsch, 2023-12-21:
Das sind zwei potentialfreie Schaltkontakte, die bei PV Überschuss für eine gewisse Zeit schalten. Darüber läuft mein Boiler.

Unten siehst du ein Bild, wie das auf dem Webserver aussieht

[image]

marq24, 2023-12-21:
also SG-Ready

patricknitsch, 2023-12-21:
Kannst du so ausdrücken. Nur dass ich keinen speziellen Heizstab habe, sondern nur einen normalen Speicher

marq24, 2023-12-21:
Leider ist das mit den Schaltern nicht ganz so einfach für mich... Wenn Du auf eine der Checkboxen clickst [z.B. Permament an] und dann speichern sagst, dann macht die WEB Oberfläche ein POST request zu Deinem Speicher...

Diesen POST Payload müsste ich mal haben :-/

patricknitsch, 2023-12-21:
Kann ich es irgendwie ausprobieren?

marq24, 2023-12-21:
Ja Du müsstes das bei Dir lokal machen - gelichzeitig vom Browser die Developertools aufmachen und eben im Netzwerk-Tab "genau" den "POST" request rausuchen... dort findet man denn auch unter den Headern die "post-data"...

Wenn die diese ganzen Begriffe nichts sagen ist das völlig OK - das alles Entwickler-Gefasel (und für den normalen User schwer verdaulich)...

Wenn es aber in einer ERSTEN Version ausreichend ist, diese Felder zu LESEN (und eben nicht zu schreiben), dann hab' ich's erstmal ;-)

patricknitsch, 2023-12-21:
Ich habe mal reingeschaut. Die post-data habe ich aber nicht gefunden. Hast du noch einen Tipp für mich, wo ich suchen kann?

[image]

marq24, 2023-12-22:
Du musst auf den "Nutzlast" tab schauen...
[image]

Wenn Du die Checkbox angeklickt hast und (ich vermute mal erst wenn Du den Save button drückst), dann gibt es leider auch nur einen weiteren Request an das lala.cgi... wonach wir suchen ist der Request, bei dem "ein Wert" auf irgendwas gesetzt ist... Hier in meinem Beispiel (und das ist leider das einzige was ich hier machen kann) wird das SAFE_CHARCE_FORCE auf `u8_01` gesetzt...

Normalerweise sind die Werte immer auf (leer) also `""` gesetzt - dann weiss der Senec "dazu sollen werte ausgeliefert werden"... Wenn Du also den Request findest, bei dem die Nutzlast Werte enthält die nicht `"FELD_NAME":""` sind, dann bitte noch den Tab auf "View Source" umschalten - alles markieren und hier reinpacken... 

Idealerweise für jede Checkbox und die Eingabefelder die Du setzen kannst getrennt (also immer nur ein Wert modifizieren)...

patricknitsch, 2023-12-22:
Ich denke ich habe da was. Unter Anweisung kann ich arbeiten :)

Schau mal ob dir das ausreicht. Habe auch die Oberfläche mit drauf, damit du die einzelnen Variablen besser zuordnen kannst. 

Wenn du noch etwas brauchst, kannst du gerne bescheid geben.

[image]

marq24, 2023-12-22:
nee - leider nein - Du guckst Dir gerade die Antwort an - dort stehen dann eben die Wert VOM Senec zu DIR an... 

Du musst im "Nutzdaten" Tab suchen....

patricknitsch, 2023-12-22:
Neuer Versuch...

[image]

patricknitsch, 2023-12-22:
[image]

marq24, 2023-12-22:
... damit kann ich wohl was anfangen... -> kannst Du das nochmal mit werten für Stunde/Minute machen?

patricknitsch, 2023-12-22:
Gerne. Diesmal für beide Kontakte

Kontakt 1:
[image]

Kontakt 2:
[image]

marq24, 2023-12-23:
Ich habe noch einen Wunsch - könntest du bitte bei einem Request mal "Quelle Anzeigen drücken und den Text der dann angezeigt wir mit der Maus COPY & PASTEN? - Danke
[image]

patricknitsch, 2023-12-23:
{"ENERGY":{"ZERO_EXPORT":""},"SOCKETS":{"ENABLE":"","USE_TIME":"","FORCE_ON":"[\"u8_00\",\"\"]","TIME_REM":"","POWER_ON":"","UPPER_LIMIT":"","LOWER_LIMIT":"","TIME_LIMIT":"","POWER_ON_TIME":"","PRIORITY":"","SWITCH_ON_HOUR":"","SWITCH_ON_MINUTE":"","NUMBER_OF_SOCKETS":"","ALREADY_SWITCHED":""}}

marq24, 2023-12-23:
"Ja iss denn heut scho' Weihnachten?"

patricknitsch, 2023-12-23:
Das ist bei dem trostlosen Wetter da draußen definitiv wie Weihnachten. Absolut geil!! Funktioniert alles. Dafür werde ich dir für deine Mühen zu Weihnachten einen Kaffee spendieren :)

closed 2023-12-24 by marq24