marq24/ha-senec-v3 #64: webAPI wrong values

h4nnes, 2023-12-22:
Hi, 
Since today I get some weird values from the webAPI.  
E.g.: senec_web_gridimport_total = 12.954,96 kWh, but the value from the app and also from mein-senec is 12.884,84 kWh

Any idea why I have this wrong values with my V3 hybrid duo?

Thanks!
Hannes

bobthebuilder4711, 2023-12-23:
Since a few weeks ago Senec likes to let the total values jump up and down. That's why a fix was introduced so that the values are only accepted by the integration if they're higher than the last recorded values.

But that also means that if the values jump up and then down to the correct numbers then the integration won't accept that and will stay on the high numbers until mein-senec.de has caught up.

I guess the integration needs some kind of plausibility check that it also won't accept values that are too high if they're not doable by the device.

h4nnes, 2023-12-23:
Hi @bobthebuilder4711 ,
Oh I see. Thanks for explanation! But that’s really bad. I’m using this values for my energy dashboard and with this jumping values it’s not reliable :(

marq24, 2023-12-23:
@h4nnes please blame Senec (which seams to be the right default reaction)... HoHoHo

h4nnes, 2023-12-23:
@marq24,
Yes sure! Not blaming someone here! I really appreciate your work!

Senec: No comment…

Merry Christmas all!

marq24, 2023-12-23:
I did not take anything from you as complain - no worries... I am start thinking about the suggestion from @bobthebuilder4711 if it would be possible to check for "implausible" increments as well - but this would (IMHO) imply to predict the expected increments over time AND predict the future creativity of Senec developers... :-)

GregorEstebanCardosoSchiller, 2023-12-23:
Maybe the best course of action would be to introduce a "prove of concept" riemann sum automatically with the integration, that sums up the values of the "now" parameters. If they defere too much from the "total" values provided by the API (difference daily) the prompt by the API should be ignored.

marq24, 2023-12-23:
> Maybe the best course of action would be to introduce a "prove of concept" riemann sum automatically with the integration, that sums up the values of the "now" parameters. If they defere too much from the "total" values provided by the API (difference daily) the prompt by the API should be ignored.

I am looking forward  to your PR ;-)

I have to admit that in my professional carrier (+40 years) I have wrote just to many lines of code in order to "correct" things other devs should have made "just right" - That's why I really like open source - if there is a bug/issue, you can fix it at the source [just like I did it with the incorrect cookie handling in the aio-http lib - now everybody will benefit from that].

This would make any kind of artificial patch framework obsolete - which anyhow will generate also false-positives for sure.