marq24/ha-senec-v3 #66: "Enable/Disable discharge" in Web UI of my Senec battery

loeffelpan, 2023-12-29:
I'm a tibber customer since a couple of week. And to use energy from cheap hours during less cheap hours I told homeassistant in an automation to charge the senec battery from the grid depending on a tibber price entity. For that I turn on the switch of your integration.

Now I'm facing some edge cases when I would like to charge the battery during cheap hours e. g. in the early morning and use that energy in the afternoon. Unfortunately the energy from the battery will be mostly consumed during the day.

I found something in the maintenance section of the web ui.
I think the left "button" is, what we do when manually charging. What about the right one?
Is it possible add a feature which enables/disables discharging?

[image]

marq24, 2023-12-29:
Looks like you are in a "lucky" position :-) I am Tibber customer since 15.10.2023 - and had exactly the same challenge.

Enable the Entity called 'switch.senec_safe_charge' and you are done... (of course use this at your own risk - as everything)... 
[image]

It take few seconds till this switch have an effect - but "enable" will do the same as pressing the "Vollladung" in the UI - and disable it again will so the same as pressing the "Entladen freigeben"...

I have few remarks/hints to share:
1. You can not "just keep the current battery level as it is" (so you can not disable the battery-discharge - there is only the option to request the battery to be loaded till it's full (aka 100% - well since we are talking about Seneic genius here - it's only 70%) - so no option to "disconnect" the battery... You only have the option to "LOAD" (till it's full) or to "RELEASE" the power

2. I also started with an two automations (trying to find the cheapest price during night and another one during the day) - while this is working ok - I find it much simpler to start with battery loading if current price is below daily average price and there are less then 6 hours ahead (where the price is lower then avg)

PS: I have to ask - did you already used a friend-invite-code with your tibber account? - if not please be so kind and use mine (My Tibber invitation Link: https://invite.tibber.com/6o0kqvzf (Code: six, oscar, zero, kilo, quebec, victor, zulu, foxtrot) - So we both get 50,-€ each) - and make sure that you also get the ha-tibber-pulse-local Integration :-)

marq24, 2023-12-29:
There is no way to "just keep the current loading state of the battery" (without recharging it)... that's the CRAP we bought... 

So my "solution" is, that I keep the "load battery" switch ON during the day (when the price is ~avg) and release it shortly before the price is some sort of expensive...

My code is far away from being optional - since currently I do not consider the possible power available from PV - but since I live in Ostwestfalen-Lippe it's anyhow just raining from Oct till end of April

loeffelpan, 2023-12-29:
> There is only the option to request the battery to be loaded till it's full (aka 100% - well since we are talking about Seneic genius here - it's only 70%) - so no option to "disconnect" the battery... You only have the option to "LOAD" (till it's full) or to "RELEASE" the power
What do you mean with "it's 70% only"`?
I'm trying to load until 70% because there are nearly no loss of charge until there. After that the charging power reduces and some of the energy from the grid is lost.
And again what a CRAP to show two buttons, if the left one is a toggle?!

> PS: I have to ask - did you already used a friend-invite-code with your tibber account? - if not please be so kind and use mine (My Tibber invitation Link: https://invite.tibber.com/6o0kqvzf (Code: [...]
Tibber Pulse is already in da house with 50€ discount. I took a code from "Kleinanzeigen".
What are my advantages when I use any Tibber integration? I just need the price sensor (from the cloud). I know senecs local/api data are sometimes messed up, but consumption in homeassistants energy data is calculated anyhow (based on the other sensors). So what should I do with local comsuption data?

> So my "solution" is, that I keep the "load battery" switch ON during the day (when the price is ~avg) and release it shortly before the price is some sort of expensive...
Seems to be the only way. I will try and calculate again what is the less lossy solution.

marq24, 2023-12-29:
> What do you mean with "it's 70% only"`?

To my best knowledge all Senec batteries have been set to a max of 70% only - if you are not affected, then you are lucky... In any other case you might had already forgot ;-)

> And again what a CRAP to show two buttons. So the left one seems to be a toggle and the right one is just for disabling?

You can try yourself (and observe the related payload that will be sent to the device)... 

For me the frustrating part was/is, that there is also a 'li_storage_mode' - that can be also enabled/disabled (in the same way as the 'safe_charge' - I was hoping that this switch would allow to disconnect the battery... But looks like that this feature have a total different meaning/functionality - but at least in my setup toggling this additional switch has no effect. [the only thing that I observed is, when disable the storage mode the device is going to a interal system check before returning to normal operations]

> What are my advantages when I use any Tibber integration? I just need the price sensor (from the cloud). I know senecs local/api data are sometimes messed up, but consumption in homeassistants energy data is calculated anyhow (based on the other sensors). So what should I do with local consumption data?

So when you have already alternative data sources for your house consumption (grid import & export) then the Senec, then you are right, then additional pulse data might have no additional value... 

I have no local access to the data of the main power-meter of my house - so the tibber pulse have value... [I would like to access the ABB EnFluRi directly - but I am clueless how this can be done]

> Seems to be the only way. I will try and calculate again what is the less lossy solution.
You might like to share your results as discussion? - would be cool

loeffelpan, 2023-12-31:
When I'm fine with my result I will share it as a discussion.

closed 2023-12-31 by loeffelpan

bobthebuilder4711, 2024-01-04:
> To my best knowledge all Senec batteries have been set to a max of 70% of their full capacity only

Only those with certain battery pack versions. Our V2.1 - installed in 2019 - never had the 70% limitation.

The charging behavior is different though. Prior to Senec disabling all power stations it would charge with 2.5kW from 0% to 100%. Now it will only charge with 1kW until around 10% then go up to 2.5kW until around 70% and then slowly go down with the power until 100%.

marq24, 2024-01-04:
thanks @bobthebuilder4711 for sharing!

loeffelpan, 2024-01-04:
Similar with our V3 hybrid duo battery from end of 2020. Never had limitation.
But the actual speed depends on the installed modules / the capacity of the battery
We have only two modules installed - 5 kWh in total. I think @bobthebuilder4711 is talking about four modules with 10 kWh capacity.

Here is my graph of a charge from 0 to 100%.
Before any upgrade we had about 1.5 kW constantly.

[image]

bobthebuilder4711, 2024-01-04:
Yes, the 2.5kW is for four modules with the V2.1.

btw:

If you really want to control how your senec power station charges and discharges then someone needs to find out how the Enfluri presents its data via Modbus. That way a raspberry pi for example could act as a Modbus master for the Enfluri and as a Modbus slave for the senec power station and change the values of grid import and grid export so the power station  charges or discharges how you want it to. Or even hold its charge by simply setting import and export values to zero.