marq24/ha-senec-v3 #67: actual values from WEBAPI no longer available

SteffenS27, 2024-01-01:
Hi,

is it "normal" that the ...now values are not longer supported?

Best regards

marq24, 2024-01-01:
well I can't confirm that - here all *_now values are available...

`{'accuimport_now': 0.00292, 'accuexport_now': 0.0, 'gridimport_now': 0.531432, 'gridexport_now': 0.0, 'powergenerated_now': 0.0, 'consumption_now': 0.537272}`

SteffenS27, 2024-01-01:
Tried it again, in the option WEPAPI only the following is visible
[image]

marq24, 2024-01-01:
what do you get, when you access mein-senec.de via your browser? [https://mein-senec.de/endkunde/#/0/status] do you have in the overview "Momentanwerte" ?! - I guess not...

SteffenS27, 2024-01-02:
Curiosly everything:
[image]

marq24, 2024-01-02:
krasse Sch*****

This overview webpage use the same request as the WebAPI-Integration - so it's quite surprsing that you see data there... (but the integration is not able to fetch data from this

`https://mein-senec.de/endkunde/api/status/getstatusoverview.php?anlageNummer=0`

You might like to check, if the entitie IDs might have changed? - At least I just can let you know, that at least for _me_ everything works as normal. I am not sure if you would like to remove and re-add the Web-API Konfigutaion in order to check, if this would solve your issue...

Also you might like to try (before that) to enable the DEBUG-Log for the integration in order to check, if there are any Exception's reported in the HA-Log? [if there are any, then it's worth sharing them]

SteffenS27, 2024-01-02:
Ok - I deleted and re-added the WEBAPI Option . That is the result in the DEBUG and even it is the same result - all sensors are there but the ...now values are not available:

> AttributeError: 'SensorEntityDescription' object has no attribute 'array_key'
2024-01-02 20:21:33.755 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up senec platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 369, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 509, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 754, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1025, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 745, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 845, in _async_write_ha_state
    state, attr = self._async_generate_attributes()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 786, in _async_generate_attributes
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 751, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/senec/sensor.py", line 100, in state
    if self.entity_description.array_key is not None:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SensorEntityDescription' object has no attribute 'array_key'
2024-01-02 20:21:45.611 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.269 seconds (success: True)

marq24, 2024-01-02:
thanks - so you are on the BETA-Channel - sorry that this did not came to my mind... a patch will be available in few minutes

SteffenS27, 2024-01-02:
Thanks a lot ! Values are back again

closed 2024-01-02 by marq24