marq24/ha-senec-v3 #69: Error requesting senec data: 504, message='Gateway Time-out'

RikyUnreal, 2024-01-11:
Hi all,
today the integration starts to show this message in the log:
Error requesting senec data: 504, message='Gateway Time-out', url=URL('https://192.168.xxxxxxx/lala.cgi')

Is it my issue or Senec changed something in the inverter firmware?

marq24, 2024-01-11:
Can you access the log of your senec hardware? [or just go the hardware itself] - probably there is a error displayed... All what the Integration is doing is to send requests via https to the IP - if the hardware does not reply via a valid http response there is not much the integration can do...

marq24, 2024-01-11:
possible things you can check/do:
https://github.com/marq24/ha-tibber-pulse-local/issues/3
or this longer here:
https://github.com/marq24/ha-tibber-pulse-local/issues/6

RikyUnreal, 2024-01-11:
I'm not at home at the moment, and all I can say is that I can see the inverter working as usual from the official Senec app on my smartphone.
I'll update you when I get home later today.

RikyUnreal, 2024-01-11:
Ok Matthias, false alarm.

I'm at home and confirm that the local web interface was also stuck, so I rebooted the inverter and everything is working again.

closed 2024-01-11 by RikyUnreal