marq24/ha-senec-v3 #70: Battery module data missing since some time

loeffelpan, 2024-01-14:
Some of the module sensors are not longer available in my instance.
I'm not using them but I thought this is worth to raise an issue.

[image]

marq24, 2024-01-14:
Since a couple of releases the integration checks, which sensors are active - and when there are sections (in your case the 'BMS') that will not be used at all, then the integration does not request the data from your SENEC device - you could simply check this enable at least one bms_current_* sensor and verify that after a short while data will be available... If yes - please close this issue - if no I would kindly ask to provide the debug log output from you HA - TIA

loeffelpan, 2024-01-14:
Nevermind.
I just found that - for any reason - I have some _2 entities from your integration.
Those without that suffix are obviously orphaned. Sorry for bothering you.

closed 2024-01-14 by loeffelpan