marq24/ha-senec-v3 #71: New NPU with old Firmware

ironbiff, 2024-01-17:
A technician today changed my NPU and so I got beneath a new MAC Address also the old NPU Firmware back.
Now I am back on 2011 instead of 2310 and my interface is back on http, which broke again your great integration.

Do somebody know a way to push the system to geht the firmware updates fast and @marq24 maybe you can give me a hint to make your adapted integration to speak http again as workaround? Or does anybody know, that I might get the firmware updates already this night again?

Thanks in advance

marq24, 2024-01-17:
Hi - the simplest is to remove the integration and re-add it - during the initial setup the code will check, if HTTP is working... and use it when it's available... as long as you keep the initial names for the integrations all "should" be fine (famous developers words)... 

And no clue how fast the UPDATE will be rolled out - quite surprising, that they do not flash them before installation...

ironbiff, 2024-01-17:
thx a lot and especially for how very fast your response was

closed 2024-01-20 by marq24

vmpr, 2024-07-02:
thanks @marq24 - I also got a new NPU today with the same problem that HTTPs doesn't work anymore, only HTTP - 

so I edited this file in home-assistant and set the value for `use_https` from true to false. after restart of HA the integration works again :) That could be an idea for an FAQ section in your repo?

```
/config/.storage/core.config_entries
line:

      {
        "entry_id": "090eb63729087c6ec4fe96c12b1f47b1",
        "version": 1,
        "minor_version": 1,
        "domain": "senec",
        "title": "senec",
        "data": {
          "name": "senec",
          "host": "10.13.37.100",
          "use_https": false,
          "scan_interval": 10,
          "type": "senec",
          "dtype_int": 19,
          "dtype": "SENEC Main-Unit",
          "dname": "SENEC.Home V3 hybrid | SENEC.Inverter V3 LV",
          "dserial": "XXX",
          "version": "App:0826 FW:2307 NPU-Image:2310(v10)"
      ...
      },
```

marq24, 2024-07-02:
> thanks @marq24 - I also got a new NPU today with the same problem that HTTPs doesn't work anymore, only HTTP -
> 
> so I edited this file in home-assistant and set the value for `use_https` from true to false. after restart of HA the integration works again :)

When you are able to edit the config_entry set `use_https` to false, then there is no need to adjust the integration code at all... since if this value is false, the code will use http... [have in mind, as soon as the firmware get's autoupdated, you need to revert this again]...

Still crazy, that they replace hardware and do not ensure that the latest firmware is installed...

vmpr, 2024-07-03:
I was hoping after the firmware update HTTP would still be working plus HTTPs on top.  let's see what happens :) yeah, I guess the technician's job would be to replace the part and trigger the firmware update but it was close to the end of the day and so he just went home. who knows.