marq24/ha-senec-v3 #72: binary_sensor.senec_wallbox_1_smart_charge_active reser after about 3 minutes

Schlolle, 2024-01-21:
Whenever I activate the smart charge mode for my SENEC- wallbox, the entity "binary_sensor.senec_wallbox_1_smart_charge_active" will be set to "on" for only a short time and then reset to "off" though the smart charge mode ist still avtive. After that, it is not possible in HA to determine, which charge mode (quick or smart) ist active, because the pop-up menu of "select.senec_wallbox_1_mode" is also reset.

marq24, 2024-01-21:
First of all thank you very much for your feedback... 

When I understood you correctly: In HA you switch the "Wallbox mode" from OFF to OPTIMIZED... this implies that the smart_charge_active sensor will set to ON... 

After a short while the HA display witch back to MODE: OFF and smart_charge_active OFF ?

If this is the case, then it looks like, that the required SYNC via the WEB-API does not work - can you please enable the DEBUG-Log for the integration and restart HA and then share the output of the LOG?

marq24, 2024-01-21:
and just to repeat - you MUST have configured the additional WebAPI-Integration to use the new Wallbox feature

Schlolle, 2024-01-21:
WebAPI-Integration is integrated indeed. Energy Dashboard is working correctly. I try to publish the log after it is generated

marq24, 2024-01-21:
Eine "idee" hätte ich... es scheint, als sei über den Jahreswechsel bei SENEC das ein oder andere WebZertifikat abgelaufen zu sein... ich hatte dieses Wochende so meine Erlebnisse als ich die neue "BETA" Version implementiert habe - dort greife ich auch auf die APP-API zurück (so wie bei der Wallbox Steuerung auch)... Wenn da also im Log bei Dir irgendwelche "could not validate certificate" Exceptions im log auftauchen... Denn gleich mal ein Update auf die aktuelle beta version machen - dann könnte das Problem schon behoben sein!

Schlolle, 2024-01-21:
Hallo Matthias,

 

das habe ich on der LOG gefunden:

 

2024-01-21 17:28:42.323 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved

Traceback (most recent call last):

  File "/mnt/dietpi_userdata/homeassistant/custom_components/senec/pysenec_ha/__init__.py", line 2928, in update

    await self.authenticate(do_update=True, throw401=False)

  File "/mnt/dietpi_userdata/homeassistant/custom_components/senec/pysenec_ha/__init__.py", line 2878, in authenticate

    async with self.web_session.post(self._SENEC_WEB_AUTH, data=auth_payload, max_redirects=20, ssl=False) as res:

  File "/home/homeassistant/.pyenv/versions/3.11.5/lib/python3.11/site-packages/aiohttp/client.py", line 1187, in __aenter__

    self._resp = await self._coro

                 ^^^^^^^^^^^^^^^^

  File "/home/homeassistant/.pyenv/versions/3.11.5/lib/python3.11/site-packages/aiohttp/client.py", line 629, in _request

    raise TooManyRedirects(

aiohttp.client_exceptions.TooManyRedirects: 0, message='', url=URL('https://mein-senec.de/auth/login')

 

Ich denke, das geht in die Richtung?

 

Werde versuchen die BETA zu installieren.

 

Melde mich dazu dann.

 

 

Mit freundlichen Grüßen

 

Ulrich Otto
Lindenbühlstraße 23
78727 Oberndorf a.N.
Telefon: 07423/1677
Telefax: 07423/1689
Mobil: 0179/5020359
E-Mail: ***@***.***

marq24, 2024-01-21:
das sieht mir nach "kaputtem" CookieHandling aus - Du sagst, dass die WebAPI Sensoren aber Daten liefern? [sieht irgendwie für mich nicht so gut aus]... aber wie es auch sei - ein Update auf die aktuellste BETA (2024.0.3) könnte helfen...

Schlolle, 2024-01-21:
Hallo Matthias,

 

leider besteht das Problem auf die neueste Beta noch gleich. Ich kann die Wallbox in HA einstellen:

Wenn man das in der SENEC-App verfolgt, sieht man unmittelbar die Änderung.

 

Nach 1-2 Minuten sieht es in HA dann wieder so aus:

 

Das Ergebnis in der SENEC-App ist aber dasselbe. D.h. dort bleibt alles eingestellt, wie es war. Das LOGFILE hat ebenfalls noch dieselbe Meldung.

 

LG

 

Ulrich Otto
Lindenbühlstraße 23
78727 Oberndorf a.N.
Telefon: 07423/1677
Telefax: 07423/1689
Mobil: 0179/5020359
E-Mail: ***@***.*** ***@***.***>

marq24, 2024-01-21:
Leider kommt hier nichts von dem an (attachments/screenshots) die Du in eine eMail Antwort steckst... das geht nur direkt hier in GitHub...

Wenn auch die aktuellste Beta version nicht wie gewünscht arbeitet, dann benötige ich leider den output, den man in dem HA-Log file findet (wenn man für die Integration den DEBUG modus aktiviert)... DEBUG Modus aktivieren - einmal den Modus von AUS auf OPTIMIERT schalten - ein - zwei Minuten warten und dann den DEBUG-LOG Modus wieder ausschalten - beim ausschalten (über einen Webbrowser), wird dann auch gleich das LOG file vom Backend mit ausgeliefert... "das" würde ich gerne haben...

Schlolle, 2024-01-21:
Hallo Matthias,

 

die Fehlermeldungen aus der Log mit dem Zertifikat traten seltsamerweise nun nicht mehr auf. Es geht nun auch länger, bis sich in HA der Smart-Charge-Status der „zurücksetzt“ (wie gesagt, in der SENEC-App bleibt trotzdem alles unverändert). Ich habe nun die DEBUG-Meldungen aus der Log gefiltert und als Datei angehängt. Vielleicht bringt das etwas mehr Erkenntnis?

 

LG

marq24, 2024-01-21:
wenn Du das File direkt 'hier' (https://github.com/marq24/ha-senec-v3/issues/72) GitHub postet - und __nicht__ als eMail Antwort sendest, dann erreicht mich das log file vielleicht auch ;-)

Schlolle, 2024-01-22:
[home-assistant_senec_2024-01-22T07-46-09.775Z.log](https://github.com/marq24/ha-senec-v3/files/14006740/home-assistant_senec_2024-01-22T07-46-09.775Z.log)

OK, hier das Logfile. Ich deaktivierte die Wallbox, startete dann die debug-Protokollierung, stellte dann die Wallbox auf "Optimiert" und wartete ca. 3 Minuten, bis HA den Schalter "binary_sensor.senec_wallbox_1_smart_charge_active" wieder zurückgesetzt hatte. Wie bereits erwähnt: die Wallbox bleibt korrekt eingestellt, was auch in der SENEC-App nachvollziehbar ist. Lediglich HA zeigt den Staus dann nicht mehr korrekt an. ich weiß auch nicht, ob es vorgesehen ist, dass der Schalter "select.senec_wallbox_1_mode" dauerhaft den eingestellten Wallbox-Modus anzeigen soll. Dort wird spätestens nach Verlassen und zurückkehren auf die Seite mit der SENEC-Erweiterung nichts  angezeigt.

marq24, 2024-01-22:
Danke für das LOG! - Ich verstehe zwar immer noch nicht was bei Dir da genau passiert - aber Dein HA Installation läuft m.E. alles andere als Rund...

hier ein paar Einträge, um die Du dich m.E. kümmern solltest/müsstest
```
2024-01-21 17:50:29.023 ERROR (SyncWorker_8) [homeassistant.util.package] Unable to install package pyatv==0.14.3: error: subprocess-exited-with-error
...
homeassistant.requirements.RequirementsNotFound: Requirements for apple_tv not found: ['pyatv==0.14.3']
...

2024-01-21 18:08:31.902 ERROR (MainThread) [homeassistant.helpers.entity] Failed to set state for sensor.audi_q3_tfsi_e_longterm_trip_data, fall back to unknown

...
2024-01-21 18:08:31.677 ERROR (MainThread) [custom_components.audiconnect.audi_connect_account] Unable to obtain the vehicle shortTerm tripdata of [VIN-HERE]: 403, message='', url=URL('https://mal-1a.prd.ece.vwg-connect.com/api/bs/tripstatistics/v1/vehicles/[VIN-HERE]/tripdata/shortTerm?type=list&from=1970-01-01T00:00:00Z&to=2024-01-21T18:38:31Z')
...
```
Vermutlich ist das "alles" diese 'audiconnect' Erweiterung...

Hier mal zusammengedampft was bei Dir passiert...

```
 746: 2024-01-21 18:18:40.543 DEBUG posting data: {'WALLBOX': {'PROHIBIT_USAGE': ['u8_00', '', '', ''], 'SMART_CHARGE_ACTIVE': ['u8_00', '', '', '']}}
 747: 2024-01-21 18:18:40.828 DEBUG post result: {'WALLBOX': {'PROHIBIT_USAGE': [0, 0, 0, 0], 'SMART_CHARGE_ACTIVE': [0, 0, 0, 0]}}
 748: 2024-01-21 18:18:40.829 DEBUG ***** APP-API: app_set_wallbox_mode(self) ********
 749: 2024-01-21 18:18:40.829 DEBUG ***** APP-API: app_post_data(self) ********
 750: 2024-01-21 18:18:40.829 DEBUG APP-API post {'mode': 'FASTEST'} to https://app-gateway-prod.senecops.com/v1/senec/anlagen/243940/wallboxes/1
 751: 2024-01-21 18:18:41.838 DEBUG APP-API HTTP:200 for post {'mode': 'FASTEST'} to https://app-gateway-prod.senecops.com/v1/senec/anlagen/243940/wallboxes/1
 752: 2024-01-21 18:18:43.840 DEBUG ***** APP-API: app_get_wallbox_data(self) ********
 753: 2024-01-21 18:18:43.840 DEBUG ***** APP-API: app_get_data(self) ********
 754: 2024-01-21 18:18:43.841 DEBUG APP-API get https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
 755: 2024-01-21 18:18:45.163 DEBUG APP-API 2sec after mode change: local set_ic_max 6.0 will be updated to 31.98
 756: 2024-01-21 18:18:45.164 DEBUG posting data: {'WALLBOX': {'SET_ICMAX': ['fl_41FFD70A', '', '', ''], 'MIN_CHARGING_CURRENT': ['fl_41FFD70A', '', '', '']}}
 757: 2024-01-21 18:18:45.177 DEBUG post result: {'WALLBOX': {'SET_ICMAX': [31.979999542236328, 0.0, 0.0, 0.0], 'MIN_CHARGING_CURRENT': [31.979999542236328, 0.0, 0.0, 0.0]}}
	
	
 791: 2024-01-21 18:20:06.382 DEBUG posting data: {'WALLBOX': {'PROHIBIT_USAGE': ['u8_00', '', '', ''], 'SMART_CHARGE_ACTIVE': ['u8_03', '', '', '']}}
 792: 2024-01-21 18:20:06.398 DEBUG post result: {'WALLBOX': {'PROHIBIT_USAGE': [0, 0, 0, 0], 'SMART_CHARGE_ACTIVE': [3, 0, 0, 0]}}
 793: 2024-01-21 18:20:06.399 DEBUG ***** APP-API: app_set_wallbox_mode(self) ********
 794: 2024-01-21 18:20:06.400 DEBUG ***** APP-API: app_post_data(self) ********
 795: 2024-01-21 18:20:06.400 DEBUG APP-API post {'mode': 'SMART_SELF_GENERATED_COMPATIBILITY_MODE'} to https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
 796: 2024-01-21 18:20:07.202 DEBUG APP-API HTTP:200 for post {'mode': 'SMART_SELF_GENERATED_COMPATIBILITY_MODE'} to https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
 797: 2024-01-21 18:20:09.205 DEBUG ***** APP-API: app_get_wallbox_data(self) ********
 798: 2024-01-21 18:20:09.205 DEBUG ***** APP-API: app_get_data(self) ********
 799: 2024-01-21 18:20:09.205 DEBUG APP-API get https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
 800: 2024-01-21 18:20:09.506 DEBUG APP-API 2sec after mode change: local set_ic_max 31.98 will be updated to 6.0
 801: 2024-01-21 18:20:09.507 DEBUG posting data: {'WALLBOX': {'SET_ICMAX': ['fl_40C00000', '', '', ''], 'MIN_CHARGING_CURRENT': ['fl_40C00000', '', '', '']}}
 802: 2024-01-21 18:20:09.520 DEBUG post result: {'WALLBOX': {'SET_ICMAX': [6.0, 0.0, 0.0, 0.0], 'MIN_CHARGING_CURRENT': [6.0, 0.0, 0.0, 0.0]}}

1146: 2024-01-21 18:24:13.371 DEBUG posting data: {'WALLBOX': {'PROHIBIT_USAGE': ['u8_00', '', '', ''], 'SMART_CHARGE_ACTIVE': ['u8_00', '', '', '']}}
1147: 2024-01-21 18:24:13.666 DEBUG post result: {'WALLBOX': {'PROHIBIT_USAGE': [0, 0, 0, 0], 'SMART_CHARGE_ACTIVE': [0, 0, 0, 0]}}
1148: 2024-01-21 18:24:13.667 DEBUG ***** APP-API: app_set_wallbox_mode(self) ********
1149: 2024-01-21 18:24:13.667 DEBUG ***** APP-API: app_post_data(self) ********
1150: 2024-01-21 18:24:13.668 DEBUG APP-API post {'mode': 'FASTEST'} to https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
1151: 2024-01-21 18:24:14.443 DEBUG APP-API HTTP:200 for post {'mode': 'FASTEST'} to https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
1152: 2024-01-21 18:24:16.445 DEBUG ***** APP-API: app_get_wallbox_data(self) ********
1153: 2024-01-21 18:24:16.445 DEBUG ***** APP-API: app_get_data(self) ********
1154: 2024-01-21 18:24:16.446 DEBUG APP-API get https://app-gateway-prod.senecops.com/v1/senec/anlagen/[ID-HERE]/wallboxes/1
1155: 2024-01-21 18:24:16.804 DEBUG APP-API 2sec after mode change: local set_ic_max 6.0 will be updated to 31.98
1156: 2024-01-21 18:24:16.805 DEBUG posting data: {'WALLBOX': {'SET_ICMAX': ['fl_41FFD70A', '', '', ''], 'MIN_CHARGING_CURRENT': ['fl_41FFD70A', '', '', '']}}
[... cut ...]
    - Zeile 757: SET_ICMAX': [31.979999542236328
    - Zeile 802: SET_ICMAX': [6.0
    - Zeile 1157: SET_ICMAX': [31.979999542236328
    und dann wird wieder auf OPTIMIZED umgestellt - aber:
    - Zeile 1809: APP-API 2sec after mode change: NO CHANGE! - local set_ic_max: 6.0 equals: 6.0

Also eigentlich steht ICMAX auf 32A - und wenn auf OPTIMIZED umgestellt wird, müsste ICMAX wieder auf 6 gestellt werden... aber "irgendwas" läuft da nicht gerade aus - die Integration meint, der Wert steht schon auf 6 ?!

Schlolle, 2024-01-22:
>>Also eigentlich steht ICMAX auf 32A - und wenn auf OPTIMIZED umgestellt wird, müsste ICMAX wieder auf 6 gestellt werden... aber "irgendwas" läuft da nicht gerade aus - die Integration meint, der Wert steht schon auf 6 ?!<<

Genau so ist es aber auch. Stelle ich aus "Schnell", dann steht ICMAX auf 31A, bei "Optimiert" auf 6A.

Schlolle, 2024-01-22:
Hallo Matthias,

 

ich habe den thread wohl aus Versehen geschlossen ☹ Wollte noch etwas bemerken: Die Einstellung von HA bzgl. Lademodus von SENEC wird direkt im Senec-Speicher lokal geändert und parallel über das Web-Api auch bei Senec so hinterlegt. Dieser Status wird nach 300 Sek. Bei Senec neu abgefragt und nicht oder falsch beantwortet. Ich habe das daran gemerkt, dass ich den optimierten Lademodus direkt in der Senec-App umgestellt habe. Dann bekommt HA den Status ohnehin nur vom Senec-Server. Das „binary_sensor.senec_wallbox_1_smart_charge_active“ bleibt von vorneherein „off“.

 

LG

 

Ulrich Otto

marq24, 2024-01-22:
Also Du sagt, wenn Du den Status über die SenecAPP änderst, dann wird der Status in HA nach einer gewissen Zeit (Backend-Synchronisation) auch korrekt übernommen & angezeigt [das ist ja die "default" Funktionalität]...

Wenn Du die Änderung, über die HA-Integration vornimmst, dann werden [im lokalen Speicher und über die App-API] die Werte "korrekt" in der SenecApp als auch im Speicher angezeigt [das ist eben die "neue" Funktionalität - und in den LOGs die Du bereitgestellt hast, sieht es eigentlich so aus, als wäre alles soweit korrekt]...

Allerdings (auch wenn SenecApp und HA sauber aussehen) gerät nach der "Default Synchronisation" zwischen Backend und Speicher irgendwas in Schieflage ?! So habe ich Dich jetzt zumindest verstanden?

Schlolle, 2024-01-23:
Ich möchte das Problem nochmals ganz genau definieren: Egal ob ich die Ladeart der Wallbox über HA oder über die SENEC-App einstelle. Die Einstellung erfolgt immer korrekt und bleibt auch so. Einzig der Status der Entität "binary_sensor.senec_wallbox_1_smart_charge_active" spätestens nach ca. stimmt nicht (mehr).
Ausgangssituation: Wallbox ist aus.
Szenario 1: Lademodus wird in HA auf "Optimiert" eingestellt. Dann geht "binary.sensor.senec_wallbox_1_prohibit_usage" von on auf off (korrekt!), der "binary_sensor.senec_wallbox_1_smart_charge_active" (korrekt!) auf on. Nach ca. 3 Minuten aber wieder auf off (ohne dass sich der Lademodus aber ändert).
Szenario 2: Lademodus wird in der SENEC-App auf "Optimiert" eingestellt. Dann dauert es ca. 3 Minuten, bis der Lademodus aktiv ist.  "binary.sensor.senec_wallbox_1_prohibit_usage" geht dann von on auf off, "binary_sensor.senec_wallbox_1_smart_charge_active" bleibt aber unverändert auf off.
Es ist also nicht mehr nachvollziehbar, ob dann aktuell der optimierte oder schnelle Lademodus aktiv ist. Nur dass ein Lademodus aktiv ist da  "binary.sensor.senec_wallbox_1_prohibit_usage" korrekt eingestellt wird.
Meine Schlussfolgerung: Alles funktioniert korrekt, außer die Statusabfrage der Art des Lademodus über das SENEC-Web-API

marq24, 2024-01-23:
kannst Du bitte nach den 3 min einmal auf Deinem Speicher lokal gucken (via https://[LOCAL-IP-HERE]//vars.html ), worauf dann der Wert SMART_CHARGE_ACTIVE steht?! (offensichtlich wechselt das vom Wert '3' auf etwas anderes ?!

die // vor dem vars.html sind wichtig...

iot-sle, 2024-01-23:
Ich habe dir die Ergebnisse der Tests geschickt. Zusammenfassung: "Ladeunterbrechung verhindern" an = 3 aus = 4 
nur bei an (Kompatibilitätsmodus) wird der min. Ladestrom berücksichtigt.

Schlolle, 2024-01-24:
Ich habe die neue BETA gestern Abend noch aufgespielt. Sieht jetzt gut aus! Ich denke, das von mir geschilderte Problem ist damit gelöst. Vielen lieben Dank Matthias!
Noch eine Frage: mit welcher Entität schalte ich in HA im solaroptimierten Modus die "Ladeunterbrechung verhindern" ein und aus? Oder geht das nur in der SENEC-App?
Außerdem kann man in der SENEC-App den Mindest-Ladestrom lediglich von 6 bis 12 A variieren, in HA geht das von 0 - 64 A. Stellt man in HA weniger als 6A ein stellt sich die Box auf 6A ein, stellt man in HA mehr als 12A ein stellt sich die Box auf 12A ein (zumindest zeigt das die App so an). Der dazu gehörende Sensor "sensor.senec_wallbox_1_min_charging_current" zeigt aber dennoch den in in HA eingestellten Mindestladestrom an. Sollte man das nicht auch in HA auf diese Range begrenzen?

iot-sle, 2024-01-24:
Hallo

schalte mal in der App die Ladeunterbrechung aus. Dann übermittelt die App leider nicht nur diesen Status, sondern auch den dort hinterlegten min. Ladestrom an deine Senec Anlage.
Die App wurde von Senec auf 6-12A "kastriert". 6A war schon immer früher ging es bis 16A. Dies ist aber eine Begrenzung in der App. Hast Du als die Ladeunterbrechung ausgeschaltet, dann kannst Du aus HA auch Ströme bis zum max. Wert Deiner Box 16 oder 32 A je nach Box einstellen. Das habe ich überprüft. Das Problem ist, dieser Ladeunterbrechungsstatus lässt sich z.Zt. nur aus der APP ändern und damit kommt automatisch die Limitierung aus der App zum Tragen. Probiere mal erst in der App die Ladeunterbrechung ausschalten, sicherheitshalber 5 Minuten warten bis das von Senec synchronisiert wurde, dann in HA 16A und das funktioniert bei mir. Als ZOE Fahrer habe ich den Ladeunterbrechungsmodus bisher nicht benutzt.

Schlolle, 2024-01-24:
> Das neue Release 2024.0.7 bietet jetzt zwei "Optimierungs-Modi" an - mit und ohne 'Ladeunterbrechung'

Wow. Super! Das ging fix und funktioniert auf den ersten Blick einwandfrei. Auch die Skala des min. Ladestroms ist jetzt von 6 auf 32A begrenzt. Herzlichen Dank für die gute Anpassung!

marq24, 2024-01-24:
also - solved - affe tot - issue zu ;-)

closed 2024-01-24 by marq24