marq24/ha-senec-v3 #73: No connection after upgrade to 2024.0.5

CodingTfC, 2024-01-23:
Today I have updated the Senec-Integration to 2024.0.5 and suddenly all entities are gone :(

I only found this Error-message in logs:

Logger: homeassistant.config_entries
Source: config_entries.py:406
First occurred: 17:43:55 (4 occurrences)
Last logged: 18:18:33

Error setting up entry senec for senec
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 406, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/__init__.py", line 108, in async_setup_entry
    if coordinator.senec.device_type is None or coordinator.senec.batt_type is None:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 183, in device_type
    value = self._raw_version[SENEC_SECTION_FACTORY]["SYS_TYPE"]
            ^^^^^^^^^^^^^^^^^
AttributeError: 'Senec' object has no attribute '_raw_version'

I also added the WebAPI function and this is working fine...

marq24, 2024-01-23:
see #74 - solved with 2024.0.6

closed 2024-01-23 by marq24

Disincentivato, 2024-01-23:
update to the 2012.0.6