marq24/ha-senec-v3 #75: New Status "Aufwachladung"

ETH1166, 2024-01-23:
Dear Community, dear Matthias @marq24,

today I had recognized a new status, which had been shown as "UNKNOWN" in Homeassistant. I am using your HA integration still on version 3.1.3 on HA 2024.1.3 (thus I am running a little late...)

This is a SENEC.home V3 Hybrid duo with the following software version:
REVISION MCU: 1827
REVISION MCU-BL: 2307
REVISION NPU-REGS: 10
REVISION NPU-IMAGE: 2310
REVISION GUI: 1967
I don't know exactly when Senec had installed the last update.

On the Display and in the WebGUI I see a status of "Aufwachladung" in German language.

Can anybody tell me, what this means? I never had seen this before.

The funny thing is, that this translates into "wake up charging", but indeed at the moment of writing the battery is being discharged as normal on a low power demand. Also on the front-panel of the Senec V3 the "Laden" (charging) LED is on, while the battery discharges. What are these gentlemen doing at Senec?

Can I help in adding this new status message to your integration or is it already integrated in version 3.1.4 or 2014.0.1?

I am a little shy to upgrade to 2014.0.1. Is thsi compatible with the 3.1.x series? Sorry for bothering you, but I read somewhere, that there are compatibility issues/considerations. I am using both, the local polling for current data and the WebAPI for total energy.

Thank you very much for your great work and support!

Best regards
Thomas

marq24, 2024-01-23:
Hi Thomas,

I don't have this new version on my V3 Hybrid ... I am still on MCU

So the "new" Status ist not present yet, if you want to help, you can share the content of the following file:
https://[SENEC-IP-HERE]/js/DE-de.js

Concerning your other question "What are these gentlemen doing at Senec?" -> I am clueless - IMHO must be a issue in the upper-management... They might need a new CIO...

ETH1166, 2024-01-23:
Hi Matthias,

you are welcome! I am glad I can help. Thank you for the link. This is obviously a new status #99.

I agree with your statement on the CIO. I guess you would do us all a big favour, if you would volunteer for this Position! ;-)

I really dislike a charging light being on, while the system discharges.

Please find attached the downloaded file.

Best regards
Thomas

P.S.: Would you mind me contacting you personally for a short off-topic Senec chat?

[SenecHybridV3duo-DE-de.txt](https://github.com/marq24/ha-senec-v3/files/14029164/SenecHybridV3duo-DE-de.txt)

ETH1166, 2024-01-23:
Just a comment on this Senec SW-Version, in case people find this interesting: (I hope you donot mind)

I have a emergency power supply added to my Home V3 duo and have set the discharging threshold to 30%, as at the moment the solat power is not sufficient to charge the battery to 70% anyhow. The last weeks discharging stopped at 30%.

Now I had looked at the battery load indicator: it is already at 18% and continues to discharge. The function "Backup Power Pro" seems to be broken by this update. As this is the first day I observe this, the update must have arrived today.

Who knows what else is broken?
Thank you Senec!

io-debug, 2024-01-24:
Hi @ETH1166, @marq24,

I'm using the "Reservekapazität" and the "Backup power pro" and had a similar idea as you @ETH1166.
I've set the "Reservekapazität" to 25% and the functionality of our integration works as expected, since I can see the expected percentage in the SENEC App as well as on https://mein-senec.de/.

Nevertheless I've ovserved two strange behaviours:

1. The reserved battery capacity is below the capacity I've set (in my example below 25%)
Thesis: If you are limited to 70% of your battery capacity ("Konditionierungsbetrieb"), the calculation for the reserved battery capacity ("Reservekapazität") is not based on 100% of your battery capacity.

To be honnest I dont have any clue what SENEC takes into consiteration for the calculation, but sometimes my battery discarted to 22%, sometimes to 18%, even if my reserved capacity is set to 25%.

2.  Sometimes my battery discarges very slowly over a long period of time and this even below my reserved capacity of 25%.

3.  I can set the reserved capacity max. to 50%.

I've started to ovserve the first two points when the "Konditionierungsbetrieb" started.
To say if the 3rd point is normal, I'm missing a propper documentation of the expected functionality by SENEC.

Hope this helps.
(My MCU Version ist also still at 826)

marq24, 2024-01-24:
@ETH1166 can you also please check the italian language file - what's the translation for the new Status 99?

electricdelicate, 2024-01-26:
> Hi Thomas,
> 
> I don't have this new version on my V3 Hybrid ... I am still on MCU 0826
> 
> So the "new" Status ist not present yet, if you want to help, you can share the content of the following file: https://[SENEC-IP-HERE]/js/DE-de.js
> 
> Concerning your other question "What are these gentlemen doing at Senec?" -> I am clueless - IMHO must be a issue in the upper-management... They might need a new CIO...

https://senec.com/de/aurelie-alemany-verlaesst-senec

electricdelicate, 2024-01-26:
Same at me, since 3 days this new message "Aufwachladung" on my Display. Charging only with 600w max. possible. Discharge not possible at all. Buhhh Senec ....

marq24, 2024-01-27:
closed with release of 2024.0.9

closed 2024-01-27 by marq24

ETH1166, 2024-02-01:
Thank you @marq24 !

Sorry for my late reply. Italian language reads:
 99: "CARICA DI RISVEGLIO"

And please ignore my ignorance: I have just updated to version 2024.0.9 and have lost the configuration of the energy dashboard. Did I miss a required manual step by updating from version 3.1.3?

I see the following:
[image]
and
[image]
One of the messages of missing entities comes from the migration from local polling. The second for the WebAPI is new.

I find the "missing" entity: senec.sensor_webapi_powergenerated_total (same name as before) in the "Verlauf" graphics section but not in the energy dashboard. Even after an hour...

Will this recover or did I do something wrong?

Thank you and best regards
Thomas

marq24, 2024-02-01:
Hi Thomas, 

please enable the debug log and restart your HA - after some minutes, disable the debug-logging again (in the GUI) and send me the log file (by eMail)...

TIA

marq24, 2024-02-01:
thanks for the logs... from the output there are two remarks:

1. check your 'sensor.gasverbrauch' - this is the one generating the unit-convert exceptions (so this is not senec related)
2. looks like, that for you the '***** APP-API: get_master_plant_id(self) ********' call is running into an unexpected exception... 

So the simples would be, to remove the WEB-API part and simply re-add it with the same/previous name [then all sensors should keep the internal names] (of course creating a backup before would not hurt - just in case something else fails)...

ETH1166, 2024-02-01:
Dear Matthias @marq24,
that simple? - at least after your advice! Thank you so much for your support!
Energy values are back in HA and I hope they appear soon in the energy dashboard again...
The only tiny detail is, that I have lost the cycle count, which should be originating from the local polling plug-in:
[image]
The configuration shows:
[image]
Do I need to change to the new values sensor.senec_bms_cycles_a_2 instead of former values sensor.senec_bms_cycles_a???
Why are there new "_2" variants?
Best regards
Thomas

ETH1166, 2024-02-01:
Thanks Matthias,
your coding is great, but is there an alternative way, which would not cause the loss of data?  -- Maybe I just keep the "_2" running for now...
The cycle counts seem to be pretty low for me. At least they are "customer friendly". I thought this might be a good idea to document those in case of warranty issues.
The 4 modules are all at 99% health and 30 cycles since June 2022. 30 cycles in about 230 days is a really low number, I agree.

marq24, 2024-02-01:
yes of course you also can just use the _2 ones (and remove the ones marked as no longer available)... and if you really want to keep long time data - make sure you have influx db running... 

but all of the 4 modules have a similar cycle count then? - I have 80, 4, 24 and 48

ETH1166, 2024-02-03:
Yes all have the same cycle count.

I am running a base installation of HAOS, which might be a limitation. I still need to learn about HA...

Can you advise a tutorial on how to upgrade an existing installation including data to Influx DB? I did not find reliable information. Or is this simply not possible?

marq24, 2024-02-03:
I am sorry Thomas - but I did not even try to find, if it would be possible to import data from HAOS to a InfluxDB - I just have both running in docker containers and have configured HA to export data to the InfluxDB (which has been basically 3 or 4 lines of additional configuration in HA)

```
influxdb:
  api_version: 2
  ssl: false
  host: <MY-INFLUX-CONTAINER-IP-HERE>
  port: <MY-INFLUX-CONTAINER-PORT-HERE>
  token: <MY-TOKEN-HERE>
  organization: <MY-INFLUX-ORGA-HERE>
  bucket: <MY-BUCKET-HERE>
  tags:
    source: HA
  tags_attributes:
    - friendly_name
  default_measurement: units
```