marq24/ha-senec-v3 #77: Temperaturschwankung nach Fernabschaltung

fugazzy, 2024-01-26:
Ich nutze den Temperatursensor zur Überwachung der Batterietemperatur.
Eben wurde das System in die Fernabschaltung versetzt.
Seit dem löst mein Sensor aus, weil die Anzeige zwischen 60 und -126 Grad hin und her springt.

marq24, 2024-01-26:
Mahlzeit! - mal kurz über Ursache und Wirkung nachgedacht... 

Diese Integration liefert die vom SENEC Speicher bereitgestellten Daten 1:1 an HA... Wenn "nicht sosehr plausible Werte" in HA ankommen, dann ist die Wahrscheinlichkeit > 0.95, dass die Ursache dieser Genialität bei SENEC und nicht bei diese Integration liegt... Mir persönlich wäre es viel zu aufwendig so ein Verhalten zu implementieren... 

Prüf doch mal, was Dein Speicher gerade so für Werte ausspuckt...

https://[LOCAL-IP-HERE]//vars.html

Wenn da die Temperaturen der Batterie-Module normal sein sollten, dann sei doch bitte so gut und mach a) einen Screenshot und b) aktiviere das Debug log für diese Integration und c) poste das ganze zusammen hier...

marq24, 2024-01-29:
Kann dieser Issue zu? [also ist es so, wie vermutet, dass der Speicher eben schon die 'nicht plausiblen' Werte liefert]

fugazzy, 2024-01-29:
Ja, sorry, die seltsamsten Daten kommen direkt aus dem Senec-System. Hätte ich nicht gedacht.

closed 2024-01-30 by marq24