marq24/ha-senec-v3 #78: how to configure energy Dashboard after 2024.0.9

melotti892, 2024-01-30:
as in the title
how do i let my HA to read the new "sensors"?
the v3 "senec.direct....." are not listed in the option to choose in the fields of the different traking sensor: grid imported/exported, solar production and battery charge/discharge field
i fixed the one "dead" from the previous webapi on the main dashboard i use for personal visual, but the one for the energy tab dashboard seems to be a little more specific
[image]

marq24, 2024-01-30:
> as in the title how do i let my HA to read the new "sensors"? the v3 "senec.direct....." are not listed in the option to choose

why do you think the sensor (your are searching for) must start with "senec.direct..."?! Either you have own internal mapper sensors with these names? - Nevertheless - Total Grid Import/Export is no longer available from the local SENEC device [depending from the MCU version installed]... 

I suggest to check #4 and #11

marq24, 2024-01-30:
Ok - then this is a "different" root cause - can you please describe, what did not work with the WebAPI part (of the integration) after you have updated to 2024.0.9?

When I understand you correct, that it's something to the selected "master" ID... Did you previously used the "auto-detect master" when added the integration before - and after the update this "old" master number did not work?

Please try to describe the situation before the update... and what was after the update... and what you "now" have changed that it's working for you again [might be that I need to provide a different update-strategy in certain situations/setups] ?!

melotti892, 2024-01-30:
i'm shure that you got the answer by yourself
if i remember there wasn't the "select your master id" when i configured it, it worked till today with 2024.0.9 and then it stopped. so i first removed the webapi one, then, found that the dashboard wasnt working, i tried to reinstall the webapi back again and i found the "automatic master", it didn't worked so i choose the 0 one (i remember that it was mine master the 0, i have to double check it but don't know how to
so now everything seems to work

marq24, 2024-01-31:
just reading again - https://github.com/marq24/ha-senec-v3/issues/8#issuecomment-1706208446 and from this information there I assume, that your "auto-detected" Master AnlagenNummer is 1...

But looks like that APP-API will provide main-unit information as '0' - is it possible for you based on the information provided here: https://documenter.getpostman.com/view/932140/2s9YXib2td#4d0f84ac-f573-42e3-b155-9a17cef309ec to make a "manual" request to the following URL:

https://app-gateway-prod.senecops.com/v1/senec/anlagen

and provide the response JSON?

melotti892, 2024-02-01:
sorry for the late response; i didn't get what i have to do inside that first link
and with the second it says "missing autorisation token"

marq24, 2024-02-01:
ok (it's totally fine, since this are nor regular-user-skills) - then let's close this... and hope for a web-developer with senec master/slave setup

melotti892, 2024-02-01:
ok, thank you for your time and patience
see you the next time 😆 
p.s. i checked from the app and compared the value reported, they are the same, even tryed to change to 0/2/3 etc but all of them say "an error occurred" wile config the app
so i guess 1 is the number

closed 2024-02-01 by marq24

mariobon123, 2024-02-23:
I'm having an issue with the energy dashboard. When I'm setting the sensors I can only find the "grid imported" (which is the daily) instead of the "grid imported power"(which is the current consumption). Same thing for the battery charge and discharge. How can I get the data in real time?