marq24/ha-senec-v3 #80: No data from mein-senec anymore

scottgoo, 2024-02-12:
Since february 4th at about 9 pm I don’t get any data anymore by mein-senec. Local data works great. Integration does not show a problem. But there are no numbers shown anymore.

Maybe it happend with the last update. I am on 2024.1.1.

Anyone else with similar issues?

scottgoo, 2024-02-12:
Sorry, I forgot to post the log.

Logger: custom_components.senec.pysenec_ha
Source: custom_components/senec/pysenec_ha/__init__.py:2527
Integration: SENEC.Home ([documentation](https://github.com/marq24/ha-senec-v3/blob/master/README.md), [issues](https://github.com/marq24/ha-senec-v3/issues))
First occurred: 10. Februar 2024 um 12:22:20 (1705 occurrences)
Last logged: 11:51:58

APP-API: Error when handling response '<ClientResponse(https://app-gateway-prod.senecops.com/v1/senec/anlagen) [200 None]> <CIMultiDictProxy('X-Content-Type-Options': 'nosniff', 'X-XSS-Protection': '1; mode=block', 'Cache-Control': 'no-cache, no-store, max-age=0, must-revalidate', 'Pragma': 'no-cache', 'Expires': '0', 'X-Frame-Options': 'DENY', 'Content-Type': 'application/json', 'Transfer-Encoding': 'chunked', 'Date': 'Mon, 12 Feb 2024 10:42:38 GMT')> ' -

It is shortend because of personal data following.

marq24, 2024-02-12:
1'st suggestion - update to the latest version
2'nd suggestion - remove the WEBAPI Integration and add it again...

scottgoo, 2024-02-12:
Thank you! I thought 2024.1.1 ist the latest version. I get no never on hacs even if I turn on beta.

marq24, 2024-02-12:
> Thank you! I thought 2024.1.1 ist the latest version. I get no never on hacs even if I turn on beta.

then there is still 2'nd and 3'rd suggestion to go...

scottgoo, 2024-02-12:
I'll take No. 3 then. ;-)

marq24, 2024-02-12:
check, if the 2024.2.1 will solve this issue... If yes - please close the issue - TIA

closed 2024-02-12 by scottgoo