marq24/ha-senec-v3 #82: my configuration doesent work

ilgio, 2024-03-08:
[image]
[image]
[image]

webapi entity are unknown

marq24, 2024-03-08:
can you please be so kind and provide the debug log?!
https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md

ilgio, 2024-03-08:
> can you please be so kind and provide the debug log?! https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md

I sent you an email

ilgio, 2024-03-08:
Maybe because I have 2 systems connected in cascade, master and slave?
[image]

marq24, 2024-03-08:
yes - this is "multi" Senec device related... Please remove the WebAPI integration and ReAdd a new one - but this time use "master_plant_id" 0 (and not "autodetect")...

ilgio, 2024-03-08:
ok, now work, thx

closed 2024-03-08 by marq24

matzeeg3, 2024-06-03:
i have kind of same issue i am not able to add the slave device. i have running with the master but adding the slave result in an unknown error. may someone can tel me how to add it correctly?

marq24, 2024-06-03:
con you please be so kind and provide DEBUG log output - TIA

matzeeg3, 2024-06-26:
hi sorry for late response.
i try to add "anlagen:id 1" becase this is my master and i recive the following output:
`homeassistant  | 2024-06-26 10:39:58.786 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_tech_data(self) ********
homeassistant  | 2024-06-26 10:39:58.786 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_tech_data(self) ********
homeassistant  | 2024-06-26 10:39:58.786 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_authenticate(self) ********
homeassistant  | 2024-06-26 10:39:58.898 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API: response:{'token': 'xxxx', 'refreshToken': ''}
homeassistant  | 2024-06-26 10:39:58.898 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: get_master_plant_id(self) ********
homeassistant  | 2024-06-26 10:39:59.229 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API response: [{'id': '59226', 'steuereinheitnummer': 'xxxx', 'gehaeusenummer': 'xxx', 'strasse': 'xxx', 'hausnummer': 'x', 'postleitzahl': 'xxxx', 'ort': 'xxxx', 'zeitzone': 'Europe/Berlin', 'wallboxIds': [], 'systemType': 'V2'}]
homeassistant  | 2024-06-26 10:39:59.229 WARNING (MainThread) [custom_components.senec.pysenec_ha] Index: 1 not available in array data: '[{'id': '59226', 'steuereinheitnummer': 'xxxx', 'gehaeusenummer': 'xxx', 'strasse': 'xxxx', 'hausnummer': 'xxx', 'postleitzahl': 'xxxx', 'ort': 'xxx', 'zeitzone': 'Europe/Berlin', 'wallboxIds': [], 'systemType': 'V2'}]'
homeassistant  | 2024-06-26 10:39:59.230 ERROR (MainThread) [custom_components.senec.pysenec_ha] APP-API: could not fetch master plant id (aka 'anlagen:id')
homeassistant  | 2024-06-26 10:39:59.230 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_tech_data(self) ********
homeassistant  | 2024-06-26 10:39:59.230 ERROR (MainThread) [aiohttp.server] Error handling request`
For Privacy reasons i removed the serial numbers and so on....
if i open up the api with the url:
https://mein-senec.de/endkunde/api/context/getAnlageBasedNavigationViewModel?anlageNummer=1
i recive data and "master":true
if i use
https://mein-senec.de/endkunde/api/context/getAnlageBasedNavigationViewModel?anlageNummer=0
i revice the data and master:false

marq24, 2024-06-26:
can you please be so kind and send me the RESPONSE JSON's from both of the requests?

- getAnlageBasedNavigationViewModel?anlageNummer=1 

and

- getAnlageBasedNavigationViewModel?anlageNummer=0

[via eMail]...

matzeeg3, 2024-06-26:
sure what is the base url?

marq24, 2024-06-26:
well - https://mein-senec.de/endkunde/api/context/

matzeeg3, 2024-06-26:
> well - https://mein-senec.de/endkunde/api/context/

oh got damm i posted the link befor ^^ sorry for that to mutch work around.
i send you the output. thanks in advanced
- matze

marq24, 2024-06-26:
So I do believe I have now finally understood your initial question Matze...

There is __no__ way to add a __SLAVE__ device to HA... When you have connected the Master (the Autodetect 'should' work) with HA, the data/values that are returned from these endpoints include also the values from the Slave (at least that's the overall/general assumption)... 

It's correct, that (currently) you can not add 'anlagen:nummer 1' -> since the response from SENEC from the ´anlagen´ endpoint only include a single unit -> this is IMHO a error in the API from SENEC - when you have master & slave then the request for Anage__n__ should return ALL systems - not only one... 

When you use "autodetect" mode, the code should find the correct master (@ analgen nummer 1)

matzeeg3, 2024-06-26:
hi thanks for analyzing this.
so at the moment i don´t have the ability to get the value for each batteriy like %-Loaded and KWH in and out i only recive what the masterunit provide. i hoped with this solution i am able to see the kwh in and out from each battery and not a summerd of both. thanks for you help :)

marq24, 2024-06-26:
you can get data via the LAN-based integration - most of the data is there - per IP of each of your devices - but not the statistic data... see #4

matzeeg3, 2024-06-26:
yes i know i used this before i choose to switch to the cloud solution. the historical data is gone so it is not nice to use this anymore....