marq24/ha-senec-v3 #83: Add Inverter v3 Duo

Roka1997, 2024-03-10:
Hello, I have added the inverter according to the instructions.

Unfortunately, I get the following error message.

[image]

Where could the error ?

marq24, 2024-03-10:
can you please provide the debug log output? https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md

Roka1997, 2024-03-10:
Thanks für answer, here ist the log:

"Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.senec
Quelle: helpers/update_coordinator.py:331
Integration: SENEC.Home (Dokumentation, Probleme)
Erstmals aufgetreten: 22:12:43 (9 Vorkommnisse)
Zuletzt protokolliert: 22:18:10

Error requesting senec data: 404, message='Not found', url=URL('http://192.168.xxx.xxx/lala.cgi')"

I've edit the url with "x"

marq24, 2024-03-11:
looks like you did not select `Internal inverter build into SENEC.Home V3 hybrid/hybrid duo` from the initial selection list when starting to configure the new device - when you did, then I would really need the full DEBUG log as described in my link above...

Roka1997, 2024-03-11:
Hello, I think I have taken the correct entry.

[image]

Sorry for the wrong log, here is the correct one.

[error_log-2.txt](https://github.com/marq24/ha-senec-v3/files/14555954/error_log-2.txt)

marq24, 2024-03-11:
`2024-03-10 22:11:31.388 WARNING (MainThread) [homeassistant.const] ENERGY_KILO_WATT_HOUR was used from senec, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/marq24/ha-senec-v3/issues`

please update to the latest version of the integration - this was fixed in JAN - https://github.com/marq24/ha-senec-v3/issues/68 - we are already on '[2024.3.0](https://github.com/marq24/ha-senec-v3/tree/2024.3.0)'

Roka1997, 2024-03-11:
I have solved it. Many thanks for the Help.

closed 2024-03-11 by marq24