marq24/ha-senec-v3 #85: SENEC integration throws errors since today. Communication issue?

thesmer, 2024-03-18:
Integration wirked for several months like a charm. Today at 13:14 German time it stopped working. The Senec App is also not responding with data. 
Is there any issue with the integration? Is my IP-address blocked? I have not found any other possibility to report or ask on the topic.
My HA is on latest version:
Core
2024.3.1
Supervisor
2024.03.0
Operating System
12.1
Frontend
20240307.0

Error messages from log file:

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

**Error Msg 1:**
Logger: custom_components.senec.pysenec_ha
Quelle: custom_components/senec/pysenec_ha/__init__.py:2582
Integration: SENEC.Home (Dokumentation, Probleme)
Erstmals aufgetreten: 13:14:00 (59 Vorkommnisse)
Zuletzt protokolliert: 18:17:16

APP-API: Calling https://app-gateway.prod.senec.dev/v2/senec/systems/299226/dashboard caused [HTTP 400]: 400, message='', url=URL('https://app-gateway.prod.senec.dev/v2/senec/systems/299226/dashboard') this is SO RIDICULOUS Senec - returning 400 without error code when TOKEN is invalid

**Error Msg 2:**
Logger: custom_components.senec.pysenec_ha
Quelle: custom_components/senec/pysenec_ha/__init__.py:2577
Integration: SENEC.Home (Dokumentation, Probleme)
Erstmals aufgetreten: 15:11:05 (1 Vorkommnisse)
Zuletzt protokolliert: 15:11:05

APP-API: Not found https://app-gateway.prod.senec.dev/v2/senec/systems/299226/dashboard [HTTP 500]: 500, message='', url=URL('https://app-gateway.prod.senec.dev/v2/senec/systems/299226/dashboard')

marq24, 2024-03-18:
Well - if your SenecApp does not return any data, then it will be "hard" for the integration to provide something useful - you can try to provide real DEBUG log output (as described here https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md) to get some additional (might be usefull) insights...

At least I can't confirm that there is a general issue - since at least for my system the webapi is returning data (I know that does not help you much here)...

thesmer, 2024-03-18:
I will provide the data soon. But first thanks for responding so fast. And is it possible, that my IP has been blocked? Can I check that somehow before debugging in the wrong direction.

marq24, 2024-03-18:
... normally in your router you can "request" a reconnect... which typically will result in a new external IP... [also you might like to disconnect your phone from the WLAN and check, if then the APP would/will provide data again - if this also/still fail, then the problem is more likely on Senec's infrastructure side...

thesmer, 2024-03-18:
provided data via direct mail.

CubieMedia, 2024-03-18:
I can confirm this is still working, seems to be a user bases connection / authentication issue. With your phone you can try to connect though a neighbours wifi. Then you definitly have a different external ip address.

Good luck!

marq24, 2024-03-22:
so do you have any update on this - did SENEC fix it?

marq24, 2024-03-23:
I have a waterkotte - so I can't help/suggest anything in this new subject...

closed 2024-03-23 by marq24