marq24/ha-senec-v3 #86: function

sbetti1973, 2024-03-23:
wath is the realy function of "senec_master_sockets_1_upper_limit" or
"senec_master_sockets_1_automatic_enable"

thank you

marq24, 2024-03-23:
These are the sensors that have been added if you have bought the technic-package -> https://github.com/marq24/ha-senec-v3/issues/63 [the names of the sensor's are indeed a bit difficult to understand - but with the translations it should match the names of the original Senec GUI

sbetti1973, 2024-03-23:
thank you for fast replay, i not have a technic-package, i not have a user/password for this, the installator not give me, you know if it's possible to have it?

marq24, 2024-03-23:
well - you have to purchase it from SENEC...

closed 2024-03-28 by marq24