marq24/ha-senec-v3 #92: Everything looks fine but entitys do not show up in energy dashbord list

toben1997, 2024-07-25:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Everything looks fine; the integration works, the entities are available and show the correct data, and the entities also have the correct attributes. However, they are not available in the energy dashboard. I have already deleted the database v2 to get rid of any problems there. I don't think it's an integration problem, but I thought I might still ask here first and then go from there.

i put everything that is conectet so senec in to the logs from the full logs because the full logs are to big i needet i can send the file 

### Add your DEBUG log output

```shell
2024-07-25 18:28:36.928 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration senec which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2024-07-25 18:28:51.023 INFO (MainThread) [custom_components.senec] Starting SENEC.Home Integration 'Senec' with interval:0:00:30 - ConfigEntry: {'data': {'name': 'Senec', 'host': '<MASKED>', 'use_https': True, 'scan_interval': 30, 'type': 'senec', 'dtype_int': 18, 'dtype': 'SENEC Main-Unit', 'dname': 'SENEC.Home V3 hybrid duo | SENEC.Inverter V3 LV', 'dserial': '<MASKED>', 'version': 'App:0827 FW:2307 NPU-Image:2310(v10)'}, 'disabled_by': None, 'domain': 'senec', 'entry_id': 'c03fccfca86737ad8639e5901742a18d', 'minor_version': 1, 'options': {}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Senec', 'unique_id': None, 'version': 1}
2024-07-25 18:28:51.024 INFO (MainThread) [custom_components.senec] ***** QUERY_BMS-DATA ********
2024-07-25 18:28:51.024 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting Senec lala.cgi integration... for host: '172.16.13.192/' with options: {'ignore_system_state': False, 'query_wallbox_data': False, 'query_wallbox_data_via_app_api': False, 'query_bms_data': True, 'query_fan_data': False, 'query_sockets_data': False}
2024-07-25 18:28:51.739 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.715 seconds (success: True)

2024-07-25 18:29:31.013 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.262 seconds (success: True)

2024-07-25 18:30:01.013 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.261 seconds (success: True)

2024-07-25 18:30:31.015 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.264 seconds (success: True)

2024-07-25 18:31:01.015 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.263 seconds (success: True)

2024-07-25 18:31:31.010 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.258 seconds (success: True)

2024-07-25 18:32:01.010 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.258 seconds (success: True)

2024-07-25 18:32:31.048 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.297 seconds (success: True)

2024-07-25 18:33:01.024 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.273 seconds (success: True)

2024-07-25 18:33:31.018 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.266 seconds (success: True)

2024-07-25 18:34:01.016 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.264 seconds (success: True)

2024-07-25 18:34:19.132 WARNING (MainThread) [homeassistant.core] Something is blocking Home Assistant from wrapping up the start up phase. We're going to continue anyway. Please report the following info at https://github.com/home-assistant/core/issues: tuya.binary_sensor, androidtv.remote, schedule, tts, xiaomi_ble.event, recorder, tuya.sensor, mobile_app.device_tracker, tuya.number, tapo.sensor, event, panel_custom, map, spotify.media_player, bosch_shc.light, dlna_dmr, senec.number, yamaha_musiccast.select, input_text, tasmota.light, tasmota.cover, rest_command, oralb.sensor, bosch_shc, switch_as_x.light, image, tradfri.sensor, unifi.sensor, websocket_api, nodered.sensor, bosch_shc.number, tapo.number, tapo.climate, number, bosch_shc.climate, bosch_shc.switch, person, switchbot.cover, sun.sensor, sonoff.remote, unifi.button, calendar, tractive.switch, frontend, tasmota, unifi, androidtv_remote.media_player, reolink.button, tuya.cover, yamaha_musiccast.media_player, intent, shelly.climate, tuya.light, tractive.binary_sensor, wiz.sensor, alexa_media, wyzeapi.lock, shelly.switch, speedtestdotnet.sensor, auth, bosch_shc.sensor, oralb, alarmo.alarm_control_panel, input_datetime, senec.sensor, alarmo, switchbot_cloud.switch, spotify, device_tracker, androidtv_remote.remote, nmap_tracker, cast, alexa_media.sensor, blueprint, cloud, matter.event, bambu_lab, history, tasmota.switch, esphome.number, wled.light, mqtt.number, tag, update, samsungtv, tuya.vacuum, cloud.stt, adaptive_lighting, sonoff.switch, webhook, androidtv, home_connect.sensor, shopping_list.todo, group.binary_sensor, shelly.event, humidifier, mqtt.light, hassio.update, wake_word, yamaha_musiccast.switch, switchbot.sensor, repairs, automation, alarm_control_panel, tractive.sensor, mobile_app.notify, wiz.switch, samsungtv.remote, tasmota.binary_sensor, logbook, wled.number, reolink.switch, switchbot_cloud.sensor, xiaomi_ble.sensor, energy.sensor, rest.sensor, tractive, bambu_lab.sensor, local_todo.todo, home_connect_alt.time, cover, group, usb, matter, dlna_dmr.media_player, backup, tuya.humidifier, androidtv.media_player, unifi.switch, senec.switch, integration.sensor, device_automation, script, tapo, camera, matter.binary_sensor, shelly.button, bosch_shc.valve, mqtt.button, tuya.fan, tapo.siren, sonoff, scheduler.switch, shelly.update, switch, esphome.button, tuya.alarm_control_panel, ssdp, matter.lock, wyzeapi.switch, home_connect, wyzeapi.alarm_control_panel, shelly, input_button, scene, wiz.binary_sensor, switchbot_cloud.climate, ble_monitor, bosch_shc.button, wled.button, ibeacon.device_tracker, weather, climate_group.climate, image_upload, tapo.switch, bambu_lab.button, wake_on_lan.switch, reolink.light, bambu_lab.fan, unifi.device_tracker, cloud.tts, switch_as_x, reolink.sensor, unifi.update, wake_on_lan, co2signal, samsungtv.media_player, sonoff.sensor, matter.cover, bambu_lab.image, tapo.light, conversation, ffmpeg, valve, system_log, template.sensor, wled.select, logger, mobile_app.sensor, wundergroundpws, matter.switch, template.binary_sensor, mqtt.update, tuya.button, hassio.sensor, unifi.image, home_connect_alt, adaptive_lighting.switch, esphome.binary_sensor, knx.sensor, shelly.valve, shelly.binary_sensor, input_number, fan, sonoff.cover, homeassistant.scene, alexa_media.light, time_date.sensor, homeassistant, tractive.device_tracker, ibeacon, esphome.sensor, default_config, yamaha_musiccast, bosch_shc.event, config, watchman, better_thermostat, wundergroundpws.sensor, wiz.light, senec.select, nodered.binary_sensor, nmap_tracker.device_tracker, hacs, yamaha_musiccast.number, todo, home_connect_alt.switch, wled.update, onboarding, home_connect.switch, xiaomi_ble.binary_sensor, reolin [... cut ...] t The system is waiting for tasks: {<Task pending name='Task-8712' coro=<HacsBase.startup_tasks() running at /config/custom_components/hacs/base.py:695> wait_for=<_GatheringFuture pending cb=[Task.task_wakeup()]> cb=[set.remove()]>}

2024-07-25 18:34:36.536 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 4.605 seconds (success: True)

2024-07-25 18:35:07.058 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.306 seconds (success: True)
```

toben1997, 2024-07-25:
here a few screenshots 

[image]
[image]
[image]

marq24, 2024-07-25:
Well looks like I have to update the documentation - for the EnergyDashboard you need ENERGY Sensors - kWHours - these sensors are no longer available directly from the SENEC device - you need to fetch the data either via the web-api-integration or you need to use Riemann Sum Integal Platform -> 

https://github.com/marq24/ha-senec-v3/issues/4
https://github.com/marq24/ha-senec-v3/issues/11

closed 2024-07-25 by marq24