marq24/ha-senec-v3 #96: Failed to load Senec Web-API since latest HA-Version

T3rr4t3c, 2024-08-09:
### With which version of the integration you have an issue?

WEB.API: mein-senec.de Portal

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Since i updated to HA Version 2024.8 the Senec Integration (WEB-API) does not work anymore. The local senec integration keeps working without problems. I have already tried to re-install and re-configure, but it keeps not loading anymore.

Today I have received my new LFP Modules and the storage had previously been switched off for several days.

But now everything is on and working again, but the integration still does not work.
[image]

I have attached the debug log.

Thank you very much in advance.

### Add your DEBUG log output

```shell
2024-08-09 10:41:22.533 ERROR (MainThread) [custom_components.senec] Error fetching senec data: 
2024-08-09 10:41:35.164 ERROR (MainThread) [custom_components.senec] Error fetching senec data: 
2024-08-09 10:41:41.936 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/senec/sensor.py", line 109, in state
    value = getattr(self.coordinator.senec, self.entity_description.key)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 1015, in bms_voltage_c
    return self._raw["BMS"]["VOLTAGE"][2]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-08-09 10:41:57.770 ERROR (MainThread) [custom_components.senec] Error fetching senec data: 
2024-08-09 10:42:11.909 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/senec/sensor.py", line 109, in state
    value = getattr(self.coordinator.senec, self.entity_description.key)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 1015, in bms_voltage_c
    return self._raw["BMS"]["VOLTAGE"][2]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-08-09 10:42:40.161 ERROR (MainThread) [custom_components.senec] Error fetching senec data: 
2024-08-09 10:42:41.926 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/senec/sensor.py", line 109, in state
    value = getattr(self.coordinator.senec, self.entity_description.key)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 1015, in bms_voltage_c
    return self._raw["BMS"]["VOLTAGE"][2]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-08-09 10:43:11.921 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/senec/sensor.py", line 109, in state
    value = getattr(self.coordinator.senec, self.entity_description.key)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/senec/pysenec_ha/__init__.py", line 1015, in bms_voltage_c
    return self._raw["BMS"]["VOLTAGE"][2]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

T3rr4t3c, 2024-08-09:
A litte addition:

The mein-senec.de shows me this now instead of current data:

[image]

But in the "Statistics Data" I can download all CSV Data with valid values for the past days including today.

marq24, 2024-08-09:
you have already LFP Modules ?! - lucky you!...

I have to admit, that I can't connect the log output with the reported error - since `self._raw["BMS"]["VOLTAGE"][2]` is code from the local Integration - and not the webAPI...

Can you please double check:
1) Remove the WebAPI
2) restart HA
3) the Exceptions should be still present ?!

T3rr4t3c, 2024-08-09:
Thank you for the fast response.

Yes since august many users get the LFP modules... Akku is working great again :)

Could you please re-check my post, I have edited it with more debug logs and added #DEBUG LOG before the new logs.

If I remove the WebAPI and add it later - can you tell me if my sensors and helpers will still be there afterwards?

marq24, 2024-08-09:
My assumption right now is, that you have 2 battery modules installed, but you are trying to read data from Module 3 (& 4)...

T3rr4t3c, 2024-08-09:
That could well be the case, I previously had 3 modules (2.5 kWh each = 7.5 kWh) and now 2 modules (5 kWh each = 10 kWh)

marq24, 2024-08-09:
so - then this is no surprise - deactivate the Sensors... (and yes - error handling can be improved)...

T3rr4t3c, 2024-08-09:
Superb, i have deactivated the module 3 entities and re-added the web-api integration and now all seems to work again. 

Thank you very much and keep up the great work!

marq24, 2024-08-09:
fixed by https://github.com/marq24/ha-senec-v3/releases/tag/2024.8.2

closed 2024-08-09 by marq24