marq24/ha-senec-v3 #97: Hallo, habe heute die neuen Lfp Akkus erhalten, MCU Version 8828, nun geht nichts mehr auf HA

Zeppelin-PV, 2024-08-13:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

marq24, 2024-08-13:
WITHOUT debug log there is no way I could help...

Zeppelin-PV, 2024-08-13:
SENEC für HA funktioniert wieder
aus Integration und aus HACS entfernt /gelöscht
Neu aufgesetzt, alles wieder da

closed 2024-08-13 by marq24