marq24/ha-senec-v3 #98: After update to home assistant 2024.8.1 WebAPI throws an error on initialization.

KalleRau, 2024-08-13:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Today I updated home assistant to the newest version 2024.8.1.

After restarting HA the integration senec_WEBAPI throws  an error on initialization.

[image]

I am using the newest integration of the senec Integration( see Screenshot) .

I am using the same incredentials like in the Mein-Senec.de portal.

[image]

In the debug output there are several division by zero errors because HA doesn't know the sensors from the WebAPI, but that doesn't really matter.

For the same reason there is also no information in the energy dashboard.

I have to mention that the system is in remote shutdown and therefore no values for storage are available, because I'm waiting for the new LFP Modules.

[image]

Before the update to HA 2024.8.1 everything worked fine.

Please find the Debug output log attached.

Currently I'm switching back to the former Version of HA (2024.7.4x). 

Good evening.

If you have any questions , please ask.

### Add your DEBUG log output

```shell
.....

2024-08-13 14:26:41.464 INFO (MainThread) [custom_components.senec] Starting SENEC.Home Integration 'senec' with interval:0:00:30 - ConfigEntry: {'created_at': '1970-01-01T00:00:00+00:00', 'data': {'name': 'senec', 'host': '<MASKED>', 'use_https': True, 'scan_interval': 30, 'type': 'senec', 'dtype_int': 18, 'dtype': 'SENEC Main-Unit', 'dname': 'SENEC.Home V3 hybrid duo | SENEC.Inverter V3 LV', 'dserial': '<MASKED>', 'version': 'App:1827 FW:2307 NPU-Image:2310(v10)'}, 'disabled_by': None, 'domain': 'senec', 'entry_id': '133d0a87af51d8d7377669c181998445', 'minor_version': 1, 'modified_at': '1970-01-01T00:00:00+00:00', 'options': {}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'senec', 'unique_id': None, 'version': 1}
2024-08-13 14:26:41.465 INFO (MainThread) [custom_components.senec] ***** QUERY_BMS-DATA ********
2024-08-13 14:26:41.465 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting Senec lala.cgi integration... for host: '192.168.178.99' with options: {'ignore_system_state': False, 'query_wallbox_data': False, 'query_wallbox_data_via_app_api': False, 'query_bms_data': True, 'query_fan_data': False, 'query_sockets_data': False}
2024-08-13 14:26:41.468 INFO (MainThread) [custom_components.senec] Starting SENEC.Home Integration 'senec_WEBAPI' with interval:0:05:00 - ConfigEntry: {'created_at': '1970-01-01T00:00:00+00:00', 'data': {'name': 'senec_WEBAPI', 'host': '<MASKED>', 'username': 'k******@*****.de', 'password': '<MASKED>', 'scan_interval': 300, 'type': 'web', 'dtype_int': '', 'dtype': 'SENEC WebAPI', 'dname': 'SENEC.Home V3 hybrid duo | SENEC.Num: 784868110', 'dserial': '<MASKED>', 'version': 'App:1967 FW:1827 Inverter: v3.16.20', 'master_plant_number': 0, 'app_token': '<MASKED>', 'app_master_plant_id': '<MASKED>', 'app_wallbox_num_max': 0}, 'disabled_by': None, 'domain': 'senec', 'entry_id': 'd6d3097381bd1d907be8846f8c661462', 'minor_version': 1, 'modified_at': '1970-01-01T00:00:00+00:00', 'options': {}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'senec_WEBAPI', 'unique_id': None, 'version': 1}
2024-08-13 14:26:41.468 INFO (MainThread) [custom_components.senec] ***** QUERY_PEAK_SHAVING! ********
2024-08-13 14:26:41.468 INFO (MainThread) [custom_components.senec.pysenec_ha] restarting MySenecWebPortal... for user: 'k******@*****.de' with options: {'query_wallbox_data': False, 'query_spare_capacity': False, 'query_peak_shaving': True, 'app_wallbox_num_max': 0, 'app_token': '<MASKED>', 'app_master_plant_id': '<MASKED>'}
2024-08-13 14:26:41.469 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** update(self) ********
2024-08-13 14:26:41.469 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_now(self) ********
2024-08-13 14:26:41.469 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_data(self) ********
2024-08-13 14:26:41.469 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API get https://app-gateway.prod.senec.dev/v2/senec/systems/287927/dashboard
2024-08-13 14:26:41.740 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 0.274 seconds (success: True)
2024-08-13 14:26:41.745 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API response: {'currently': {'powerGenerationInW': 3054.27221760797, 'powerConsumptionInW': 817.58, 'gridFeedInInW': 2242.5249, 'gridDrawInW': 1e-05, 'batteryChargeInW': 1e-05, 'batteryDischargeInW': 1e-05, 'batteryLevelInPercent': 1e-05, 'selfSufficiencyInPercent': 100.0, 'wallboxInW': 1e-05}, 'today': {'powerGenerationInWh': 5423.828125, 'powerConsumptionInWh': 8271.48, 'gridFeedInInWh': 2096.19140625, 'gridDrawInWh': 4943.359375, 'batteryChargeInWh': 0.0, 'batteryDischargeInWh': 0.0, 'batteryLevelInPercent': 1e-05, 'selfSufficiencyInPercent': 40.24, 'wallboxInWh': 0.0}, 'timestamp': '2024-08-13T12:21:44Z', 'electricVehicleConnected': False}
2024-08-13 14:26:41.745 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_update_total(self) ********
2024-08-13 14:26:41.746 DEBUG (MainThread) [custom_components.senec.pysenec_ha] ***** APP-API: app_get_data(self) ********
2024-08-13 14:26:41.746 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API get https://app-gateway.prod.senec.dev/v1/senec/monitor/287927/data/custom?startDate=2018-01-01&endDate=2024-09-01&locale=de_DE&timezone=GMT
2024-08-13 14:26:42.861 DEBUG (MainThread) [custom_components.senec.pysenec_ha] APP-API response: {'overviewMonitorCard': {'gridFeedInInKwh': {'value': 4467.0048828125, 'unit': 'kWh'}, 'gridFeedOutInKwh': {'value': 2418.169921875, 'unit': 'kWh'}, 'generatedElectricityInKwh': {'value': 7292.87060546875, 'unit': 'kWh'}, 'consumedElectricityInKwh': {'value': 5207.794921875, 'unit': 'kWh'}}, 'energySourcesMonitorCard': {'pvUsageInPercent': {'value': 71.21464254453542, 'unit': '%'}, 'batteryUsageInPercent': {'value': 5.184897813021035, 'unit': '%'}, 'gridUsageInPercent': {'value': 23.60045964244354, 'unit': '%'}}, 'pvStatsMonitorCard': {'selfConsumptionInPercent': {'value': 38.748332111325276, 'unit': '%'}, 'gridFeedInInPercent': {'value': 61.251667888674724, 'unit': '%'}, 'pvGeneration': {'value': 7292.87060546875, 'unit': 'kWh'}}, 'savingsMonitorCard': {'savedElectricityCosts': {'value': 2708.572142871094, 'unit': 'euro'}, 'avoidedCo2Emissions': {'value': 2924.441112792969, 'unit': 'kg'}, 'treesToBePlanted': 186.27013457280057}, 'aggregation': {'startDate': '2018-01-01T00:00:00Z', 'gridConsumption': {'value': 2418.169921875, 'unit': 'kWh'}, 'gridFeedIn': {'value': 4467.0048828125, 'unit': 'kWh'}, 'storageLoad': {'value': 565.8899536132812, 'unit': 'kWh'}, 'storageConsumption': {'value': 530.9692993164062, 'unit': 'kWh'}, 'storageLevel': {'value': 7.5471272468566895, 'unit': '%'}, 'generation': {'value': 7292.87060546875, 'unit': 'kWh'}, 'totalUsage': {'value': 5207.794921875, 'unit': 'kWh'}}, 'intervals': [{'startDate': '2018-01-01T00:00:00Z', 'gridConsumption': {'value': 0.0, 'unit': 'kWh'}, 'gridFeedIn': {'value': 0.0, 'unit': 'kWh'}, 'storageLoad': {'value': 0.0, 'unit': 'kWh'}, 'storageC [... cut ...] : 8.371176, 'latitude': 48.993487, 'timezone': 'Europe/Berlin'}, 'onlineMonitoringAllowed': True, 'website': 'www.hatrapv.de'}, 'slaveSystemInfoDtos': []}
2024-08-13 14:26:43.454 INFO (MainThread) [custom_components.senec.pysenec_ha] ***** authenticate(self) ********
2024-08-13 14:26:43.767 ERROR (MainThread) [custom_components.senec] Error fetching senec data: 
2024-08-13 14:26:43.768 DEBUG (MainThread) [custom_components.senec] Finished fetching senec data in 2.299 seconds (success: False)
```

marq24, 2024-08-13:
it could be that the APP-TOKEN that is stored in the config_entry is invalid with the updated HA installation?! - Can you please be so kind and remove the current webAPI config and simply create a new one (that should generated an updated APP-Token) ?!

marq24, 2024-08-13:
additionally you might like to check, if this issue will disapear, once you disable the `*_gridexport_limit` entity ?!

marq24, 2024-08-13:
ok - please disable the sensors:

`*_gridexport_limit`
`*_peakshaving_mode`
`*_peakshaving_capacitylimit`
`*_peakshaving_enddate`

and then restart HA... this should TEMP fix the issue (till I found the root cause)

closed 2024-08-13 by marq24

marq24, 2024-08-13:
thanks for your report - this is also related to #95 - but only occoured if peakshaving sensos are active - so finally this update of aiohttp lib related issue is fixed now with the new release (2024.8.3)

KalleRau, 2024-08-14:
Thanks for the quick response.

The update to version 8.3 did fix the issue.

The intergration works again like expected.

Thanks again for the your help.
Keep up the good work.

👍 😄 👍

commit 71017be8 (2024-08-13): fix #98