marq24/ha-senec-v3 #100: Missing storage module values (Module SoH, Cylces etc.)

cuesheet2, 2024-08-24:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Firmware: App:0826 FW:2307 NPU-Image:2310(v10)

Since 18.08., the power storage unit no longer displays the data for the individual storage modules like SoH, cycles etc.. Does anyone else have a similar problem? Or has something been changed on Senec side? 

[image]

marq24, 2024-08-24:
Well - was soll ich sagen? Ich benötige den DEBUG Log output - und idealerweise den direkten output von einem lokalen lala.cgi POST an Deinen Speicher... Bekommst Du das mit so einem Tool wie CURL hin?

marq24, 2024-08-24:
bitte mal auf die 2024.8.4 updaten...

cuesheet2, 2024-08-25:
Wollte gerade die Debuglogs hochladen. Fehler wurde durch den Patch behoben. Vielen lieben Dank 👍

closed 2024-08-25 by marq24