marq24/ha-senec-v3 #101: Update not possible

electricdelicate, 2024-08-25:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Since the new version (2.0) of HACS I can´t update this intergration anymore.

[image]

marq24, 2024-08-25:
well

1)
this is HACS 3.0... updates should now pop up in the HA**SS** settings ?!

2)
you have obviously NOT installed the latest version of the integration - so why checking then the checkbox?

electricdelicate, 2024-08-25:
I´m just a little bit confused .... I refered to this:
[image]

But right now, I´ve found the solution. From now the HACS downloaded Repo updates will be shown here: 
[image]

closed 2024-08-25 by marq24

ETH1166, 2024-08-25:
Marc, I would Like to apologize in advance, but could you please explain, whether your Senec Plugin is compatible with HACS 2.0? 
Reading the changelog I do not dare to try the Update without asking...

marq24, 2024-08-25:
> Marc, I would Like to apologize in advance, but could you please explain, whether your Senec Plugin is compatible with HACS 2.0? Reading the changelog I do not dare to try the Update without asking...

of course it is... don't get confused - it's just that HACS report you updates different than before...

ETH1166, 2024-08-25:
> of course it is... don't get confused - it's just that HACS report you updates different than before...

Thank you very much for the clarification!