marq24/ha-senec-v3 #104: Accessing `hass.helpers.aiohttp_client` will stop working in HA 2024.11

sehraf, 2024-09-18:
### With which version of the integration you have an issue?

WEB.API: mein-senec.de Portal

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I've setup the web api and found this in the logs: Home Assistant complains about usage of a deprecated feature:
> WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'senec' accesses hass.helpers.aiohttp_client. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from aiohttp_client directly at custom_components/senec/config_flow.py, line 177: web_session = self.hass.helpers.aiohttp_client.async_create_clientsession(auto_cleanup=False), please create a bug report at https://github.com/marq24/ha-senec-v3/issues

While 2024.11 isn't around the corner, the "will stop working" part might be of your interest.

### Add your DEBUG log output

```shell
2024-09-18 12:55:40.536 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'senec' accesses hass.helpers.aiohttp_client. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from aiohttp_client directly at custom_components/senec/config_flow.py, line 177: web_session = self.hass.helpers.aiohttp_client.async_create_clientsession(auto_cleanup=False), please create a bug report at https://github.com/marq24/ha-senec-v3/issues
```

marq24, 2024-09-18:
thanks for the hint - will be fixed till Nov 2024

marq24, 2024-10-31:
done...

closed 2024-10-31 by marq24