marq24/ha-senec-v3 #105: Angezeigte Werte der Sensoren lassen sich nach dem Update auf HASS 2024.9.2 nicht mehr ändern/ einstellen

Schlolle, 2024-09-18:
### With which version of the integration you have an issue?

SENEC.Home V3 hybrid/SENEC.Home V3 hybrid duo/SENEC.Home V2.1 (or older) 

### Multiple Systems Environment?

- [ ] I have multiple SENEC Systems (master/slave configuration)

### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - Of course this could be always the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the SENEC.Home device itself already provided this data
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Ich habe die beiden Module senec und senec_WEBAPI in Version 2024.9.0 installiert. Nach dem HASS-Update von 2024.9.1 auf 2024.9.2 erscheinen die Werte der senec-sensoren wieder mit allen Dezimalstellen hinter dem Komma. Das ist auch nicht mehr anders zu konfigurieren.

marq24, 2024-09-18:
works fine in my office - more info required

Schlolle, 2024-09-18:
[image]
[home-assistant_senec_2024-09-18T17-49-17.069Z.log](https://github.com/user-attachments/files/17048286/home-assistant_senec_2024-09-18T17-49-17.069Z.log)
[image]
[image]
Hier die Screenshots. Nach dem Update lässt die die Anzeigegenauigkeit der Werte nicht mehr einstellen.

Schlolle, 2024-09-18:
Ich sehe jetzt erst, dass nach dem Update im Einstelldialog das Popup zum Einstellen der Maßeinheit komplett fehlt.

marq24, 2024-09-18:
Du sagst also, dass es nach dem Update (von HA) bei Dir keine Auswahlbox zum Einstellen der Anzeigegenauigkeit mehr gibt? 

Und as ist NUR bei der Senec Integration der Fall?!

Schlolle, 2024-09-18:
Nicht ganz, nach dem Update gibt es schon noch die Auswahlbox. Ein Ändern der Genauigkeit ist aber unwirksam. Die Auswahlbox hat auch kein Pop-up mehr zum Einstellen der Einheit, z.B. von W auf kW. Die angezeigten Werte haben dann auch keine Einheit mehr, die hinter dem Wert erscheinen würde.

marq24, 2024-09-18:
wie gesagt - __hier ist alles normal__ - ggf mal die Integration rauswerfen und wieder neu hinzufügen - kann sein das was an Deiner Entity DB quer liegt...

Schlolle, 2024-09-20:
Ich habe nun heute die Integration in HACS komplett herausgeworfen, neu heruntergeladen und installiert, die Geräte (Speicher und webapi) neu installiert und eingerichtet. Das Ergebnis ist dasselbe: die SENEC-Werte erscheinen mit allen Nachkommastellen, ein Änderung in der Dialogbox hat keinen Effekt. Die Popups zum Wechseln der Einheit z.B. von W auf kW sind verschwunden.

marq24, 2024-09-20:
es tut mir leid - ich habe gestern eine komplett neue HA Instanz aufgesetzt (Docker) - HACS installiert, die Integration installiert - und alles läuft wie es soll/muss... und nu?!

Es kann sich nochmal lohnen die DEBUG logs zu aktivieren, um festzustellen, ob es da irgendwo in der INIT Phase eine Herausforderung bei Dir lokal gibt... Wenn dem nicht so ist einmal bitte mit "Scully & Mulder" telefonieren...

Schlolle, 2024-09-24:
Ich denke ich habe das Problem gefunden. Vermutlich ist es der Browsercache gewesen. Ich habe den Browser während und nach dem Update auf der HASS-Config-Seite geöffnet gelassen. Die Seite hat sich zwar im Hintergrund aktualisiert, dann aber die falsch gerundeten Werte angezeigt. Nach dem Schließen, Wiederöffnen und Neuanmelden sehen die Werte nun gut aus. Auch das Popup mit der Einstellung der Einheiten ist zurückgekehrt. Ich habe das nicht gemerkt, weil ich das Update immer umgehend wieder rückgängig gemacht habe. Sorry für die Umstände!

closed 2024-09-24 by marq24