marq24/ha-senec-v3 #107: New Wallbox Charge Mode

schimi42, 2024-10-19:
The SENEC App provides a new Wallbox charge mode "Komfort-Laden". This comes in addition to the old modes "solar optimized" and "fast". There are also additional settings for this mode. 

If you are interested in adding this mode to the Home Assistant integration I can offer my help with providing any required data and/or with testing.

marq24, 2024-10-19:
sure! - looking forward to get details!

schimi42, 2024-10-19:
I did some testing with the app and monitored the log file. Looks like this now comfort mode is handled on the server, not locally. It just sets the smart charge on or off, as well as the interchange. So nothing to add to the integration.