marq24/ha-senec-v3 #108: Some values in the senec-webapi are unavailable

Schlolle, 2024-11-01:
Without having anything changed in HA, some values in the senec-webapi are unavailable since midnight.
[image]

marq24, 2024-11-01:
check mein-senec.de if the data is there... if yes - restart HA - check the debug log for any login/credential issues - you might want to remove and re-add the installation...

vmpr, 2024-11-01:
my senec app told me today that they deployed a new API :(

Schlolle, 2024-11-01:
Means that an adaption of the senec-webapi-integration is necessary?

unavailable (Issue #108)

view it on GitHub, or
M5WDAVCNFSM6AAAAABQ74DJ5WVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDINJRGUZDM
OBQGI> unsubscribe.
<https://github.com/notifications/beacon/A47BLHYBOQQG7M4X3GHACGDZ6M5WDA5CNFS
M6AAAAABQ74DJ5WWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU

marq24, 2024-11-01:
When you check the ha logs, then you can see, that the requests:

```
Calling https://app-gateway.prod.senec.dev/v1/senec/monitor/XXX/data/custom?startDate=2018-01-01&endDate=2024-12-01&locale=de_DE&timezone=GMT caused [HTTP 400]: 400, message='', url='https://app-gateway.prod.senec.dev/v1/senec/monitor/XXX/data/custom?startDate=2018-01-01&endDate=2024-12-01&locale=de_DE&timezone=GMT' this is SO RIDICULOUS Senec - returning 400 without error code when TOKEN is invalid
```

failing - so indeed looks like, that SENEC have shut down the old monitor endpoint... so an integration update will be required... good that I am located in NRW and we have public holiday!

marq24, 2024-11-01:
update to https://github.com/marq24/ha-senec-v3/releases/tag/2024.11.0 should solve the problem... thanks SENEC for ruin my All Saints Day!

marq24, 2024-11-01:
from the 3 dot's in HACS for the Integration you can select 'Aktualisieren' - then in your Settings you have the option to update...

Schlolle, 2024-11-01:
Thanks a lot Matthias for your time you spend into this project!

Tranquality450, 2024-11-01:
Thanks for you super-fast update!!

ETH1166, 2024-11-01:
> update to https://github.com/marq24/ha-senec-v3/releases/tag/2024.11.0 should solve the problem... thanks SENEC for ruin my All Saints Day!

Thank you so much.
You are marvellous!

And have a nice remaining All Saints day!

Thunderbolt, 2024-11-01:
Thanks a lot for the quick fix!!!
I received a survey in the Senec App last night, 'Would you recommend Senec to your friends, acquaintances?' I answered it negatively, because of the missing open interfaces and cloud connection... but when I tried to submit it, an error occurred and my answer couldn't be transmitted. How Senec-like😅

closed 2024-11-03 by marq24